Koenkk/zigbee2mqtt #548: Unable to connect to MQTT Server

MNEWI, 2018-10-28:
Dear all,

need your input for continuing setting up my MQTT Server for use with ioBroker...
Most likely caused by a stupid thing which I could not find... 

**Facts:**
Running on a CC2531 stick with latest FW
Installation on a PI3 according to "Running the bridge"section
port is listed in devices and active
Green LED is on when error appears (unplug stick , plug in and reset for next start)

**Log file shows:**
2018-10-28 18:28:00 - info: Logging to directory: '/opt/zigbee2mqtt/data/log/2018-10-28.18-28-00'
2018-10-28 18:28:01 - info: Starting zigbee2mqtt version 0.1.8 (commit #97cc56b)
2018-10-28 18:28:01 - info: Starting zigbee-shepherd
2018-10-28 18:28:02 - info: zigbee-shepherd started
2018-10-28 18:28:02 - info: Coordinator firmware version: '20180815'
2018-10-28 18:28:02 - info: Currently 0 devices are joined:
2018-10-28 18:28:02 - warn: `permit_join` set to  `true` in configuration.yaml.
2018-10-28 18:28:02 - warn: Allowing new devices to join.
2018-10-28 18:28:02 - warn: Set `permit_join` to `false` once you joined all devices.
2018-10-28 18:28:02 - info: Zigbee: allowing new devices to join.
2018-10-28 18:28:02 - info: Connecting to MQTT server at mqtt://192.168.178.47
2018-10-28 18:28:03 - info: zigbee-shepherd ready
2018-10-28 18:28:13 - error: Not connected to MQTT server!
2018-10-28 18:28:23 - error: Not connected to MQTT server!
2018-10-28 18:28:33 - error: Not connected to MQTT server!
2018-10-28 18:28:38 - error: Not connected to MQTT server!
2018-10-28 18:28:38 - error: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
2018-10-28 18:28:38 - info: zigbee-shepherd stopped

**Config is as follows:**
# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://192.168.178.47'
  # MQTT server authentication, uncomment if required:
# user: pi
# password: mypwd

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyACM0

Tried also to define the long termination for the port but didn't work either
Checked the Wiki and issues but didn't find anything similar so far...

Each advice is very much appreciated.
Thanks !

Regards
Michael

MNEWI, 2018-10-29:
Hi Koen, I checked it out and hope I did configure MQTT.FX correctly.
But->No success, no connection established..

Any further ideas ?

Koenkk, 2018-10-30:
That means the MQTT server isn't exposed. What MQTT broker do you run?

MNEWI, 2018-10-30:
I am going to use ioBroker with the standard Zigbee adapter.
But I understood that the iobroker Zigbee adapter is communicating with the MQTT Server and that on the other hand with the stick ?!

Koenkk, 2018-10-30:
No, you can use iobroker as a replacement for zigbee2mqtt

MNEWI, 2018-10-30:
I have to admit now I was confused ...
Using zigbee adapter 0.7.7 in ioBroker refers to your github page with the firmware to flash  "CC2531ZNP-Prod.hex". Following the "Running bridge" section I have to install the MQTT server.
Would be helpful to mention in that section that it is not valid for iobroker... :)

Re-installing the ioBroker zigbee adapter I was now able to pair devices.

Thanks a lot !

closed 2018-10-30 by MNEWI