Koenkk/zigbee2mqtt #549: How to add support for an unsupported device when running the docker container?

poldim, 2018-10-28:
I'm running this in a Docker.  I'm trying to figure out how to expose the `node_modules/zigbee-shepherd-converters` folder? 

I've tried both of these volume mappings but they create error in starting the container:

- `- ./appdata/home-assistant/zigbee2mqtt2/:/app/`

- `- ./appdata/home-assistant/zigbee2mqtt/zigbee-shepherd-converters/:/app/node_modules/zigbee-shepherd-converters`

This is my `docker-compose` config for running the container:

```
  homeassistant_zigbee2mqtt:
    image: koenkk/zigbee2mqtt
    container_name: homeassistant_zigbee2mqtt
    restart: always
    network_mode: host
    volumes:
      - ./appdata/home-assistant/zigbee2mqtt/:/app/data/
    devices:
      - "/dev/ttyACM0"
    environment:
      - TZ=${TZ}
```

Koenkk, 2018-10-30:
First checkout zigbee-shepherd-converters locally:
```
cd /opt
git clone https://github.com/Koenkk/zigbee-shepherd-converters.git
cd zigbee-shepherd-converters
npm install
```

To use local checkout in the docker container run the container with the volume:
```bash
/opt/zigbee-shepherd-converters:/app/node_modules/zigbee-shepherd-converters
```

Now make changes in `/opt/zigbee-shepherd-converters/devices.js` and restart the docker container in order to test.

poldim, 2018-10-30:
No go, it errors out using this external volume:
```

Using '/app/data' as data directory

> zigbee2mqtt@0.1.8 start /app

> node index.js

  zigbee2mqtt:info 2018-10-30 15:05:11 Logging to directory: 'data/log/2018-10-30.22-05-11'

module.js:538

    throw err;

    ^

Error: Cannot find module 'debounce'

    at Function.Module._resolveFilename (module.js:536:15)

    at Function.Module._load (module.js:466:25)

    at Module.require (module.js:579:17)

    at require (internal/module.js:11:18)

    at Object.<anonymous> (/app/node_modules/zigbee-shepherd-converters/converters/fromZigbee.js:3:18)

    at Module._compile (module.js:635:30)

    at Object.Module._extensions..js (module.js:646:10)

    at Module.load (module.js:554:32)

    at tryModuleLoad (module.js:497:12)

    at Function.Module._load (module.js:489:3)

npm ERR! code ELIFECYCLE

npm ERR! errno 1

npm ERR! zigbee2mqtt@0.1.8 start: `node index.js`

npm ERR! Exit status 1



 

npm ERR! Failed at the zigbee2mqtt@0.1.8 start script.

npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:

npm ERR!     /root/.npm/_logs/2018-10-30T22_05_12_039Z-debug.log
```

closed 2019-01-06 by stale[bot]

tanus10, 2019-04-02:
It's worth to mention that sometimes git cloned zigbee-shepherd-converters only works with -dev tagged docker image. Currently anyway.

Nosstrom, 2019-09-22:
I dont undestand what do u mean by docker do we have to open ssh session, log in to hassio and do all the thing via command line and VI ? I havent found any documentation about, is really frustrated i have search all over the internet! Please support the people who dont understand this please