Koenkk/zigbee2mqtt #550: Cant change Colortemp on Tradfri bulbs with opnehab

ebivan, 2018-10-29:
Hey there, 
I already posted a question on openhab, but it seems nobody can help me there, well there is not to much of a zigbee2mqtt community over there yet. So I try my luck over here. I want to control my Tradfri GU10 tuneable white bulbs which are connected via zigbee2mqtt with openhab.

Well, the title is not right, I can change the colortemps, but only between 255 and 250, which ist just about the threshold between pure cold light and turning on the warmer ones on their lowest level.

When I change Brightness everything looks fine:

```
  zigbee2mqtt:info 2018-10-29 07:15:38 Zigbee publish to '0x90fd9ffffedcadf4', genLevelCtrl - moveToLevelWithOnOff - {"level":63.75,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2018-10-29 07:15:39 MQTT publish, topic: 'zigbee2mqtt/tradfri-flur-1', payload: '{"state":"ON","brightness":232,"color_temp":250}'
  zigbee2mqtt:info 2018-10-29 07:15:40 MQTT publish, topic: 'zigbee2mqtt/tradfri-flur-1', payload: '{"state":"ON","brightness":63,"color_temp":250}'
```

And when I try to set the colortemp:

```
zigbee2mqtt:info 2018-10-29 07:29:54 Zigbee publish to '0x90fd9ffffedcadf4', lightingColorCtrl - moveToColorTemp - {"colortemp":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2018-10-29 07:29:55 MQTT publish, topic: 'zigbee2mqtt/tradfri-flur-1', payload: '{"state":"ON","brightness":30,"color_temp":255}'
  zigbee2mqtt:info 2018-10-29 07:29:56 MQTT publish, topic: 'zigbee2mqtt/tradfri-flur-1', payload: '{"state":"ON","brightness":30,"color_temp":255}'
  zigbee2mqtt:info 2018-10-29 07:30:01 Zigbee publish to '0x90fd9ffffedcadf4', lightingColorCtrl - moveToColorTemp - {"colortemp":28.05,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2018-10-29 07:30:03 MQTT publish, topic: 'zigbee2mqtt/tradfri-flur-1', payload: '{"state":"ON","brightness":30,"color_temp":250}'
  zigbee2mqtt:info 2018-10-29 07:30:03 MQTT publish, topic: 'zigbee2mqtt/tradfri-flur-1', payload: '{"state":"ON","brightness":30,"color_temp":250}'
```

FutureCow, 2018-10-29:
You might use higher values than 255. My Innr GU10 uses 200 till 455.

nikolaykasyanov, 2018-11-03:
@HKC001 interesting, I have a similar issue with Trådfri color temperature, here's how it looks like in Home Assistant: 
[image]

note how it jumps to an intermediate value after setting it to maximum or minimum.

@Koenkk does it deserve a separate ticket or the temperature range should be customized on Home Assistant side?

Koenkk, 2018-11-03:
@nikolaykasyanov the range home assistant has for color temperature is broader than the tradfri bulbs can handle.

closed 2019-01-10 by stale[bot]