Koenkk/zigbee2mqtt #551: Pair CC2530 Router to a Coordinator

Imperial-Guard, 2018-10-30:
Yesterday I managed to successfully flash 2 CC2530 devices as a router. 

Somehow I added one to zigbee2mqtt, but I did something stupid and removed the router.

How can I repair it again? Somehow it will not pair with my coordinator. 
Any thoughts on this? 

See below the versions and the amount of connected devices:
`> zigbee2mqtt@0.1.8 start /zigbee2mqtt-0.1.8
> node index.js
  zigbee2mqtt:info 2018-10-30 07:08:11 Logging to directory: '/share/zigbee2mqtt/log/2018-10-30.07-08-11'
  zigbee2mqtt:info 2018-10-30 07:08:13 Starting zigbee2mqtt version 0.1.8 (commit #unknown)
  zigbee2mqtt:info 2018-10-30 07:08:13 Starting zigbee-shepherd
  zigbee2mqtt:info 2018-10-30 07:08:14 zigbee-shepherd started
  zigbee2mqtt:info 2018-10-30 07:08:14 Coordinator firmware version: '20180820'
  zigbee2mqtt:info 2018-10-30 07:08:14 Currently 12 devices are joined:
  zigbee2mqtt:info 2018-10-30 07:08:14 hallway/sensor/0003/motionsensor (0x00158d0002b4863d): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
  zigbee2mqtt:info 2018-10-30 07:08:14 living_room/sensor/0003/gardendoorsensor (0x00158d000232c194): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
  zigbee2mqtt:info 2018-10-30 07:08:14 living_room/sensor/0004/doorsensor (0x00158d00022d2d2c): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
  zigbee2mqtt:info 2018-10-30 07:08:14 scullery/sensor/0001/gardendoorsensor (0x00158d000243753d): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
  zigbee2mqtt:info 2018-10-30 07:08:14 scullery/sensor/0002/smokedetector (0x00158d000282298f): JTYJ-GD-01LM/BW - Xiaomi MiJia Honeywell smoke detector (EndDevice)
  zigbee2mqtt:info 2018-10-30 07:08:14 bathroom/sensor/0002/windowsensor (0x00158d00024e548d): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
  zigbee2mqtt:info 2018-10-30 07:08:14 kitchen/sensor/0001/motionsensor (0x00158d0002732adb): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
  zigbee2mqtt:info 2018-10-30 07:08:14 hallway_up/sensor/0001/motionsensor (0x00158d0002753dde): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
  zigbee2mqtt:info 2018-10-30 07:08:14 hallway/sensor/0004/doorsensor (0x00158d000251b6f2): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
  zigbee2mqtt:info 2018-10-30 07:08:14 hallway_up/sensor/0002/smokedetector (0x00158d00028ac409): JTYJ-GD-01LM/BW - Xiaomi MiJia Honeywell smoke detector (EndDevice)
  zigbee2mqtt:info 2018-10-30 07:08:14 spare/sensor/0001/motionsensor (0x00158d0002b45f88): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
  zigbee2mqtt:info 2018-10-30 07:08:14 0x001788011032d314 (0x001788011032d314): 324131092621 - Philips Hue dimmer switch (EndDevice)`

ptvoinfo, 2018-10-30:
@Imperial-Guard Did you allow joining in the settings?

Imperial-Guard, 2018-10-30:
> @Imperial-Guard Did you allow joining in the settings?

Yes I did. Is there a reset function on the CC2530 with dupont wires?

ptvoinfo, 2018-10-30:
@Imperial-Guard Re-fashing fully resets the device.

Imperial-Guard, 2018-10-30:
> @Imperial-Guard Re-fashing fully resets the device.

After flashing the device , is 3.3V and a GND sufficient to pair the device?

ptvoinfo, 2018-10-30:
@Imperial-Guard If you use my firmware, then the answer is 'yes'.

Chrischi-, 2018-10-30:
> @Imperial-Guard Re-fashing fully resets the device.

just fyi & to @all who flash via Ti's SerialBootTool
it seems that this method does **not** clear the flash/database.

Imperial-Guard, 2018-10-31:
Woohoo finally paired to the coordinator. 

Only the link quality between the CC2531 (Coordinator) and the CC2530 (Router) is a bit poor, linkquality +- 18. 

The coordinator is placed in a wooden cabinet and to the router in next room it's about 8 - 10 meter.

khzd, 2018-11-02:
Hello @Imperial-Guard,
I got also a pair of CC2530 from aliexpress and look for instructions to pair them, can you give your detailed steps? 
thanks

Imperial-Guard, 2018-11-02:
> Hello @Imperial-Guard,
> I got also a pair of CC2530 from aliexpress and look for instructions to pair them, can you give your detailed steps?
> thanks
I used this page:

http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/

khzd, 2018-11-02:
thanks for your propmt response, i did the same.
how did you pair with zigbee2mqtt? any action?
thanks

khzd, 2018-11-03:
My bad : my "Allow devices to join" HA automation is not working properly. After rebooting all the system, both CC2530 joined immediately the network without any action :)

after a night of running, my network seems more reliable, to follow...  

Great Work **@ptvoinfo** and thanks for such support.

Imperial-Guard, 2018-11-03:
Solved and working  :)

closed 2018-11-03 by Imperial-Guard

MikeVRT, 2018-11-19:
> @Imperial-Guard If you use my firmware, then the answer is 'yes'.

@ptvoinfo, can you said what must happens in this cases:
1. end device connectet near CC2530 repeater will be closer moved to CC2531 coordinator (where no repeater signal) - end device will works well auto moved to CC2531 coordinator?
2. as 1 but end device near CC2531 coordinator will moved near to CC2530 repeater
3. CC2530 repeater lose power for few hours/days for example (end device near repeater to far from coordinator for connection) and than CC2530  got power again - end device will be losed from network o automatic appears again?
4. end device between ~same signal level from coordinator & repeater - any problems, or always swap and works?

Imperial-Guard, 2018-11-19:
@ptvoinfo same question as @MikeVRT, powered off and on the CC2530 router but it seems to be offline for some reason. Any thoughts on this?

ptvoinfo, 2018-11-20:
@Imperial-Guard All answers are 'yes', but I suppose that device firmware does not contain bugs.

Imperial-Guard, 2018-11-20:
@ptvoinfo Thanks for your info. A few days ago did some testing and saw like every minute the router publishing a message with all the connected notes. 

Right now I don't see any message....

ptvoinfo, 2018-11-20:
@Imperial-Guard The router has 2 versions (debug and normal). Is it possible you flashed the normal version?

nanosonde, 2019-02-13:
Is reflashing the only option to re-pair a CC2530 router to the coordinator again after the database.db was deleted?

Stefan300381, 2019-03-13:
> 
> 
> Is reflashing the only option to re-pair a CC2530 router to the coordinator again after the database.db was deleted?

same question. i have paired the router to a different zigbee network and need to repair

austrisv, 2019-09-01:
foud on the ptvo site, as of fw version 2019-02-04
> СС2530 can be re-paired if you power on/power off it three times (power on, wait 2 seconds, power off, repeat this cycle three times).

vhristev, 2019-10-16:
Devices:

Old Pi3 + HassOS + HassIO + CC2531 without antenna ( coordinator)
New Pi4 + Rasberian + HassIO + CC2531 WITH antenna (coordinator)
Zigbee Router ( CC2530 + CC2591 )

Hello, guys recently I move from Pi3-CC2531 to Pi4 - CC2531- with an antenna. With Pi3 I configure CC2530 + CC2591 based on documentation and when I plug it into the wall socket after few minutes it appear.

The problem is when i move to new Pi4 with new CC2531 with antenna I cannot see CC2530 + CC2591 ( router )

I check the voltage and I can see that CC2530 + CC2591 board get 5V which means power is fine

Any idea if I need to re-flash it when I move from one coordinator to another. I’m not sure if there is some limitation to which coordinator to join

I also don’t know how to check if CC2530 + CC2591 is working fine because there are no LED indicators … nothing i test it with a multimeter. 

Any ideas on how to troubleshoot. Im not aware how to sniff Zigbee network to check if CC2530 + CC2591 is there. I'm not sure if a router joins coordinator network it uses just that network and cannot join others.

Tried re-paring with power on/off 3 times based on documentation.