Koenkk/zigbee2mqtt #552: zigbee2mqtt hangs

borpin, 2018-10-30:
Device IKEA TRADFRI LED bulb

Bug report
- First make sure your on the latest version of zigbee2mqtt. **yes**
- Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware. **yes - flashed this week**
- Provide a clear description of the problem.

Installed on a RPi Zero - some bluetooth activity.

Starting zigbee2mqtt as a service.  One device connected (although it says 2 for some reason - no other zigbee devices in house). Able to switch light on / off a couple of times via the `set` MQTT topic then bridge hangs and eventually reports `zigbee2mqtt/bridge/state offline` (nothing in log).

Process `node index.js` goes to 100%

Restart service and repeat.

- Provide a log with `log_level` set to `debug`, see 
https://gist.github.com/borpin/a23fbea03d4130ac14ded5e09b2b63ce

MQTT activity during period of log - note last 3 set commands unreported.

```
zigbee2mqtt/bridge/state online
zigbee2mqtt/Standing Lamp {"state":"on"}
zigbee2mqtt/Standing Lamp {"state":"on"}
zigbee2mqtt/Standing Lamp/set {"state": "off"}
zigbee2mqtt/Standing Lamp {"state":"off"}
zigbee2mqtt/Standing Lamp {"state":"OFF"}
zigbee2mqtt/Standing Lamp/set {"state": "on"}
zigbee2mqtt/Standing Lamp {"state":"on"}
zigbee2mqtt/Standing Lamp {"state":"ON"}
zigbee2mqtt/Standing Lamp {"state":"ON"}
zigbee2mqtt/Standing Lamp/set {"state": "off"}
zigbee2mqtt/Standing Lamp {"state":"off"}
zigbee2mqtt/Standing Lamp {"state":"OFF"}
zigbee2mqtt/Standing Lamp/set {"state": "on"}
zigbee2mqtt/Standing Lamp {"state":"on"}
zigbee2mqtt/Standing Lamp {"state":"ON"}
zigbee2mqtt/Standing Lamp/set {"state": "off"}
zigbee2mqtt/Standing Lamp/set {"state": "off"}
zigbee2mqtt/Standing Lamp/set {"state": "off"}
zigbee2mqtt/bridge/state offline
```

Koenkk, 2018-10-31:
Can you try if this issue also happens when running on a different device?

borpin, 2018-10-31:
Hi, not really.  It is the only RPiW I have.

Koenkk, 2018-10-31:
Can you provide the log when running with `DEBUG=* npm start`?

borpin, 2018-10-31:
https://gist.github.com/borpin/cebcfe3bfea3c8dd8987f8dc226fc086

MQTT messages published from another device and the published messages published from a third machine.

Note log stops after last mqtt message is sent by zigbee2mqtt.

Found this in my mqtt log (on a different run)
```
1541012406: Client mqttjs_2869dda5 has exceeded timeout, disconnecting.
1541012406: Socket error on client mqttjs_2869dda5, disconnecting.
```

Up till then there were PINGREQ but difficult to tell properly.

```
zigbee2mqtt/bridge/state online
zigbee2mqtt/Standing Lamp {"state":"on"}
zigbee2mqtt/Standing Lamp {"state":"on"}
zigbee2mqtt/Standing Lamp/set {"state":"off"}
zigbee2mqtt/Standing Lamp {"state":"off"}
zigbee2mqtt/Standing Lamp {"state":"OFF"}
zigbee2mqtt/Standing Lamp {"state":"OFF"}
zigbee2mqtt/Standing Lamp/set {"state":"on"}
zigbee2mqtt/Standing Lamp {"state":"on"}
zigbee2mqtt/Standing Lamp {"state":"ON"}
zigbee2mqtt/Standing Lamp {"state":"ON"}
zigbee2mqtt/Standing Lamp/set {"state":"off"}
zigbee2mqtt/Standing Lamp {"state":"off"}
zigbee2mqtt/Standing Lamp {"state":"OFF"}
zigbee2mqtt/Standing Lamp {"state":"OFF"}
zigbee2mqtt/Standing Lamp/set {"state":"on"}
zigbee2mqtt/Standing Lamp {"state":"on"}
zigbee2mqtt/Standing Lamp {"state":"ON"}
zigbee2mqtt/Standing Lamp {"state":"ON"}
zigbee2mqtt/Standing Lamp/set {"state":"off"}
zigbee2mqtt/Standing Lamp {"state":"off"}
zigbee2mqtt/Standing Lamp {"state":"OFF"}
zigbee2mqtt/Standing Lamp {"state":"OFF"}
zigbee2mqtt/Standing Lamp/set {"state":"on"}
zigbee2mqtt/Standing Lamp {"state":"on"}
zigbee2mqtt/Standing Lamp {"state":"ON"}
zigbee2mqtt/Standing Lamp {"state":"ON"}
zigbee2mqtt/Standing Lamp/set {"state":"off"}
zigbee2mqtt/Standing Lamp/set {"state":"off"}
zigbee2mqtt/Standing Lamp/set {"state":"off"}
zigbee2mqtt/bridge/state offline
```

borpin, 2018-10-31:
```
node -v
v11.0.0
```

Koenkk, 2018-11-01:
Can you try with nodejs 8? That is the recommended version: https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge#2-installing

borpin, 2018-11-02:
> Can you try with nodejs 8? 

Hi @Koenkk, I can confirm it works with node V11. I have edited the title appropriately.

Cheers

sehraf, 2018-11-03:
Hi, i can confirm the described behavior. Running v11 and process hangs at ~100% CPU shortly after starting.
```
node -v       
v11.0.0
```

Koenkk, 2018-11-03:
@borpin thanks!

aktentasche, 2018-11-04:
I just did my first install using node 11 on an Orange Pi PC running armbain mainline and I cannot confirm. If I can be of help let me know.

sehraf, 2018-11-04:
@aktentasche v11 or v11.1?
@borpin can you try out version v11.1?
I've played around with it yesterday and it seems to be running fine now (it still rarely hangs sometimes though).

aktentasche, 2018-11-04:
@sehraf node v11.1.0

alex-voigt, 2018-11-04:
I also had problems with Osram smart plugs using version v11.0. After updating to v11.1.0 it seems to work fine so far

xoxys, 2018-11-05:
> Can you try with nodejs 8? That is the recommended version: https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge#2-installing

@Koenkk Really? WIth nodejs 8.12 i get:
```
3652 verbose stack Error: serialport@6.2.2 install: `prebuild-install || node-gyp rebuild`
3652 verbose stack Exit status 1
3652 verbose stack     at EventEmitter.<anonymous> (/opt/nodejs/node-v8.12.0-linux-armv6l/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
3652 verbose stack     at emitTwo (events.js:126:13)
3652 verbose stack     at EventEmitter.emit (events.js:214:7)
3652 verbose stack     at ChildProcess.<anonymous> (/opt/nodejs/node-v8.12.0-linux-armv6l/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
3652 verbose stack     at emitTwo (events.js:126:13)
3652 verbose stack     at ChildProcess.emit (events.js:214:7)
3652 verbose stack     at maybeClose (internal/child_process.js:915:16)
3652 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
3653 verbose pkgid serialport@6.2.2
3654 verbose cwd /opt/zigbee2mqtt
3655 verbose Linux 4.14.27-v7.1.el7
3656 verbose argv "/opt/nodejs/node-v8.12.0-linux-armv6l/bin/node" "/opt/nodejs/latest/bin/npm" "install"
3657 verbose node v8.12.0
3658 verbose npm  v6.4.1
3659 error code ELIFECYCLE
3660 error errno 1
3661 error serialport@6.2.2 install: `prebuild-install || node-gyp rebuild`
3661 error Exit status 1
3662 error Failed at the serialport@6.2.2 install script.
3662 error This is probably not a problem with npm. There is likely additional logging output above.
3663 verbose exit [ 1, true ]
```

Koenkk, 2018-11-06:
@xoxys what system are you running it on? What happens after `rm -rf node_modules && npm install`.

xoxys, 2018-11-06:
@Koenkk CentOS7 ARM on a RPi. I have updated my comment. It works now after running npm install as non-root user... Seems to be a requirement of serialport

danielo515, 2020-07-30:
I am on the same situation. This was worning fine before, but now this is what I'm getting: 

```

Zigbee2mqtt requires node version ^10 || ^12 || ^13 || ^14, you are running v11.5.0!
```

The only thing that changed was a reboot. If it works with node 10 and 11 why will it not work with 11?

danielo515, 2020-07-30:
Just added node 11 to the list of engines on package.json, fixed some problems with the log permissions files (the update script seems to make them root owned) and now it is working again, so seems that node 11 it's indeed supported.

Koenkk, 2020-07-30:
@danielo515 Zigbee2MQTT only supports supported node versions. Node 11 is not supported anymore and therefore not supported by Zigbee2MQTT. This however doesn't mean that it won't work. The message you seeing is just a warning.

[image]

danielo515, 2020-07-30:
> @danielo515 Zigbee2MQTT only supports supported node versions. Node 11 is not supported anymore and therefore not supported by Zigbee2MQTT. This however doesn't mean that it won't work. The message you seeing is just a warning.
> 
> [image]
> 

Well, if it is just a warning then the problem was the log folder permissions. Maybe it was failing because the folder thing but systemd was only showing up to that warning.