Koenkk/zigbee2mqtt #553: IKEA TRADFRI wireless dimmer (ICTC-G-1) No converter available

salopette, 2018-10-31:
`-- Logs begin at Wed 2018-10-31 15:13:28 CET. --
Okt 31 16:08:43 raspberrypi npm[577]:   zclMsg:
Okt 31 16:08:43 raspberrypi npm[577]:    { frameCntl: { frameType: 1, manufSpec: 0, direction: 0, disDefaultRsp: 1 },
Okt 31 16:08:43 raspberrypi npm[577]:      manufCode: 0,
Okt 31 16:08:43 raspberrypi npm[577]:      seqNum: 48,
Okt 31 16:08:43 raspberrypi npm[577]:      cmdId: 'move',
Okt 31 16:08:43 raspberrypi npm[577]:      payload: { movemode: 1, rate: 195 } } }
Okt 31 16:08:43 raspberrypi npm[577]:   zigbee2mqtt:warn 2018-10-31 16:08:43 No converter available for 'ICTC-G-1' with cid 'genLevelCtrl', type 'cmdMove' and data '{"cid":"genLevelCtrl","data":{"movemode":1,"rate":195}}'
Okt 31 16:08:43 raspberrypi npm[577]:   zigbee2mqtt:warn 2018-10-31 16:08:43 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
Okt 31 16:08:43 raspberrypi npm[577]:   zigbee2mqtt:info 2018-10-31 16:08:43 MQTT publish, topic: 'zigbee2mqtt/Ikea Dimmer', payload: '{"brightness":152,"device":{"ieeeAddr":"0x000b57fffe2ba3da","friendlyName":"Ikea Dimmer","type":"EndDevice","nwkAddr":53963,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI wireless dimmer","status":"offline"}}'
Okt 31 16:08:44 raspberrypi npm[577]:   zigbee2mqtt:info 2018-10-31 16:08:44 MQTT publish, topic: 'zigbee2mqtt/Ikea Dimmer', payload: '{"brightness":121,"device":{"ieeeAddr":"0x000b57fffe2ba3da","friendlyName":"Ikea Dimmer","type":"EndDevice","nwkAddr":53963,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI wireless dimmer","status":"offline"}}'
Okt 31 16:08:56 raspberrypi npm[577]: { groupid: 0,
Okt 31 16:08:56 raspberrypi npm[577]:   clusterid: 8,
Okt 31 16:08:56 raspberrypi npm[577]:   srcaddr: 53963,
Okt 31 16:08:56 raspberrypi npm[577]:   srcendpoint: 1,
Okt 31 16:08:56 raspberrypi npm[577]:   dstendpoint: 1,
Okt 31 16:08:56 raspberrypi npm[577]:   wasbroadcast: 0,
Okt 31 16:08:56 raspberrypi npm[577]:   linkquality: 31,
Okt 31 16:08:56 raspberrypi npm[577]:   securityuse: 0,
Okt 31 16:08:56 raspberrypi npm[577]:   timestamp: 10179774,
Okt 31 16:08:56 raspberrypi npm[577]:   transseqnumber: 0,
Okt 31 16:08:56 raspberrypi npm[577]:   len: 5,
Okt 31 16:08:56 raspberrypi npm[577]:   data: { '0': 17, '1': 50, '2': 5, '3': 0, '4': 195 },
Okt 31 16:08:56 raspberrypi npm[577]:   zclMsg:
Okt 31 16:08:56 raspberrypi npm[577]:    { frameCntl: { frameType: 1, manufSpec: 0, direction: 0, disDefaultRsp: 1 },
Okt 31 16:08:56 raspberrypi npm[577]:      manufCode: 0,
Okt 31 16:08:56 raspberrypi npm[577]:      seqNum: 50,
Okt 31 16:08:56 raspberrypi npm[577]:      cmdId: 'moveWithOnOff',
Okt 31 16:08:56 raspberrypi npm[577]:      payload: { movemode: 0, rate: 195 } } }
Okt 31 16:08:56 raspberrypi npm[577]: cmd { groupid: 0,
Okt 31 16:08:56 raspberrypi npm[577]:   clusterid: 8,
Okt 31 16:08:56 raspberrypi npm[577]:   srcaddr: 53963,
Okt 31 16:08:56 raspberrypi npm[577]:   srcendpoint: 1,
Okt 31 16:08:56 raspberrypi npm[577]:   dstendpoint: 1,
Okt 31 16:08:56 raspberrypi npm[577]:   wasbroadcast: 0,
Okt 31 16:08:56 raspberrypi npm[577]:   linkquality: 31,
Okt 31 16:08:56 raspberrypi npm[577]:   securityuse: 0,
Okt 31 16:08:56 raspberrypi npm[577]:   timestamp: 10179774,
Okt 31 16:08:56 raspberrypi npm[577]:   transseqnumber: 0,
Okt 31 16:08:56 raspberrypi npm[577]:   len: 5,
Okt 31 16:08:56 raspberrypi npm[577]:   data: { '0': 17, '1': 50, '2': 5, '3': 0, '4': 195 },
Okt 31 16:08:56 raspberrypi npm[577]:   zclMsg:
Okt 31 16:08:56 raspberrypi npm[577]:    { frameCntl: { frameType: 1, manufSpec: 0, direction: 0, disDefaultRsp: 1 },
Okt 31 16:08:56 raspberrypi npm[577]:      manufCode: 0,
Okt 31 16:08:56 raspberrypi npm[577]:      seqNum: 50,
Okt 31 16:08:56 raspberrypi npm[577]:      cmdId: 'moveWithOnOff',
Okt 31 16:08:56 raspberrypi npm[577]:      payload: { movemode: 0, rate: 195 } } }
Okt 31 16:08:56 raspberrypi npm[577]:   zigbee2mqtt:warn 2018-10-31 16:08:56 No converter available for 'ICTC-G-1' with cid 'genLevelCtrl', type 'cmdMoveWithOnOff' and data '{"cid":"genLevelCtrl","data":{"movemode":0,"rate":195}}'
Okt 31 16:08:56 raspberrypi npm[577]:   zigbee2mqtt:warn 2018-10-31 16:08:56 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
Okt 31 16:08:56 raspberrypi npm[577]:   zigbee2mqtt:info 2018-10-31 16:08:56 MQTT publish, topic: 'zigbee2mqtt/Ikea Dimmer', payload: '{"brightness":121,"device":{"ieeeAddr":"0x000b57fffe2ba3da","friendlyName":"Ikea Dimmer","type":"EndDevice","nwkAddr":53963,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI wireless dimmer","status":"offline"}}'
Okt 31 16:08:56 raspberrypi npm[577]:   zigbee2mqtt:info 2018-10-31 16:08:56 MQTT publish, topic: 'zigbee2mqtt/Ikea Dimmer', payload: '{"brightness":163,"device":{"ieeeAddr":"0x000b57fffe2ba3da","friendlyName":"Ikea Dimmer","type":"EndDevice","nwkAddr":53963,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI wireless dimmer","status":"offline"}}'
`

Chrischi-, 2018-11-14:
This is a supported device, please update and paste ([pastebin.com](pastebin.com)) a debug log after trying it again.

mattydanby, 2018-11-16:
Hi,

I'm having the same errors reported:
- Device: Raspberry Pi 3B+
- Hassio: 0.81.6
- Hass.io supervisor: 139
- HassOS: 1.12
- zigbee2mqtt: 0.1.8

Is this what you need?

[https://pastebin.com/LxsHNz1E](https://pastebin.com/LxsHNz1E)

The above is for:
1. - slow turn clockwise
2. - slow turn counter-clockwise
3. - quick short turn clockwise
4. - quick short turn counter-clockwise

Let me know if I can provide any more info

Thanks

salopette, 2018-11-16:
https://pastebin.com/0zmKp7Pz

salopette, 2018-11-17:
How can I switch to DEV?

cd / opt / zigbee2mqtt
git fetch
git checkout dev
npm install

get errors:

Abbruch

p01c, 2018-11-19:
# Stop zigbee2mqtt and go to directory
sudo systemctl stop zigbee2mqtt
cd /opt/zigbee2mqtt
# Backup configuration
**cp -R data data-backup**
# Revert changes
**git checkout data/configuration.yaml**
# Update
git checkout dev
git pull
rm -rf node_modules
npm install
# Restore configuration
cp -R data-backup/* data
rm -rf data-backup
# Start zigbee2mqtt
sudo systemctl start zigbee2mqtt