Koenkk/zigbee2mqtt #554: Problen with start zigbee2mqtt

salopette, 2018-10-31:
Hi folks, my zigbee2mqtt has suddenly stopped working?
Can anyone help me, I have already restarted Pi, but it does not help.

`pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@0.1.6 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 2018-10-31 17:52:27 Logging to directory: 'data/log/2018-10-31.17-52-27'
  zigbee2mqtt:info 2018-10-31 17:52:28 Starting zigbee2mqtt version 0.1.6 (commit #98302be)
  zigbee2mqtt:info 2018-10-31 17:52:28 Starting zigbee-shepherd
/opt/zigbee2mqtt/node_modules/nedb/lib/datastore.js:77
    if (err) { throw err; }
               ^

Error: Can't insert key 10, it violates the unique constraint
    at _AVLTree.insert (/opt/zigbee2mqtt/node_modules/binary-search-tree/lib/avltree.js:273:19)
    at AVLTree.insert (/opt/zigbee2mqtt/node_modules/binary-search-tree/lib/avltree.js:307:27)
    at Index.insert (/opt/zigbee2mqtt/node_modules/nedb/lib/indexes.js:77:15)
    at Index.insertMultipleDocs (/opt/zigbee2mqtt/node_modules/nedb/lib/indexes.js:114:12)
    at Index.insert (/opt/zigbee2mqtt/node_modules/nedb/lib/indexes.js:69:33)
    at Index.reset (/opt/zigbee2mqtt/node_modules/nedb/lib/indexes.js:55:23)
    at /opt/zigbee2mqtt/node_modules/nedb/lib/datastore.js:107:21
    at Array.forEach (<anonymous>)
    at Datastore.resetIndexes (/opt/zigbee2mqtt/node_modules/nedb/lib/datastore.js:106:29)
    at /opt/zigbee2mqtt/node_modules/nedb/lib/persistence.js:293:23
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.1.6 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@0.1.6 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2018-10-31T16_52_28_870Z-debug.log
pi@raspberrypi:/opt/zigbee2mqtt $
`

Koenkk, 2018-11-01:
Did you manually edit your `database.db`?

salopette, 2018-11-01:
Yes i have.
but now I have everything reinstalled and it works!

What's the problem with the DB?

Koenkk, 2018-11-01:
If the problem is solved, nvm.

closed 2018-11-01 by Koenkk