Koenkk/zigbee2mqtt #557: Philips 6916531PH Living Whites

arteck, 2018-11-02:
any ideas ??

WagnerBeccard, 2018-11-02:
Wich Ideas? Is it supported? 
https://github.com/Koenkk/zigbee2mqtt/wiki/Supported-devices#philips - Nope 
if there ll be support? 
https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices - maybe. After Its Philipps it's likely they used the same protocol.
From some Amazon ratings you need to use Touchlink to pair the device.

arteck, 2018-11-02:
the problem is this Plug can not connected direct to the Bridge.
Touchlink  is here the magic word... ok

vladitep, 2018-11-29:
it can connected direct to bridge. To pair it with bridge you need to put 6916531PH  close to bridge, just put it on the bridge and it connects. In old bridge firmware it does. I have one of this and its work ok couple years, when i got it and connect to bridge. But now, when i removed it from bridge, i cant connect it again.

karolzlot, 2021-07-04:
@Koenkk  can you reopen?

I have this device: 6916531PH

It has no buttons, so I think it should be possible to link it by Touchlink. I can help with logs, but I am not sure where to start.

I tried to put it near zigbee2mqtt cooridinator and permit, but nothing happens.

Koenkk, 2021-07-05:
Try putting the device close to the coordinator (almost against it), and press the scan button on the touchlink page of the z2m frontend: 
[image]

If it detects it click the factory reset button

karolzlot, 2021-07-05:
@Koenkk 

I tried a few times. Usually detected nothing.
Once it detected one of my already connected devices. (Osram plug).

[image]

Koenkk, 2021-07-05:
That's strange, is it really Zigbee?

karolzlot, 2021-07-05:
More information:

> Hi, LivingWhite products can't be paired to deCONZ since these use a security key only known and supported by Philips Hue bridge. They predate Hue products which use inter-operable security keys.

https://github.com/ebaauw/homebridge-hue/issues/291#issuecomment-366313967

---

Manual:

[image]
[image]
[image]
[image]

Koenkk, 2021-07-05:
> LivingWhite products can't be paired to deCONZ since these use a security key only known and supported by Philips Hue bridge. They predate Hue products which use inter-operable security keys.

This explains why this won't work.

closed 2021-07-13 by github-actions[bot]