Koenkk/zigbee2mqtt #558: How to replace coordinator (CC2531) with  CC2530+CC2591 within existing Zigbee network?

blauhorn, 2018-11-03:
Hello everybody and thank you very much for this great project.

I am currently using the cc2531 as coordinator and several sensors and actuators with zigbee2mqtt via the FHEM module on a bananapi.
 However, I am not quite satisfied with the range of the cc2531, and now I would like to replace the coordinator to a cc2530+cc2591, I have the hardware here, and also know how to flash the coordinator firmware.
But how can I easily replace the coordinator in the network without having to re-pair all enddevices?
Can someone kindly give me a hint here?

Thank you for any ideas
blauhorn

Koenkk, 2018-11-03:
At the moment this is not possible, you have to repair everything.

lolorc, 2018-11-03:
it is possible but it's tricky, make sure the panid & channel stay the same, Once flashed shield the cc2530 so that it can't detect your devices/router using those parameters (unplug routers) when you plug it.
https://github.com/Koenkk/zigbee2mqtt/issues/282#issuecomment-426207544

blauhorn, 2018-11-05:
Thank you very much for the answers. I don't think I can do the tricky trick, so now I have some work to do to pair all the devices again.

blauhorn

closed 2018-11-05 by blauhorn

VivantSenior, 2019-06-30:
[Now it's possible](https://www.zigbee2mqtt.io/information/what_does_and_doesnt_require_repairing.html). Just change "serial_port" to appropriate value (`ls -l /dev/serial/by-id`) and restart RPi (`sudo reboot`).