Koenkk/zigbee2mqtt #559: The parameter 'modelld' is missing

loglux, 2018-11-03:
I have different versions of Aqara Square Mini Switch that can be recognisable and distingish by modelld (they have the same look but the different modelld and a set of functions)

The problem is not possible to distinguish different devices because the parameter modelld is missing from MQTT messages. All of them have a similar parameter model: "WXKG11LM". However, all of them completely different devices. 

Nov 03 15:28:19 housespirit npm[9560]:   zigbee2mqtt:info 2018-11-3 15:28:19 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"devices","message":

{"ieeeAddr":"0x00158d0001e89451","type":"EndDevice",**"model":"WXKG11LM"**,"friendly_name":"0x00158d0001e89451"}]}'

**zigbee2mqtt/data/database.db**, however, has a correct reference to the modelld:

{"id":6,"type":"EndDevice","ieeeAddr":"0x00158d0001e89451","nwkAddr":57,"manufId":4151,"epList":[1],"status":"online","joinTime":1541104864,"endpoints":{"1":{"profId":260,"epId":1,"devId":259,"inClusterList":[0,3,18],"outClusterList":[0$
{"id":6,"type":"EndDevice","ieeeAddr":"0x00158d0001e89451","nwkAddr":57,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.remote.b1acn01\u0000\u0000\u0000\u0000\u0000\u0000","epList":[1],"status":"online","joinTi$
{"id":6,"type":"EndDevice","ieeeAddr":"0x00158d0001e89451","nwkAddr":57,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.remote.b1acn01\u0000\u0000\u0000\u0000\u0000\u0000","epList":[1],"status":"online","joinTi$

loglux, 2018-11-05:
In my case, the parament modelId is present in database.db. 
The problem is that zigbee2mqtt doesn't relay on modelId, using the parameter 'model' instead that leads to confusion because the one model can have different modelId with different functionality. 
I have two different Aqara Square mini switches that have the same model - WXKG11LM but different modelId and different functionality.

Koenkk, 2018-11-05:
Why do you want to distinguish devices?

loglux, 2018-11-05:
It demanded to distinguish one version from another.
For example, I have two similar Aqara Square mini switches, both of them are positioned as a model WXKG11LM
This is the version **sensor_switch.aq2**
[image]
And this is the version **lumi.remote.b1acn01**
[image]
There is also another version that hasn't Long Click and Press Release but uses Triple Click and Quadruple Click instead. 
Therefore the modelId is needed to establish a correct device in HA.

closed 2019-01-12 by stale[bot]