Koenkk/zigbee2mqtt #560: Hue Lightstrips always report being offline despite being listed as routers

ben423423n32j14e, 2018-11-03:
An interesting observation regarding Hue Lightstrips v2

They function fine but are always reported as offline in Zigbee2MQTT despite being listed as a router.

See Graphviz screenshot below, in the center is a Xiaomi power plug and either side is Hue Lightstrips

[image]

davekr8, 2018-11-04:
I see the same issue with my osram plugs. I think it's a generic issue with the online/offline state.

Additionally, I have seen devices with no edge to an upstream router/coordinator - but being fully functional an present in my mesh. The network layout seems to be misleading to some extend.

ben423423n32j14e, 2018-11-16:
After I reflashed my CC2531 and created a new network, I notice that Hue Lightstrips are acting as router and listed as online. Closing ticket.

closed 2018-11-16 by ben423423n32j14e