Koenkk/zigbee2mqtt #562: Problem with devices which contain "/"

berrywhite96, 2018-11-03:
Named a device with an "/", but zigbee2mqtt cant set new settings with a mqtt msg. So maybe you give the user a warning that you cant rename a device with "/" in the friendly name or you fix this problem :D.

My log (I know I should have post it on pastebin, but are just two lines):
`zigbee2mqtt:info 2018-11-4 00:40:59 MQTT publish, topic: 'zigbee2mqtt/room_1/hue_go_1', payload: '{"state":"OFF","brightness":254,"color_temp":153,"color":{"x":0.156,"y":0.142}}'
  zigbee2mqtt:error 2018-11-4 00:42:16 Cannot handle 'zigbee2mqtt/room_1/hue_go_1/set' because deviceID of 'room_1' cannot be found`

I hope I can find all bugs and help this project, really like this project very good work!

Koenkk, 2018-11-04:
Which version are you running?

berrywhite96, 2018-11-04:
Installed it yesterday step by step with the wiki, so I have the clone from the master.
Btw. I changed the friendly name with the configuration.yaml.

dirstel, 2018-11-05:
Currently nested Topics are not supported:
Duplicate of #470

Koenkk, 2018-11-07:
As @dirstel mentioned, duplicate of #470

closed 2018-11-07 by Koenkk