Koenkk/zigbee2mqtt #563: homeassistant link quality with WXKG11LM

dmwyatt, 2018-11-04:
My Xiaomi WXKG11LM reports a `linkquality` attribute which is currently not supported in `homeassistant.js` where it's configured to use `configurations.sensor_click`.

```{"battery":"100.00","voltage":3165,"linkquality":26,"click":"single"}```

I'm getting ready to add the linkquality attribute to MQTT discovery for this device.  I was just going to add it to `configurations.sensor_click`, but I'm not sure if that is supposed to be more generic or if there's some reason `linkquality` isn't in there already?

Koenkk, 2018-11-05:
`linkquality` should be added to all `json_attributes`. Can you make a pr?

dmwyatt, 2018-11-05:
Yep, I'll get one out today.

dmwyatt, 2018-11-05:
Done, see PR #566

closed 2018-11-05 by dmwyatt