Koenkk/zigbee2mqtt #564: Zigbee Smart Meters?

joshdinsdale, 2018-11-05:
Hi Guys

Wondering if anyone has had any like using z2mqtt with smart meters? I am having one fitted soon and it uses zigbee to communicate with the in home display screen, it would be great if i could pipe the data directly into my HA system, and ideally use the display as well, i dont know if that would be possible?

Apparently the system is compatible with Smart Energy Profile (SEP 1.1) if that helps?

closed 2018-11-05 by joshdinsdale

bachoo786, 2019-06-12:
any luck on this @joshdinsdale ?

joshdinsdale, 2019-06-12:
Unfortunately it's a huge 'No' with a side-order of 'aint gonna happen', not in the UK at least anyway. 
Firstly, the smart meter acts as a coordinator on its own ZigBee network (HAN), obviously you can't have two coordinators on the same network.
Secondly even if we could somehow find a way around the above issue, the security around smart metering means that devices have to be certified in order to be allowed to access the HAN. Only the energy supplier can access the meter to enable pairing of devices.

There is talk of CAD devices that will be allowed to connect to the HAN and then to your WiFi in order to send the meter data to other services, but there aren't any that are easily available that are certified to use in the UK.

ccorderod, 2019-06-12:
@joshdinsdale @bachoo786 I am using at home meazon meters (dinrail and plugs) - meazon.com. I pushed the definition for them a few weeks ago and they work well. I also have a smartmeter from my energy provider, but it is fully locked.
Anyway, it doesn't matter. One of my dinrail meters is reading the main, so I get the same data than the smartmeter.

joshdinsdale, 2019-06-12:
@ccorderod where did you buy the meazon and how does it work, do you need to do any wiring to get it in line with your mains?

ccorderod, 2019-06-13:
@joshdinsdale I got my meters from Meazon directly, I may pm you a few contacts if needed. Dinrail meters DO need wiring (line and neutral); not an issue as they have to be on electric closet, then a magnetic clamp monitor electric consumption on the line where you have the clamp. I may send a pic if you want of my setup.
Bizy plugs do not need wiring as they get powered when plugged on the electric socket.

jd1900, 2020-09-11:
> @joshdinsdale I got my meters from Meazon directly, I may pm you a few contacts if needed. Dinrail meters DO need wiring (line and neutral); not an issue as they have to be on electric closet, then [...]

could you send me those contact?
I'd like to buy a meazon din rail on/off device (https://www.zigbee2mqtt.io/devices/MEAZON_DINRAIL.html)