Koenkk/zigbee2mqtt #568: Incorrect Battery Readings

RodBr, 2018-11-06:
I'm getting erratic readings on Xiaomi battery readings.  The Mi Home app gives readings of 16% (I think, I've now shut down the hub and moved everything over) for the two sensors below.  
Both of these are 'WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor', but it seems to be happening on all sensors.

One is reporting back 99% while the other is reporting back 27%. 

```
  zigbee2mqtt:info ... payload: '{"temperature":26.78,"linkquality":0,"humidity":61.01,"battery":"99.00","voltage":3005}'
  zigbee2mqtt:info ..., payload: '{"temperature":28.16,"linkquality":31,"humidity":57.57,"battery":"27.00","voltage":2955}'
```

subzero79, 2018-11-06:
The xiaomi devices report their battery level in voltage, based on that the level is calculated according to this array

https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L14

I don't recall the mi home app reporting the devices batteries.

RodBr, 2018-11-06:
After a bit of troubleshooting it appears that as soon as linkquality drops to 0 the voltage defaults to 3005 even though valid temp and humidity readings are coming through.  Move the sensor closer and the voltage readings work again.

marcodlima, 2018-11-11:
I got 2 aqara temp/hum sensors and both in less than 5 days dropped to 31% battery. One is near the sniffer and the other is on another room. If this readings are correct then each sensor lasts 1 week!

Koenkk, 2018-11-12:
@marcodlima what's the voltage?

nerdosity, 2018-11-16:
I don't have battery reading in door sensors.
Two Mijia, 4 Aqara.
I had battery reading after some hours for PIR, temp/humidity sensors (still from Xiaomi)

JoschaMiddendorf, 2019-01-02:
> Ok, don't be worried that the sensors only last a few days (should be 1 year+).
> 
> The percentages are not calibrated correctly because we don't know at which voltage the sensors stops working. Voltage comes from the device, based on that zigbee2mqtt calculates a percentage.

Wouldn't it be better to calculate the battery reading in a more optimistic manner?

JoschaMiddendorf, 2019-01-02:
> Ok, don't be worried that the sensors only last a few days (should be 1 year+).
> 
> The percentages are not calibrated correctly because we don't know at which voltage the sensors stops working. Voltage comes from the device, based on that zigbee2mqtt calculates a percentage.

Wouldn't it be better to calculate the battery reading in a more optimistic manner, until we know the lowest possible voltage, than to make everybody replacing their half used batteries?
I am currently calculating an actual percentage of a CR2032 battery discharge curve.

```
if ($voltage >= 3000) {
		$batteryLevel = 100;
	}
	elsif ($voltage > 2900)  {

		$batteryLevel = 100 - ((3000 - $voltage) * 58) / 100;
	}
	elsif ($voltage > 2740)  {

		$batteryLevel = 42 - ((2900 - $voltage) * 24) / 160;
	}
	elsif ($voltage > 2440)  {

		$batteryLevel = 18 - ((2740 - $voltage) * 12) / 300;
	}
	elsif ($voltage > 2100) {

		$batteryLevel = 6 - ((2440 - $voltage) * 6) / 340;
	}
	else {

		$batteryLevel = 0;
	};
```
Would be an honor to me to have that integrated in your code.

Koenkk, 2019-01-02:
that indeed looks much better, could you make a pr? https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L148

JoschaMiddendorf, 2019-01-02:
Have a look on PR #190, its my first try on JS so don't be to hard with me and feel free to apply corrections.

JoschaMiddendorf, 2019-01-02:
Have a look on PR #191, my first attempt was a formatting disaster. ;-)

closed 2019-03-10 by stale[bot]