Koenkk/zigbee2mqtt #570: Unable to start zigbee2mqtt after update

tb205gti, 2018-11-06:
Just updated to 0.1.8 (https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge#6-for-later-update-zigbee2mqtt-to-the-latest-version) - just like I have done multiple times before. Made a backup of the data folder, update from git, delete the node_modules, npm install, restore data folder. Just like normal.

But now I'm unable to start zigbee2mqtt. I get the following error:

```
/opt/zigbee2mqtt# npm start

> zigbee2mqtt@0.1.8 start /opt/zigbee2mqtt
> node index.js

/opt/zigbee2mqtt/lib/controller.js:553
let messagePayload = {...payload};
                      ^^^

SyntaxError: Unexpected token ...
    at createScript (vm.js:74:10)
    at Object.runInThisContext (vm.js:116:10)
    at Module._compile (module.js:533:28)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:503:32)
    at tryModuleLoad (module.js:466:12)
    at Function.Module._load (module.js:458:3)
    at Module.require (module.js:513:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/index.js:1:82)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.1.8 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@0.1.8 start script.
```

What the heck - I do not the that line 553 has been changed in any recent commits - so why is it suddenly failing??

I even tried to start all over again with a completely fresh checkout - but I get the same error.

Node v8.0.0
npm v6.4.1
Ubuntu 16.04.5 LTS

tb205gti, 2018-11-06:
Tried to revert to 0.1.7 - not working, reverting to 0.1.6 - wohhoo it works.

..Now why doesn't 0.1.8 work anymore? Am I doing something completely wrong?

Koenkk, 2018-11-06:
What is the output of `node --version`.

tb205gti, 2018-11-06:
like above: 
v8.0.0

I found out that it stops working from commit: 03d913defc86bcb30647a4d994da4da2a78d8658

Koenkk, 2018-11-06:
Is it fixed when using node v8.12?

tb205gti, 2018-11-06:
Just tried - nope same issue.

closed 2018-11-06 by tb205gti