Koenkk/zigbee2mqtt #572: Error RPI bridge with CC2530 as connector

HouseFox, 2018-11-07:
Hello community,

at first i have to say that i´m not so familia with this because i´m a rookey.

In the past i got into trouble to Flash the cc2531 with the ccdebugger, like in the documentation of https://github.com/Koenkk/zigbee2mqtt/wiki/Getting-started. 
The result was every time, that i got no connetion (green lamp) between ccdebugger, adapter and cc2531. And maybe, i don´t know why and how but i damaged the cc2531 usb stick. Not one, two of them. No green light anymore at cc2531.

But i have also by my side one of cc2530 with antenna. 
So my thougths are going to this direction. 
To use this with a FTDI at my raspberry pi 3+.

I used for flashing the code from https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/CC2530/bin --> CC2530ZNP-Prod.hex.
Flashing result with SmartRF Flash Programmer looked good.

So i go forward with the instuctions of https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge and when "Koenkk" speak about cc2531 i used cc2530.

When i starting zigbee2mqtt i get into error and i don´t know why....

Can anybody help me please, what is wrong...
Is the flash of cc2530 wrong?
Is some installation wrong?

I do not know how to continue.
Please see my attached files.

Thanks to you all!

[image]
[image]
[image]
[Uploading whatihavedone.txt…]()
[image]

hamstie, 2018-11-15:
i think it isn't a communication problem, i think this is a node.js issue
- check if the node.js is running (try a hello word.js)
- check the version of node.js
- check version of npm

HouseFox, 2018-11-16:
@hamstie what do you mean with "check if the node.js is running (try a hello word.js)"
how i can check this?

Using username "pi".
Linux raspberrypi 4.14.71-v7+ #1145 SMP Fri Sep 21 15:38:35 BST 2018 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Nov 12 22:11:23 2018 from 192.168.2.102
pi@raspberrypi:~ $ node --version
v8.12.0
pi@raspberrypi:~ $ npm --version
6.4.1
pi@raspberrypi:~ $

closed 2019-01-23 by stale[bot]