Koenkk/zigbee2mqtt #573: Split out device config from configuration.yml

xoxys, 2018-11-07:
Hi,
i try to setup zigee2mqtt with ansible but it is very hard to manage zigbee2mqtt with a config management system if you have a config file with static and dynamic values (like devices). So i would love to see an option to specify a separate file for device informations.

Maybe something like:
devices:
    file: /opt/zigbee2mqtt/data/devices.yml

Thanks for your great work

xoxys, 2018-11-08:
I like it! Will have a look on this at the next days

thinkl33t, 2018-12-03:
Is anyone looking at this?  It would be very useful for our ansible-orchestrated zigbee2mqtt systems :)

xoxys, 2018-12-04:
I have done some first steps locally. But i can't find a good library for that the most of them are outdated. 

The other way is to use a simple string compare/regex to detect !import but it feels a little bit hacky and dirty. @Koenkk what do you think?

Koenkk, 2018-12-04:
That feels indeed hacky, could you find out how home assistant does this?

clockbrain, 2018-12-04:
Don't know how home assistant does it but surely leveraging something like [https://www.npmjs.com/package/yaml-include](https://www.npmjs.com/package/yaml-include)
 or 
[https://www.npmjs.com/package/yaml-import](https://www.npmjs.com/package/yaml-import) would be the way to go about it.

thinkl33t, 2018-12-04:
https://github.com/home-assistant/home-assistant/blob/dev/homeassistant/util/yaml.py

Looks like they've extended the yaml parser.

xoxys, 2018-12-05:
@clockbrain yes that are the libraries i am talking about. But both of them are outdated and i am not sure if one of them is a solid base to start with..

scoootec, 2018-12-15:
@Koenkk could we consider having a seperate devices.yaml by default?

That way configuration.yaml is limited to static configuration only. It would allow much easier automatic config generation in a hassio-addon as an example.

xoxys, 2018-12-16:
I think this would be easier to achieve as struggling around with an include setup. Because this would be a breaking change we can make it optional with an opt-in attribute in the config.

closed 2019-02-21 by stale[bot]

rasmusson, 2023-06-06:
@xoxys any chance of this coming to life?

xoxys, 2023-06-06:
Devices and groups can be managed in a dedicated file already https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html#extract-config-to-separate-files

rasmusson, 2023-06-06:
Awesome I interpreted the stale notice as it just died