Koenkk/zigbee2mqtt #576: CC2531 timeout

mattydanby, 2018-11-09:
HI,

I'm using Hassio with a CC2531 debugger and I have previously had this working with some xiaomi sensors however after I failed to add my first Ikea tradfri light, everything stopped.

I have tried various methods to reset the debugger, including unplugging, using reset button, rebooting Hassio and reinstalling Hassio addon's but I'm now unable to communicate/join any devices.

I have now started a fresh using the following setup and still I'm unable to get ANY devices to join.
Hassio v1.4.0
hassio-zigbee2mqtt v0.1.8

My main area of concern (see below & pastebin) is the timeout messages and the fact that the LED no longer disables.

https://pastebin.com/Kzu2pg6S

How can debug if the issue is debugger firmware or zigbee2mqtt?

Many thanks,

Matty

```
  zigbee2mqtt:debug 2018-11-9 00:21:40 Received mqtt message on topic 'zigbee2mqtt/bridge/config/permit_join' with data 'true'
  zigbee2mqtt:info 2018-11-9 00:21:40 Zigbee: allowing new devices to join.
Fri, 09 Nov 2018 00:21:40 GMT zigbee-shepherd:request REQ --> ZDO:mgmtPermitJoinReq
  zigbee2mqtt:info 2018-11-9 00:21:43 
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/zigbee2mqtt-0.1.8/node_modules/cc-znp/lib/ccznp.js:255:22)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at Timeout.<anonymous> (/zigbee2mqtt-0.1.8/node_modules/cc-znp/lib/ccznp.js:234:18)\n    at ontimeout (timers.js:498:11)\n    at tryOnTimeout (timers.js:323:5)\n    at Timer.listOnTimeout (timers.js:290:5)"}
  zigbee2mqtt:debug 2018-11-9 00:25:40 Saving state to file /share/zigbee2mqtt/state.json

Koenkk, 2018-11-09:
"error":"Unable to get firmware version"`` indicates you are running an older firmware version, could you try upgrading to the latest?

mattydanby, 2018-11-09:
I'm not sure what version I'm running as I bought the device flashed.  I think when I first started looking at zigbee2mqtt I was running v0.1.6.

Could the firmware get corrupted and need reflashing (that seems odd to me), or is it that the firmware is incompatible with the current version of zigbee2mqtt that I'm now using v0.1.8?

Koenkk, 2018-11-11:
It's an older version for sure, otherwise it would show that firmware version. Normally unplugging -> reset button (the one close to the usb port) -> plugging en starting zigbee2mqtt fixes it.

mattydanby, 2018-11-11:
I'll order a debugger so I can flash it with the latest firmware.
But I haven't had any luck with performing a reset. :-(
Thanks

mattydanby, 2018-11-15:
I have been able to flash the debugger and now it is working again.
`zigbee2mqtt:info 2018-11-15 00:03:29 Coordinator firmware version: '20180815'`
Many thanks for your work on this project and your advice.

closed 2018-11-15 by mattydanby