Koenkk/zigbee2mqtt #577: Settings for older Netvox smartplug

shmergin, 2018-11-09:
I have a very old version of the netvox Z809A plug, which doesnt  work with the existing z809A settings. It was made in 2012 and doesnt seem to support `haElectricalMeasurement`, only `seMetering`. 

After a week or two of fiddling I have gotten it working a little by pasting in other peoples code, at this point it reports current power usage and can be switched on and off.  I have no idea what I am doing however, could you please advise the correct devices.js/fromzigbee.js settings?

**Here is my database.db entry for one plug and coordinator:**

> {"id":5,"type":"Router","ieeeAddr":"0x00137a0000008a99","nwkAddr":16629,"manufId":4255,"manufName":"netvox","powerSource":"Mains (single phase)","modelId":"Z809AE3R","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":9,"inClusterList":[0,3,4,5,6,21,1794],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":3,"appVersion":10,"stackVersion":45,"hwVersion":3,"manufacturerName":"netvox","modelId":"Z809AE3R","dateCode":"20120605","powerSource":1,"locationDesc":"         ","physicalEnv":0,"deviceEnabled":1}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":128}},"genScenes":{"dir":{"value":1},"attrs":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}},"genCommissioning":{"dir":{"value":1},"attrs":{"shortress":65535,"extendedPANId":"0xffffffffffffffff","panId":65535,"channelmask":51955712,"startupControl":3,"trustCenterress":"0x0000000000000000","trustCenterMasterKey":{"0":0,"1":0,"2":0,"3":0,"4":0,"5":0,"6":0,"7":0,"8":0,"9":0,"10":0,"11":0,"12":0,"13":0,"14":0,"15":0},"networkKey":"0":255,"1":255,"2":255,"3":255,"4":255,"5":255,"6":255,"7":255,"8":255,"9":255,"10":255,"11":255,"12":255,"13":255,"14":255,"15":255},"useInsecureJoin":1,"preconfiguredLinkKey":{"0":57,"1":48,"2":101,"3":99,"4":110,"5":97,"6":105,"7":108,"8":108,"9":65,"10":101,"11":101,"12":66,"13":103,"14":105,"15":90},"networkKeySeqNum":0,"networkKeyType":5,"networkManagerress":0,"scanAttempts":5,"timeBetweenScans":100,"rejoinInterval":60,"maxRejoinInterval":3600,"indirectPollRate":0,"parentRetryThreshold":0,"concentratorFlag":0,"concentratorRus":15,"concentratorDiscoveryTime":0}},"seMetering":{"dir":{"value":1},"attrs":{"currentSummDelivered":{"0":0,"1":539},"instantaneousDemand":0,"status":0,"unitOfMeasure":0,"multiplier":1,"divisor":1000,"summaFormatting":251,"meteringDeviceType":0}}}}},"_id":"DbOGhQioXI9tPBub"}

> {"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018ed33a6","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1541740748,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}},"_id":"uBh6ZL1lWOGwPEjX"}
> 

**And my devices.js entry:**
```
// Netvox
    {
        zigbeeModel: ['Z809AE3R'],
        model: 'Z809',
        vendor: 'Netvox',
        description: 'Power socket with power consumption monitoring',
        supports: 'on/off, power measurement',
        fromZigbee: [fz.generic_state, fz.ignore_onoff_change, fz.ignore_electrical_change, fz.Z809A_gen, fz.Z809A_seMetering, fz.Z809A_seMeteringatt],
        toZigbee: [tz.onoff],
    },
```

**And my Fromzigbee.js entries:**

    Z809A_gen: {
        cid: 'genOnOff',
        type: 'attReport',
        convert: (model, msg, publish, options) => {
            console.log(msg.data)
        },
        },
    Z809A_seMetering: {
        cid: 'seMetering',
        type:'attReport',
       convert: (model, msg, publish, options) => {
            console.log(msg.data) 
          return {
                power: msg.data.data['instantaneousDemand'],
                current: msg.data.data['currentSummDelivered'],
         };
       },
       },
     Z809A_seMeteringatt: {
        cid: 'seMetering',
        type:'devChange',
       convert: (model, msg, publish, options) => {
          return {
                power: msg.data.data['instantaneousDemand'],
                current: msg.data.data['currentSummDelivered'],
            };
       },
       },  

**This is the data received from seMetering:**

	`Received zigbee message of type 'attReport' with data '{"cid":"seMetering","data":{"57344":78,"57345":248,"57346":12,"57347":546}}'`

**Below are several logs:**

**I connected a lamp to the plug and turned on the switch:**
https://pastebin.com/JXQxWvRW
 
**Turned on the lamp at 17:24, then turned it off at 17:33**
https://pastebin.com/eSquZZGV

**Looking at some documentation for this version of the Z809A I found some information which may help:**

> "The related Attribute ID of Simple Metering Cluster ID: 
>  Current Attribute ID: 0xE000; unit: mA
>  Voltage Attribute ID: 0xE001; unit: V
>  Power Attribute ID: 0xE002; unit: W
>  Energy Attribute ID: 0xE003; related to AttributeID CurrentSummationDeliver (0x0000); unit: Wh
> 
> The related Attribute ID of Electrical Measurement Cluster ID: 
>  Current Attribute ID: 0x0508
>  Voltage Attribute ID: 0x0505
>  Power Attribute ID: 0x050B
>  Power Factor Attribute ID: 0x0510"

I also found this information about my specific z809A model which might help:

> "This utilizes Simple Metering cluster (0x0702) defined within Smart Energy Profile (SEP). Voltage, Current, Power and Energy are Netvox proprietary attributes.  
>  
> CurrentSummationDeliver Attribute of SEP corresponds to Energy Attribute of Netvox proprietary. (Unit kwh); 
> InstantaneousDemand Attribute of SEP corresponds to Power Attribute of Netvox proprietary. (Unit kw); 
> Current Attribute of Netvox proprietary. (Unit mA); 
> Voltage Attribute of Netvox proprietary. (Unit V)"

Koenkk, 2018-11-12:
Looks good already. 

We need to find out what the values of `{"57344":83,"57345":245,"57346":13,"57347":548}}` mean.

Can you try plugging the bulb out of the socket while having it turned on. How much power does your lamp consume?

shmergin, 2018-11-15:
Hi there,

For plug 0x00137a0000008a99

This is the log with the lamp unplugged (nothing using power): 
https://pastebin.com/Da7bHaSM

This is the log after i plugged in and turned on the lamp:  
https://pastebin.com/DZDS0S7N

To me it looks like:
'57344': I don't know... 
'57345': is Voltage - V
'57346': is current power usage (at that moment) - watts?
'57347': is total power used (over the lifetime of the plug) - kWh?

what do you think?

shmergin, 2018-11-17:
@Koenkk It is a CFL bulb listed as 12Watts

**Full bulb details:**
740lm @ 61lm/W
220-240V

Koenkk, 2018-11-19:
`57346` is indeed the current power usage in watt.

I'm not sure about `57345`, 249 volt is too much.

closed 2019-01-25 by stale[bot]