Koenkk/zigbee2mqtt #578: Documentation question

TinajaLabs, 2018-11-09:
Just came across this and it looks exciting.

I'm looking at the ```get started``` part and I see references to flashing a device with stuff from TI.  For me it would be helpful to understand why I'm doing all that.

I have xbee radios that I configured with xctu and I have a raspi with my central controller xbee and about a dozen xbees sending data to into that.  I have a python script that collects the data input and immediately sends packets to an MQTT node in node-red running on another raspi.  It works but it's very tedious.  As a result, I mostly prefer devices like the esp8266 wifi micro-controllers.

Your solution looks like it can work for a broader set of zigbee (maybe xbee?) devices but I don't understand how the hardware fits and why we flash the device that, too me, looks like a specialized FTDI connection.  Is it for loading configs into the devices?  Is it for running the devices?  Both?

Thanks,
Chris.

Koenkk, 2018-11-09:
The device needs to be flashed with a coordinator firmware which is required to execute commands and retrieve values from devices.

TinajaLabs, 2018-11-12:
So the device would become a type of gateway for all my zigbee devices and convert their output into an MQTT stream?  Is discovery of zigbee devices part of the feature set?

Thanks,
Chris.

Koenkk, 2018-11-12:
Yes the cc2531 is the gateway.

What do you mean with discovery?

closed 2019-01-18 by stale[bot]