Koenkk/zigbee2mqtt #581: zigbee2mqtt with a cc2530 as connector

HouseFox, 2018-11-10:
Hello,

how i can use zigbee2mqtt form Koen Kanters "Koenkk" with a cc2530 as connector.
I want to use a cc2530 with external antenna.
Is there a step by step instruction for beginners.
Is there a working solution?

Thanks in advanced.

khzd, 2018-11-12:
Hi you mean as coordinator ?

HouseFox, 2018-11-12:
Hi, sry yes i mean coordinator.

khzd, 2018-11-15:
Hi
I use one as a router and is working fine :)
Do you try https://github.com/kirovilya/files/blob/master/CC2530ZNP-Pro-Secure_LinkKeyJoin.hex
?
Cheers

HouseFox, 2018-11-17:
hi @khzd im a Little bit busy at the moment

khzd, 2018-11-24:
hi @HouseFox  did you succeed to install it as a  coordinator ? 
cheers

HouseFox, 2018-11-26:
Hi @khzd it seems so that i have the worng wireing to cc2530. 
So i search the right instructions...
Do you have some good News?

khzd, 2018-12-01:
Hi sorry for the delay, was busy with a new livolo switch, cannot Change now, will try later

khzd, 2019-02-04:
Hi 
Sorry for the delay, I set up 2 cc2530 this we to have 2 separate network on different channels.
It work fine for the moment , will give feed back later if any issue.
The board are those in my previous post.
I never had a so good linkquality :)
Cheers

closed 2019-02-11 by stale[bot]