Koenkk/zigbee2mqtt #584: zigbee2mqtt start error

HouseFox, 2018-11-10:
Hi community,
can someone tell me what this error is all about?
Is this error known and resolved?

Last login: Sat Nov 10 23:56:48 2018 from 192.168.2.102
pi@raspberrypi:~ $ cd /opt/zigbee2mqtt
pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@0.1.8 start /opt/zigbee2mqtt
> node index.js

/opt/zigbee2mqtt/lib/util/logger.js:10
const level = settings.get().advanced.log_level;
                                     ^

TypeError: Cannot read property 'log_level' of null
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/util/logger.js:10:38)
    at Module._compile (module.js:653:30)
    at Object.Module._extensions..js (module.js:664:10)
    at Module.load (module.js:566:32)
    at tryModuleLoad (module.js:506:12)
    at Function.Module._load (module.js:498:3)
    at Module.require (module.js:597:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/mqtt.js:2:16)
    at Module._compile (module.js:653:30)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.1.8 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@0.1.8 start script.
npm ERR! This is probably not a problem with npm. There is likely additional log                                                                          ging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2018-11-10T23_29_24_523Z-debug.log
pi@raspberrypi:/opt/zigbee2mqtt $

pi@raspberrypi:/opt/zigbee2mqtt $ nano /home/pi/.npm/_logs/2018-11-10T23_29_24_523Z-debug.log
  GNU nano 2.7.4                                 Datei: /home/pi/.npm/_logs/2018-11-10T23_29_24_523Z-debug.log

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.4.1
3 info using node@v8.12.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@0.1.8~prestart: zigbee2mqtt@0.1.8
6 info lifecycle zigbee2mqtt@0.1.8~start: zigbee2mqtt@0.1.8
7 verbose lifecycle zigbee2mqtt@0.1.8~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@0.1.8~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/$
9 verbose lifecycle zigbee2mqtt@0.1.8~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@0.1.8~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@0.1.8~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@0.1.8~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@0.1.8 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at emitTwo (events.js:126:13)
13 verbose stack     at EventEmitter.emit (events.js:214:7)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at emitTwo (events.js:126:13)
13 verbose stack     at ChildProcess.emit (events.js:214:7)
13 verbose stack     at maybeClose (internal/child_process.js:915:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
14 verbose pkgid zigbee2mqtt@0.1.8
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.14.71-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v8.12.0
19 verbose npm  v6.4.1
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@0.1.8 start: `node index.js`
22 error Exit status 1

Koenkk, 2018-11-11:
Can you post the contents of your `data/configuration.yaml`?

HouseFox, 2018-11-14:
@Koenkk

HouseFox, 2018-11-14:
@Koenkk sry for the formation...

HouseFox, 2018-11-14:
@Koenkk

Koenkk, 2018-11-16:
These are audit reports which you can ignore.

HouseFox, 2018-11-16:
Hi @Koenkk , do think it is really a Report that i can ignore?
But i think that is my Problem. 
Can you give me please Support, or is it possible that you do the same installation with the actually Version of node they i installed.
What can i test to find the mistake/Problem...

Thanks for your help!

Using username "pi".
Linux raspberrypi 4.14.71-v7+ #1145 SMP Fri Sep 21 15:38:35 BST 2018 armv7l
The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.
Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Nov 12 22:11:23 2018 from 192.168.2.102
pi@raspberrypi:~ $ node --version
v8.12.0
pi@raspberrypi:~ $ npm --version
6.4.1
pi@raspberrypi:~ $

HouseFox, 2018-11-16:
@Koenkk you think that my Installation of zigbee2mqtt ins´t right?
can you give me support how i uninstall zigbee2mqtt in a clean way.
I think i had installed zigbee2mqtt like your insttruction.
https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge
At least i reinstalled it like Code below.
Could be mosquitto be the Problem?

Using username "pi".
Linux raspberrypi 4.14.71-v7+ #1145 SMP Fri Sep 21 15:38:35 BST 2018 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Nov 16 21:41:08 2018 from 192.168.2.102
pi@raspberrypi:~ $ # Setup Node.js repository
pi@raspberrypi:~ $ # For Pi Zero use https://raw.githubusercontent.com/sdesalas/node-pi-zero/master/install-node-v8.10.0.sh
pi@raspberrypi:~ $ sudo curl -sL https://deb.nodesource.com/setup_8.x | sudo -E bash -

## Installing the NodeSource Node.js 8.x LTS Carbon repo...

## Populating apt-get cache...

+ apt-get update
Holen:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15,0 kB]
Holen:2 http://archive.raspberrypi.org/debian stretch InRelease [25,3 kB]
Holen:3 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11,7 MB]
OK:4 https://deb.nodesource.com/node_8.x stretch InRelease
Holen:5 http://archive.raspberrypi.org/debian stretch/main armhf Packages [199 kB]
Holen:6 http://archive.raspberrypi.org/debian stretch/ui armhf Packages [39,1 kB]
Es wurden 11,9 MB in 13 s geholt (875 kB/s).
Paketlisten werden gelesen... Fertig

## Confirming "stretch" is supported...

+ curl -sLf -o /dev/null 'https://deb.nodesource.com/node_8.x/dists/stretch/Release'

## Adding the NodeSource signing key to your keyring...

+ curl -s https://deb.nodesource.com/gpgkey/nodesource.gpg.key | apt-key add -
OK

## Creating apt sources list file for the NodeSource Node.js 8.x LTS Carbon repo...

+ echo 'deb https://deb.nodesource.com/node_8.x stretch main' > /etc/apt/sources.list.d/nodesource.list
+ echo 'deb-src https://deb.nodesource.com/node_8.x stretch main' >> /etc/apt/sources.list.d/nodesource.list

## Running `apt-get update` for you...

+ apt-get update
OK:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
OK:2 http://archive.raspberrypi.org/debian stretch InRelease
OK:3 https://deb.nodesource.com/node_8.x stretch InRelease
Paketlisten werden gelesen... Fertig

## Run `sudo apt-get install -y nodejs` to install Node.js 8.x LTS Carbon and npm
## You may also need development tools to build native addons:
     sudo apt-get install gcc g++ make
## To install the Yarn package manager, run:
     curl -sL https://dl.yarnpkg.com/debian/pubkey.gpg | sudo apt-key add -
     echo "deb https://dl.yarnpkg.com/debian/ stable main" | sudo tee /etc/apt/sources.list.d/yarn.list
     sudo apt-get update && sudo apt-get install yarn

pi@raspberrypi:~ $
pi@raspberrypi:~ $ # Install Node.js
pi@raspberrypi:~ $ sudo apt-get install -y nodejs git make g++ gcc
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
g++ ist schon die neueste Version (4:6.3.0-4).
gcc ist schon die neueste Version (4:6.3.0-4).
git ist schon die neueste Version (1:2.11.0-3+deb9u4).
make ist schon die neueste Version (4.1-9.1).
nodejs ist schon die neueste Version (8.12.0-1nodesource1).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 27 nicht aktualisiert.
pi@raspberrypi:~ $ # Verify that the correct nodejs and npm (automatically installed with nodejs)
pi@raspberrypi:~ $ # version has been installed
pi@raspberrypi:~ $ node --version  # Should output v8.X.X (at time of writing v8.11.2)
v8.12.0
pi@raspberrypi:~ $ npm --version  # Should output > 5.X.X (at time of writing 5.6.0)
6.4.1
pi@raspberrypi:~ $ # Clone zigbee2mqtt repository
pi@raspberrypi:~ $ sudo git clone https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt
fatal: Zielpfad '/opt/zigbee2mqtt' existiert bereits und ist kein leeres Verzeichnis.
pi@raspberrypi:~ $ sudo chown -R pi:pi /opt/zigbee2mqtt
pi@raspberrypi:~ $ # Install dependencies
pi@raspberrypi:~ $ cd /opt/zigbee2mqtt
pi@raspberrypi:/opt/zigbee2mqtt $ npm install
audited 1830 packages in 25.2s
found 3 moderate severity vulnerabilities
  run `npm audit fix` to fix them, or `npm audit` for details
pi@raspberrypi:/opt/zigbee2mqtt $ node-pre-gyp info ok
-bash: node-pre-gyp: Kommando nicht gefunden.
pi@raspberrypi:/opt/zigbee2mqtt $ nano /opt/zigbee2mqtt/data/configuration.yaml
  GNU nano 2.7.4                     Datei: /opt/zigbee2mqtt/data/configuration.yaml                      Verändert

https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge# Home Assistant integration (MQTT discovery)
....homeassistant: false
..
# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
    server: 'mqtt://localhost'
  # server: 'mqtt://192.168.2.105'
  # MQTT server authentication, uncomment if required:
   user: xxxxxxxxx
   password: xxxxxxxxx

# Serial settings
serial:
    baudrate: 115200

  # Location of FTDI/ CC2530
    port: /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0

advanced:
rtscts: false

HouseFox, 2018-11-17:
Hi @Koenkk thanks for your support!
I investigate my yaml and i get a lot of mistakes...
So i copy yours from https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge ;
save it and try to run...
But i get the error.
Also when i insert my Password inside yaml.
Below two starts with and without Password
Also the log at leased.

```
# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://localhost'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

Using username "pi".
Linux raspberrypi 4.14.71-v7+ #1145 SMP Fri Sep 21 15:38:35 BST 2018 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sat Nov 17 20:11:00 2018 from 192.168.2.102
pi@raspberrypi:~ $ nano /opt/zigbee2mqtt/data/configuration.yaml
pi@raspberrypi:~ $ cd /opt/zigbee2mqtt
pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@0.1.8 start /opt/zigbee2mqtt
> node index.js

/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:171
  throw generateError(state, message);
  ^
YAMLException: end of the stream or a document separator is expected at line 4, column 18:
    ....homeassistant: false
                     ^
    at generateError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:165:10)
    at throwError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:171:9)
    at readDocument (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1512:5)
    at loadDocuments (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1548:5)
    at load (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1569:19)
    at Object.safeLoad (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1591:10)
    at read (/opt/zigbee2mqtt/lib/util/settings.js:29:17)
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/util/settings.js:17:16)
    at Module._compile (module.js:653:30)
    at Object.Module._extensions..js (module.js:664:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.1.8 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@0.1.8 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output abov        e.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2018-11-17T19_40_44_717Z-debug.log
pi@raspberrypi:/opt/zigbee2mqtt $
pi@raspberrypi:/opt/zigbee2mqtt $

######################################

pi@raspberrypi:/opt/zigbee2mqtt $ nano /opt/zigbee2mqtt/data/configuration.yaml
  GNU nano 2.7.4               Datei: /opt/zigbee2mqtt/data/configuration.yaml
[... 63 lines cut ...]
8 verbose lifecycle zigbee2mqtt@0.1.8~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle$
9 verbose lifecycle zigbee2mqtt@0.1.8~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@0.1.8~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@0.1.8~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@0.1.8~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@0.1.8 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/$
13 verbose stack     at emitTwo (events.js:126:13)
13 verbose stack     at EventEmitter.emit (events.js:214:7)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/$
13 verbose stack     at emitTwo (events.js:126:13)
13 verbose stack     at ChildProcess.emit (events.js:214:7)
13 verbose stack     at maybeClose (internal/child_process.js:915:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
14 verbose pkgid zigbee2mqtt@0.1.8
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.14.71-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v8.12.0
19 verbose npm  v6.4.1
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@0.1.8 start: `node index.js`
```

Koenkk, 2018-11-19:
Use the following yaml: https://github.com/Koenkk/zigbee2mqtt/blob/master/data/configuration.yaml

HouseFox, 2018-11-19:
Hi @Koenkk,

thanks for your link.

```
pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@0.1.8 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 2018-11-19 20:15:45 Logging to directory: '/opt/zigbee2mqtt/data/log/2018-11-19.20-15-45'
  zigbee2mqtt:info 2018-11-19 20:15:46 Starting zigbee2mqtt version 0.1.8 (commit #97cc56b)
  zigbee2mqtt:info 2018-11-19 20:15:46 Starting zigbee-shepherd
  zigbee2mqtt:info 2018-11-19 20:15:46 Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
  zigbee2mqtt:info 2018-11-19 20:16:46 Starting zigbee-shepherd
  zigbee2mqtt:error 2018-11-19 20:16:46 Error while starting zigbee-shepherd!
  zigbee2mqtt:error 2018-11-19 20:16:46 Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 2018-11-19 20:16:46 Failed to start
        {"message":"Error: No such file or directory, cannot open /dev/ttyACM0","stack":"Error: Error: No such file or directory, cannot open /dev/ttyACM0"}
npm start
```

I copy paste only your yaml, but i think i cant use it in this style...

Because,
1. i want to use a CC2530 with external antenna.
2. and i have a user and Password defined as i installed mqtt mosquitto.
So regarding this i Change yoour yaml a Little bit.
Do you think that is ok?

```
# Home Assistant integration (MQTT discovery)

homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt

  # MQTT server URL
  server: 'mqtt://localhost'

  # MQTT server authentication, uncomment if required:
   user: ameissner
   password: test1234

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  # port: /dev/ttyACM0
  # Location of FTDI/ CC2530
    port: /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controll$
```

With the change of your Code i get this error...

```
pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@0.1.8 start /opt/zigbee2mqtt
> node index.js

/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:171
  throw generateError(state, message);
  ^
YAMLException: bad indentation of a mapping entry at line 17, column 4:
       user: ameissner
       ^
    at generateError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:165:10)
    at throwError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:171:9)
    at readBlockMapping (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1080:7)
    at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1332:12)
    at readBlockMapping (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1062:11)
    at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1332:12)
    at readDocument (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1492:3)
    at loadDocuments (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1548:5)
    at load (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1569:19)
    at Object.safeLoad (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1591:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.1.8 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@0.1.8 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2018-11-19T19_33_08_931Z-debug.log
pi@raspberrypi:/opt/zigbee2mqtt $
```

Error log Looks like...
```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.4.1
3 info using node@v8.12.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@0.1.8~prestart: zigbee2mqtt@0.1.8
6 info lifecycle zigbee2mqtt@0.1.8~start: zigbee2mqtt@0.1.8
7 verbose lifecycle zigbee2mqtt@0.1.8~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@0.1.8~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin$
9 verbose lifecycle zigbee2mqtt@0.1.8~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@0.1.8~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@0.1.8~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@0.1.8~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@0.1.8 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:$
13 verbose stack     at emitTwo (events.js:126:13)
13 verbose stack     at EventEmitter.emit (events.js:214:7)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:$
13 verbose stack     at emitTwo (events.js:126:13)
13 verbose stack     at ChildProcess.emit (events.js:214:7)
13 verbose stack     at maybeClose (internal/child_process.js:915:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
14 verbose pkgid zigbee2mqtt@0.1.8
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.14.71-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v8.12.0
19 verbose npm  v6.4.1
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@0.1.8 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@0.1.8 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

So works the Code only with CC2531 ?
Is that really the reason?
What is so different between CC2530 and CC2531?
How i can use CC2530?
Is it possible to test the yaml and Installation without a sniffer?
What tells the error log?
What do you think is the Problem?

Koenkk, 2018-11-19:
You have an error in your YAML format: `YAMLException: bad indentation of a mapping entry at line 17, column 4:`. Please read the yaml documentation an verify the configuration.yaml using a YAML linter (http://www.yamllint.com/)

HouseFox, 2018-11-19:
for my Background which yaml documentation do you prefer?
It seems so that my yaml doesn´t like user and Password... 
for me not clear why...?
but without user and Password.... can´t work or???

I´m very thankful for every help i can get!

```
pi@raspberrypi:~ $ nano /opt/zigbee2mqtt/data/configuration.yaml
  GNU nano 2.7.4                     Datei: /opt/zigbee2mqtt/data/configuration.yaml                      Verändert

# Home Assistant integration (MQTT discovery)

homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt

  # MQTT server URL
  server: 'mqtt://localhost'

  # MQTT server authentication, uncomment if required:
  # user: ameissner
  # password: test1234

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  # port: /dev/ttyACM0
  # Location of FTDI/ CC2530
    port: /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controll$
```

Is that a zigbee-shepherd, a usb or .... Problem?
At cc2530 i cant press a button, there is no button.

```
pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@0.1.8 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 2018-11-19 21:30:25 Logging to directory: '/opt/zigbee2mqtt/data/log/2018-11-19.21-30-24'
  zigbee2mqtt:info 2018-11-19 21:30:26 Starting zigbee2mqtt version 0.1.8 (commit #97cc56b)
  zigbee2mqtt:info 2018-11-19 21:30:26 Starting zigbee-shepherd
  zigbee2mqtt:info 2018-11-19 21:30:26 Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
  zigbee2mqtt:info 2018-11-19 21:31:26 Starting zigbee-shepherd
  zigbee2mqtt:error 2018-11-19 21:31:26 Error while starting zigbee-shepherd!
  zigbee2mqtt:error 2018-11-19 21:31:26 Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 2018-11-19 21:31:26 Failed to start
        {"message":"Error: No such file or directory, cannot open /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controll$","stack":"Error: Error: No such file or directory, cannot open /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controll$"}
```

Koenkk, 2018-11-19:
No yaml error, so that ok now.

The location of your CC device is not ok (port `/dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controll$`). 

What is the output of `ls /dev`?

HouseFox, 2018-11-19:
```
pi@raspberrypi:~ $ nano /opt/zigbee2mqtt/data/configuration.yaml
  GNU nano 2.7.4                     Datei: /opt/zigbee2mqtt/data/configuration.yaml                      Verändert

# Home Assistant integration (MQTT discovery)

homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt

  # MQTT server URL
  server: 'mqtt://localhost'

  # MQTT server authentication, uncomment if required:
  # user: ameissner
  # password: test1234

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  # port: /dev/ttyACM0
  # Location of FTDI/ CC2530
    port: /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controll$
```

```
pi@raspberrypi:~ $ ls -l /dev/serial/by-id
insgesamt 0
lrwxrwxrwx 1 root root 13 Nov 19 21:37 usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0 -> ../../ttyUSB0
```

sehraf, 2018-11-19:
> port: /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controll$

See the $ at the end? That's not the complete name/path.

HouseFox, 2018-11-20:
@Koenkk @sehraf i don´t know why but the port is wrong...

```
pi@raspberrypi:~ $ ls -l /dev/serial/by-id
insgesamt 0
lrwxrwxrwx 1 root root 13 Nov 19 22:41 usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0 -> ../../ttyUSB0
pi@raspberrypi:~ $

```
I change the yaml to the right seen in ls /dev.
Also i activate user and Password.
```
Using username "pi".
Linux raspberrypi 4.14.71-v7+ #1145 SMP Fri Sep 21 15:38:35 BST 2018 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Nov 19 22:40:11 2018 from 192.168.2.102
pi@raspberrypi:~ $ nano /opt/zigbee2mqtt/data/configuration.yaml
pi@raspberrypi:~ $ nano /opt/zigbee2mqtt/data/configuration.yaml
  GNU nano 2.7.4                     Datei: /opt/zigbee2mqtt/data/configuration.yaml

# Home Assistant integration (MQTT discovery)

homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt

  # MQTT server URL
  server: 'mqtt://localhost'

  # MQTT server authentication, uncomment if required:
    user: ameissner
    password: test1234

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  # port: /dev/ttyACM0
  # Location of FTDI/ CC2530
  # port: /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controll$
    port: /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0
```
Error again.
And i don´t understand why i have a YAMLExecption with my user...
_**YAMLException: bad indentation of a mapping entry at line 17, column 5:**_
Can someone explain me this?

```
pi@raspberrypi:~ $ cd /opt/zigbee2mqtt
pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@0.1.8 start /opt/zigbee2mqtt
> node index.js

/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:171
  throw generateError(state, message);
  ^
YAMLException: bad indentation of a mapping entry at line 17, column 5:
        user: ameissner
        ^
    at generateError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:165:10)
    at throwError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:171:9)
    at readBlockMapping (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1080:7)
    at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1332:12)
    at readBlockMapping (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1062:11)
    at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1332:12)
    at readDocument (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1492:3)
    at loadDocuments (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1548:5)
    at load (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1569:19)
    at Object.safeLoad (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1591:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.1.8 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@0.1.8 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2018-11-20T19_58_13_893Z-debug.log
pi@raspberrypi:/opt/zigbee2mqtt $
```
```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.4.1
3 info using node@v8.12.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@0.1.8~prestart: zigbee2mqtt@0.1.8
6 info lifecycle zigbee2mqtt@0.1.8~start: zigbee2mqtt@0.1.8
7 verbose lifecycle zigbee2mqtt@0.1.8~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@0.1.8~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin$
9 verbose lifecycle zigbee2mqtt@0.1.8~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@0.1.8~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@0.1.8~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@0.1.8~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@0.1.8 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:$
13 verbose stack     at emitTwo (events.js:126:13)
13 verbose stack     at EventEmitter.emit (events.js:214:7)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:$
13 verbose stack     at emitTwo (events.js:126:13)
13 verbose stack     at ChildProcess.emit (events.js:214:7)
13 verbose stack     at maybeClose (internal/child_process.js:915:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
14 verbose pkgid zigbee2mqtt@0.1.8
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.14.71-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v8.12.0
19 verbose npm  v6.4.1
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@0.1.8 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@0.1.8 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

pixeldoc2000, 2018-11-20:
YAML require a very strict format.

Change:
```
  # MQTT server authentication, uncomment if required:
    user: ameissner
    password: test1234
```

To:
```
  # MQTT server authentication, uncomment if required:
  user: ameissner
  password: test1234
```
^ NOTE: only **two whitespaces in front** of user and everywhere else! 

`    port=` is also incorrectly formated.

**Check your indentation everywhere!**

HouseFox, 2018-11-20:
Hello @pixeldoc2000 thanks for your answer and help. @Koenkk @sehraf 
i changed the yaml to your instructions.
```
Using username "pi".
Linux raspberrypi 4.14.71-v7+ #1145 SMP Fri Sep 21 15:38:35 BST 2018 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Tue Nov 20 22:25:13 2018 from 192.168.2.102
pi@raspberrypi:~ $ nano /opt/zigbee2mqtt/data/configuration.yaml
  GNU nano 2.7.4                               Datei: /opt/zigbee2mqtt/data/configuration.yaml

# Home Assistant integration (MQTT discovery)

homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt

  # MQTT server URL
  server: 'mqtt://localhost'

  # MQTT server authentication, uncomment if required:
  user: ameissner
  password: test1234

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  # port: /dev/ttyACM0
  # Location of FTDI/ CC2530
  # port: /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controll$
  port: /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0
```

Now with this modifications i run into the following error.
At error text i see "Press the reset button on the stick (the one closest to the USB) and start again" but i use the CC2530 mit external antenna and there is no button to press... 
The following messages i don´t understand what is the meaning?

```
pi@raspberrypi:~ $ cd /opt/zigbee2mqtt
pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@0.1.8 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 2018-11-20 22:40:12 Logging to directory: '/opt/zigbee2mqtt/data/log/2018-11-20.22-40-12'
  zigbee2mqtt:info 2018-11-20 22:40:12 Starting zigbee2mqtt version 0.1.8 (commit #97cc56b)
  zigbee2mqtt:info 2018-11-20 22:40:12 Starting zigbee-shepherd
  zigbee2mqtt:info 2018-11-20 22:40:15 Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
  zigbee2mqtt:info 2018-11-20 22:41:15 Starting zigbee-shepherd
  zigbee2mqtt:error 2018-11-20 22:41:19 Error while starting zigbee-shepherd!
  zigbee2mqtt:error 2018-11-20 22:41:19 Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 2018-11-20 22:41:19 Failed to start
        {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:255:22)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:234:18)\n    at ontimeout (timers.js:498:11)\n    at tryOnTimeout (timers.js:323:5)\n    at Timer.listOnTimeout (timers.js:290:5)"}

```

Koenkk, 2018-11-22:
zigbee-shepherd fails to communicate with the CC2530, probably something is not connected right. 

I would suggest starting with a CC2531 (which is easier to use).

HouseFox, 2018-11-26:
Hi @Koenkk i´m little bit busi at the moment...
At the Moment i have only cc2530 by my side.
Also i prefer the extented range of cc2530 with antenna.
You mean that i have wired the cc2530 wrong.
Can you give me a instruction or URL which works with cc2530 and zigbee-shepherd?
thanks

Koenkk, 2018-11-27:
There are no clear instructions yet, but you can find some here: https://github.com/Koenkk/zigbee2mqtt/issues/465

closed 2019-02-02 by stale[bot]