Koenkk/zigbee2mqtt #585: Hue Dimmer Switch: long/short press

Bearonic, 2018-11-10:
Hi,
I use an Hue Dimmer Switch (ioBroker zigbee adapter 0.7.7 / flashed CC2531 on RPi3). In principle it does work. It was easily paired and button presses are recognized.
Previosly, I had it paired with an Hue Bridge 2.1 and used a script with node-hue-api in ioBroker to poll the states. That way it was possible to distinguish between short and long (> ~1s) presses for all 4 buttons.

With zigbee2mqtt it is not possible to detect short and long presses. Rather, the "I" and "O" buttons do only toggle one state and the darker/brighter buttons always create a "button down" followed by a "button released" event. This reduces the flexibility and range of functions of the Dimmer Switch significantly.

I would love to see the same behavior as when using a Hue Bridge. Would this be possible? I already digged into the 'devices.js' and 'fromZigbee.js', but I have no idea what's to do. 

Note:
Interestingly, the flashing of the LED on the Dimmer Switch differs from when paired with the Hue Bridge. I guess, this means, that the Hue Bridge sends messages to the Dimmer Switch as well.

sehraf, 2018-11-11:
Hi,
for me the Dimmer Switch produces the following events:
"I" and "O" button: `"action":"on"` / `"action":"off"`
Darker/brighter: `"action":"down-press"` and when pressed long additionally `"action":"down-hold"` followed by `"down-hold-release"` (similar for `up`)

Using zigbee2mqtt 1.8 with a CC2531.

When pressing the "I" or "O" button long the Dimmer Switch flashes red, that's indeed interesting (i don't have a Hue Bridge to test the behavior when connected to it)

sti0, 2019-01-03:
I watched the same behavior. Pressing dim buttons short will give a "action":"up-press" / "action":"down-press". Holding the dim buttons I get an "action":"up-hold" event following by an "action":"up-hold-release" event after releasing. This seams to work great.

But holding the I/O buttons doesn't have the same functionality (like mentioned by @Bearonic ). There are only "action":"on" / "action":"off" responses. Would be great if there were pressing responses like dim buttons. 

You can configurate this events on the hue bridge via thrid party apps like all4hue or HueEssentials. This would increase the functionality of the dim switch with 2 more actions, which i need in my smart home environment. Therefore I have to stay on hue bridge till this is available on zigbee2mqtt.

james-fry, 2019-01-22:
Agree it would be great if this switch can support:

Single, double, triple and quadruple click
Long press
Long press release

This would really open these controllers up for a lot of functions.
e.g. 
click off - turn off my beside lamp
double click - turn off all lamps in room
triple click - turn off all lamps in house (bed time mode :))

Especially long press and long press release are useful for the on and off buttons since these are not also used for the brightness up/down count.

I guess this needs to be implemented in code with click response counting with timers etc since I'm assuming that the dimmer itself is not sending these events.

Koenkk, 2019-01-23:
The Hue dimmer switch does not provide a key up event, so we cannot detect the time between press and release.

james-fry, 2019-01-23:
Hi @Koenkk 
That is weird - it seems to be supported by the dimmer when used with hue hub.
This HA custom component supports it:
https://github.com/robmarkcole/Hue-sensors-HASS

> The RWL dimmer remote has a total of 8 possible states, corresponding to the remote having 4 buttons, where each button can be short pressed (clicked) and long pressed (when holding the button for 2 sec you will see the LED blink twice). The ZPG (or Tap) remote has 4 buttons.

Also this article corroborates:
https://www.howtogeek.com/244803/how-to-re-program-the-hue-dimmer-switch-to-do-anything-with-your-lights/

For the Single, double, triple and quadruple click can a timer based implementation work?

Koenkk, 2019-01-23:
@james-fry you can add some logging here https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/af.js#L689 to check if some commands are ignored. (`console.log(msg.zclMsg.cmdId)`)

I doubt about the timer thing as this is very timing critical. When there is a little lag on the network a double presses could easily be missed. After the first click you have to wait for X time to check if another click is received, this will also decrease the responsiveness for the single click.

sti0, 2019-01-25:
On hue bridge and custom HA component by @robmarkcole you can't use double tab, triple tab etc. But you can detect hold and release events for the on and off button.

St0Ma, 2019-02-12:
@Koenkk I have added the suggested console.log(..) entry in af.js. But there are no other states published. 

The same model of my hue dimmer switch connected to the hue bridge fires continuesly events when holding on button (i think because the green light continuously blinks. With zigbee2mqtt continously holding the on button results in a red light.

In another thread somebody suggested that the remote has a possibility to bind directly to lights:

https://github.com/Koenkk/zigbee2mqtt/issues/36

Could that be the trick? How can i help?

Koenkk, 2019-02-13:
@St0Ma binding will probably give a much better experience. This is explained in #782

james-fry, 2019-02-13:
I wonder if the coordinator that the dimmer is paired with can change how it operates. It is possible that an end device can do that? Would the sequence of interactions during pairing allow that?

Koenkk, 2019-02-13:
@james-fry yes, every device can set this up for a different device, I just has to send to commands to the device.

St0Ma, 2019-02-13:
@Koenkk 
I don´t get it. THe Hue Switch can be paired to a device / bulb with a long press. But how can that change the behaviour? 

Maybe it´s only a Software Thing in the Hue Bridge, that the next pressed ok button is assigned to a different scene?

Koenkk, 2019-02-13:
@St0Ma sniffing Hue traffic is possible, see https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html (be sure to search for *The Hue bridge uses a different Trust Center link key* on that page)

ugrug, 2019-02-20:
@Koenkk 
I just started to migrate my hue devices including dimmer and also just realised the missing feature.
Here is the result of sniffing the hue network.
1. press key 1 (on button):
[image]
2. release key 1 (on button):
[image]
3. holding key 1 (on button):
a.
[image]
b.
[image]
c.
[image]

Does it helps?

ugrug, 2019-02-22:
Hi @Koenkk,
Did you get sniffed network traffic about philips hue dimmer pairing and long-pressing on/off buttons?
I found remote binding after pairing on 0xfc00 cluster.

[image]

Koenkk, 2019-02-23:
@ugrug no, I don't have the hue dimmer/hue bridge. Can you check what is send when you long press the dimmer switch?

ugrug, 2019-02-24:
@Koenkk 
The messages paired with philips bridge:
[image]

All message as on the screenshot: cluster 0xfc00, dst endpoint 65, manufacturer-specific, manufacturer-code: 0x100b, command, 0x00, the 8 data bytes are different:
1. ON button:
- Click: 01 00 00 30 00 21 00 00
- Hold (800ms after click message and in every 800ms):
(1st) 01 00 00 30 01 21 08 00
(2nd) 01 00 00 30 01 21 10 00
(3rd) 01 00 00 30 01 21 18 00
(4th) 01 00 00 30 01 21 20 00
...
- Release: 01 00 00 30 03 21 23 00

2. UP button:
- Click: 02 00 00 30 00 21 00 00
- Hold (800ms after click message and in every 800ms): 
(1st) 02 00 00 30 01 21 08 00
(2nd) 02 00 00 30 01 21 10 00
...
- Release: 02 00 00 30 03 21 24 00

3.  DOWN button:
- Click: 03 00 00 30 00 21 00 00
- Hold (800ms after click message and in every 800ms): 
(1st) 03 00 00 30 01 21 08 00
(2nd) 03 00 00 30 01 21 10 00
...
- Release: 03 00 00 30 03 21 21 00

4. OFF button:
- Click: 04 00 00 30 00 21 00 00
- Hold (800ms after click message and in every 800ms):
(1st) 04 00 00 30 01 21 08 00
(2nd) 04 00 00 30 01 21 10 00
(3rd) 04 00 00 30 01 21 18 00
...
- Release: 04 00 00 30 03 21 2b 00

The pairing process:
[image]
[image]
[image]

I'm currently on master branch.
I checked, on and off message working currently for on/off button,
press and hold message for up/down button on zigbee2mqtt. 

I missing hold for on/off button but unfortunatelly there is red light on hue dimmer this case with zigbee2mqtt currently.

For me this writing request was interesting also
[image]
I was playing with bind to 0xfc00 cluster, tried to put missing unknown manufacturer write foundation after pairing with no success.

It should be also nice to implement release message also.

james-fry, 2019-02-24:
Thanks for this investigation! I hope it can lead to more comprehensive support for these dimmers.

Koenkk, 2019-02-24:
@ugrug can you add `console.log('HELLO WORLD', msg.zclMsg.cmdId)` here https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/af.js#L693 and trigger the long press? We probably need to add the missing cmdids here: https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/af.js#L689

ugrug, 2019-02-25:
@Koenkk 
I tried your suggestion, added 
`console.log('ugrug_log_cmdId ', msg.zclMsg.cmdId);` at (https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/af.js#L693)
`console.log('ugrug_log_dispatchIncomingMsg` ', `dispatchIncomingMsg(): type: ${type}, msg: ${JSON.stringify(msg)}`); at (https://github.com/Koenkk/zigbee-shepherd/blob/3c5d7b4a3048eed19f71f618b7cc70bb866a2818/lib/components/af.js#L591)
unfortunatelly with no luck. I uploaded a video about the test: https://drive.google.com/file/d/1Wp4_vp-5ubdOLMyrYtTfSDjxNtirgh91/view?usp=sharing

I still have the guess the remote does not send any zigbee message as there could be some missing cluster bind, but probably I'm wrong. Without binding all button press causes lighting the red led.
Do you have any suggestion to check?

Koenkk, 2019-02-25:
Could you add the missing binds here: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L828 ?

Example of adding a missing cluster definition: https://github.com/Koenkk/zcl-id/commit/9c123fa4c4d48b8e87d27d9dd52a5048f79191f1

ugrug, 2019-02-25:
@Koenkk 
I tried this, no success.

I just started to digging information about the dimmer.
As I see can be paired directly with a bulb without hub, will try this also.

ugrug, 2019-02-25:
@Koenkk 

@Koenkk 
Great news: Added cluster definiton, modified device configuration and finally long-click working now.

I have to dig what cause exception now, but at least the dimmer send long press ON and OFF also.

This data received from the 0xfc00 philips manufacturer specific cluster:
```
Feb 25 21:31:16 cubie npm[32673]: 2019-02-25T20:31:16.418Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":64512,"srcaddr":1096,"srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"linkquality":65,"securityuse":0,"timestamp":2925319,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[29,11,16,131,0,1,0,0,48,0,33,0,0]}}
Feb 25 21:31:16 cubie npm[32673]: /usr/lib/zigbee2mqtt/node_modules/zcl-packet/lib/functional.js:33
Feb 25 21:31:16 cubie npm[32673]:         throw new Error('Unrecognized command');
Feb 25 21:31:16 cubie npm[32673]:         ^
Feb 25 21:31:16 cubie npm[32673]: Error: Unrecognized command
Feb 25 21:31:16 cubie npm[32673]:     at new FuncPayload (/usr/lib/zigbee2mqtt/node_modules/zcl-packet/lib/functional.js:33:15)
Feb 25 21:31:16 cubie npm[32673]:     at /usr/lib/zigbee2mqtt/node_modules/zcl-packet/lib/zcl.js:33:22
Feb 25 21:31:16 cubie npm[32673]:     at Dissolve.<anonymous> (/usr/lib/zigbee2mqtt/node_modules/zcl-packet/lib/zcl.js:121:9)
Feb 25 21:31:16 cubie npm[32673]:     at Object.onceWrapper (events.js:315:30)
Feb 25 21:31:16 cubie npm[32673]:     at emitOne (events.js:116:13)
Feb 25 21:31:16 cubie npm[32673]:     at Dissolve.emit (events.js:211:7)
Feb 25 21:31:16 cubie npm[32673]:     at Dissolve.<anonymous> (/usr/lib/zigbee2mqtt/node_modules/dissolve-chunks/index.js:73:29)
Feb 25 21:31:16 cubie npm[32673]:     at emitNone (events.js:106:13)
Feb 25 21:31:16 cubie npm[32673]:     at Dissolve.emit (events.js:208:7)
Feb 25 21:31:16 cubie npm[32673]:     at emitReadable_ (/usr/lib/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_readable.js:456:10)

```

james-fry, 2019-02-26:
Nice!
Would this be a breaking change for existing devices?
(I actually didnt pair mine up yet, so not a prob for me :) )

Koenkk, 2019-02-26:
@ugrug Great! In addition to changing zcl-id, you also need to make a similar update to zcl-packet, see https://github.com/Koenkk/zcl-packet/commit/09daf830a481bc20ef001da6181fe83464aa618b for an example.

Bearonic, 2019-02-26:
Guys, I don't have lots of time for all this smart home stuff at the moment, but that sounds great! Seems like I can finally abandon the Hue Bridge soon. :)
Great work so far, thanks for digging into this!

ugrug, 2019-02-26:
@Koenkk 
I fixed now the cluster and message definition, af get the messages.

I added logging `debug(`ugrug_log_dispatchIncomingMsg: type: ${type}, msg: ${JSON.stringify(msg)}`);` at (https://github.com/Koenkk/zigbee-shepherd/blob/3c5d7b4a3048eed19f71f618b7cc70bb866a2818/lib/components/af.js#L590)

The issue now comparing to standard up/down messages received as `zclIncomingMsg` type, this messages received as `incomingMsg `type so not handled now.

```
Feb 26 19:03:36 cubie npm[813]: 2019-02-26T18:03:36.228Z zigbee-shepherd:af ugrug_log_dispatchIncomingMsg: type: incomingMsg, msg: {"groupid":0,"clusterid":64512,"srcaddr":1096,"srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"linkquality":65,"securityuse":0,"timestamp":14425051,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[29,11,16,61,0,3,0,0,48,0,33,0,0]}}
Feb 26 19:03:36 cubie npm[813]: 2019-02-26T18:03:36.229Z zigbee-shepherd:af ugrug_log_dispatchIncomingMsg: return6
Feb 26 19:03:36 cubie npm[813]: 2019-02-26T18:03:36.237Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
Feb 26 19:03:36 cubie npm[813]: 2019-02-26T18:03:36.240Z zigbee-shepherd:af ugrug_log_dispatchIncomingMsg: type: zclIncomingMsg, msg: {"groupid":0,"clusterid":8,"srcaddr":1096,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":65,"securityuse":0,"timestamp":14425033,"transseqnumber":0,"len":7,"data":{"0":1,"1":60,"2":2,"3":1,"4":30,"5":9,"6":0},"zclMsg":{"frameCntl":{"frameType":1,"manufSpec":0,"direction":0,"disDefaultRsp":0},"manufCode":0,"seqNum":60,"cmdId":"step","payload":{"stepmode":1,"stepsize":30,"transtime":9}}}

```
Could you help me how to solve this?

Koenkk, 2019-02-26:
Could you post the debug log of ONLY the long click?

ugrug, 2019-02-27:
@Koenkk 
Sure, here are the messages:

OFF button down:
Feb 27 17:00:15 cubie npm[12270]: 2019-02-27T16:00:15.502Z zigbee-shepherd:af ugrug_log_dispatchIncomingMsg: type: incomingMsg, msg: {"groupid":0,"clusterid":64512,"srcaddr":1096,"srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"linkquality":68,"securityuse":0,"timestamp":10629645,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[29,11,16,87,0,4,0,0,48,0,33,0,0]}}

OFF button hold:
Feb 27 17:00:16 cubie npm[12270]: 2019-02-27T16:00:16.298Z zigbee-shepherd:af ugrug_log_dispatchIncomingMsg: type: incomingMsg, msg: {"groupid":0,"clusterid":64512,"srcaddr":1096,"srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"linkquality":68,"securityuse":0,"timestamp":10632144,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[29,11,16,89,0,4,0,0,48,1,33,8,0]}}
Feb 27 17:00:17 cubie npm[12270]: 2019-02-27T16:00:17.135Z zigbee-shepherd:af ugrug_log_dispatchIncomingMsg: type: incomingMsg, msg: {"groupid":0,"clusterid":64512,"srcaddr":1096,"srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"linkquality":68,"securityuse":0,"timestamp":10634645,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[29,11,16,90,0,4,0,0,48,1,33,16,0]}}
Feb 27 17:00:17 cubie npm[12270]: 2019-02-27T16:00:17.903Z zigbee-shepherd:af ugrug_log_dispatchIncomingMsg: type: incomingMsg, msg: {"groupid":0,"clusterid":64512,"srcaddr":1096,"srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"linkquality":70,"securityuse":0,"timestamp":10637144,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[29,11,16,91,0,4,0,0,48,1,33,24,0]}}

OFF button up:
Feb 27 17:00:18 cubie npm[12270]: 2019-02-27T16:00:18.581Z zigbee-shepherd:af ugrug_log_dispatchIncomingMsg: type: incomingMsg, msg: {"groupid":0,"clusterid":64512,"srcaddr":1096,"srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"linkquality":70,"securityuse":0,"timestamp":10639267,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[29,11,16,92,0,4,0,0,48,3,33,30,0]}}

Correlated to the previously sniffed values just add (29,11,16,92,0 zigbee header):
OFF Click: 04 00 00 30 00 21 00 00 -> dec: 4, 0, 0, 48, 0, 33, 0, 0
OFF Hold (1st) 04 00 00 30 01 21 08 00 -> dec: 4, 0, 0, 48, 1, 33, 8, 0
  (2nd) 04 00 00 30 01 21 10 00 -> dec 4, 0, 0, 48, 1, 33, 16, 0
  (3rd) 04 00 00 30 01 21 18 00 -> dec 4, 0, 0, 48, 1, 33, 24, 0
OFF Release: 04 00 00 30 03 21 2b 00 -> dec 4, 0, 0, 48, 3, 33, 43, 0

So it seems that bytes related to events that way [29,11,16,92,0,4,0,0,48,3,33,30,0]:
header: [29,11,16,92,0]
button: [4] (1: ON, 2: UP, 3: DOWN, 4: OFF]
unknown: [0, 0, 48]
type: [0] (0: press, 1: hold, 3: release
unknown: [33]
time from press [0]
unknown: [0]

Koenkk, 2019-02-28:
Can you replace what you add to zcl-packet with (fill in the params):
```js
"MYCOMMAND":{
                "params":[
                    {"value":"uint16"}
                ],
                "dir":1},
```

ugrug, 2019-03-01:
Great. The message decoded now. 👍 
```
Mar  1 18:35:44 cubie npm[31235]: 2019-03-01T17:35:44.370Z zigbee-shepherd:af ugrug_log_dispatchIncomingMsg: type: zclIncomingMsg, msg: {"groupid":0,"clusterid":64512,"srcaddr":1096,"srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"linkquality":86,"securityuse":0,"timestamp":4159004,"transseqnumber":0,"len":13,"data":{"0":29,"1":11,"2":16,"3":25,"4":0,"5":4,"6":0,"7":0,"8":48,"9":3,"10":33,"11":18,"12":0},"zclMsg":{"frameCntl":{"frameType":1,"manufSpec":1,"direction":1,"disDefaultRsp":1},"manufCode":4107,"seqNum":25,"cmdId":"eventNotification","payload":{"button":4,"unknown1":3145728,"type":3,"unknown2":0,"time":18}}}
Mar  1 18:35:44 cubie npm[31235]: 2019-03-01T17:35:44.374Z zigbee-shepherd:af ugrug_log_dispatchfinal: msg.zclMsg: {"frameCntl":{"frameType":1,"manufSpec":1,"direction":1,"disDefaultRsp":1},"manufCode":4107,"seqNum":25,"cmdId":"eventNotification","payload":{"button":4,"unknown1":3145728,"type":3,"unknown2":0,"time":18}
```
How to deliver it to HomeAssistant now?

ugrug, 2019-03-01:
@Koenkk 
The converter implemented now, the previous PR updated.
Everython working now on my configuration.
Also added the new command to zigbee-shepherd (https://github.com/Koenkk/zigbee-shepherd/pull/15).

The breaking change is only sending now 'on-press' and 'off-press' instead of 'on' and 'off'.
Added 'on-hold', 'on-hold-release', 'off-hold', 'off-hold-release' actions, and also includes duration attribute contains hold duration in seconds.

Its ready to merge and test. :)

james-fry, 2019-03-01:
Great job! Hopefully this will make it into the edge hassio addon soon :)

Koenkk, 2019-03-02:
Merged everything, will be available in edge in an hour.

ugrug, 2019-03-02:
Great, thanks!

Evildime, 2019-03-03:
@ugrug I updated to the latest dev and now all my dimmer does is create errors in the log. 
`No converter available for '324131092621' with cid 'genOnOff', type 'cmdOn' and data '{"cid":"genOnOff","data":{}}`

Same for type `cmdStep` and `cmdOffWithEffect`

I see that you removed those in your PR. Is there something else I need to do to make them work again? I already tried resetting and pairing again with both my Hue Dimmers.

thehaxxa, 2019-03-07:
`@ugrug I updated to the latest dev and now all my dimmer does is create errors in the log.
No converter available for '324131092621' with cid 'genOnOff', type 'cmdOn' and data '{"cid":"genOnOff","data":{}}`

Same issue as referenced here: https://github.com/Koenkk/zigbee2mqtt/issues/531

Koenkk, 2019-03-07:
@ugrug perhaps we should readd the old converters and only use the manuSpecificPhiliphs cluster for the long presses, would this be a solution?

ugrug, 2019-03-07:
@Koenkk 
My fault to remove converters seems to be the current initialisation makes the dimmer only operate with binding to genOnOff and genLevelCtrl clusters. (https://github.com/Koenkk/zigbee-shepherd-converters/pull/322)

I still looking for the similar solution as the dimmer works with the hue network.

I miss one command at the dimmer initialisation, probably causing that the dimmer messages not sent.
Could you help me how to send this message at the initialisasion to the dimmer?
[image]
After this, followed by a write attribute command:
[image]
And finally there is 2 bind command to the cluster 0xfc00 and 0x0001
[image]

Koenkk, 2019-03-07:
1. I don't know about the first one, this is not a ZCL command
2.  Could you expand the application support layer data (need to see the cluster)
3/4. these seems to be bind request from the remote **to** the coordinator.

Question: why do we need this? What's not working currently?

ugrug, 2019-03-07:
This is the different behaviour paired with hue and zigbee2mqtt:
- Hue: (https://drive.google.com/file/d/1_RpMoLkbaXEsYo8wBDy-N-LNPFVK_SJn/view?usp=sharing)
- Zigbee2mqtt: (https://drive.google.com/file/d/1LB0AZ9Cfg5jOlzFqpw4nz88v51eOqmh1/view?usp=sharing)
All button works including long-press, just holding on after about 2 sec dimmer starts to re-pair.
This not happens with hue.

ugrug, 2019-03-07:
I missed your comment about expanding the application support layer data. I replaced the screenshot now.

Koenkk, 2019-03-08:
Converters are not automatically updated, I have to bump the zigbee-shepherd-converters version zigbee2mqtt is using.

For the write, please try with:
```js
(cb) => ep2.foundation('genBasic', 'write', [{attrId: 0x0031, dataType: 0x19, attrData: 0x000B}], {manufSpec: 1, disDefaultRsp: 1, manufCode: 0x100B}, cb),
```

ugrug, 2019-03-08:
Great! This fixed the repairing issue I showed on the video.
I also implemented now multiple clicks (double, triple etc.)
Should I make new PR for this?

Koenkk, 2019-03-08:
Great! Yes please, then I can work on the zigbee2mqtt 1.2 release :) 

What is the procedure for existing users? Do they need to re-pair?

ugrug, 2019-03-08:
Finally the existing binding kept, just the new bind and the write request necessary to add.
I think repairing not necessary just the dimmer has to be awake for configuration.

From the previously used messages 'on' and 'off' modified to 'on-press' and 'off-press' to be consistent.

PR: https://github.com/Koenkk/zigbee-shepherd-converters/pull/323

Koenkk, 2019-03-08:
Integrated everything, can this be closed?

closed 2019-03-09 by Koenkk

jarrah31, 2019-03-21:
Sorry to comment on this after it was closed, but I can't see multiple clicks either in the Zigbee logs or on Home Assistant.

HA: 0.89.1
Zigbee2mqtt: 1.2.1
Firmware: 20190223

Single presses show action as 'on-press', and holding works with 'on-hold' and 'on-hold-release', but double or triple clicks just show 'on-press'.

I thought perhaps the 'counter' attribute would count the clicks, but I can't reliable get it to detect 2, let alone 3 click.

Here the logs shows me pressing twice in quick succession, but no double click detected:
```
Mar 21 19:13:07 raspberrypi npm[7766]:   zigbee2mqtt:info 3/21/2019, 7:13:07 PM MQTT publish: topic 'zigbee2mqtt/hue_dimmer_ensuite', payload '{"battery":94.5,"linkquality":42,"last_seen":"2019-03-21T19:13:07.070Z","action":"on-press","duration":0,"counter":1}'
Mar 21 19:13:07 raspberrypi npm[7766]:   zigbee2mqtt:info 3/21/2019, 7:13:07 PM MQTT publish: topic 'zigbee2mqtt/hue_dimmer_ensuite', payload '{"battery":94.5,"linkquality":42,"last_seen":"2019-03-21T19:13:07.329Z","action":"on-press","duration":0,"counter":1}'
```
HA State shows the following:
[image]

ugrug, 2019-03-22:
@jarrah31 
The default timeout is 0.25 (means 250 ms), what requires really fast clicking. 
In my instance its working right, ok for 2-3 clicks, if you are quick.
Please try to increase `multiple_press_timeout`,

jarrah31, 2019-03-22:
@ugrug ah thanks! 

Where would I set multiple_press_timeout as searching for this option in this repo and googling it with zigbee2mqtt or home assistant gives no answers?

ugrug, 2019-03-22:
@jarrah31 
can be configured in device settings:
`/usr/lib/zigbee2mqtt/data/configuration.yaml `
```
devices:
  '0x00xxxxxxxxxxxxxx':
    friendly_name: yyyyyyyyyyy
    multiple_press_timeout: 1
    retain: true
```

james-fry, 2019-03-23:
IMO 250ms is just perfect.
I have also the old round xiaomi buttons and the hue feels like the for multi-click.
I also have the new square xiaomi buttons and they have a longer double-click period and it feels wrong.

jarrah31, 2019-03-23:
Thank you @ugrug

james-fry, 2019-04-08:
Just noticed that the device support docs is out of date for this dimmer now.

It says

> 324131092621 | Philips Hue dimmer switch (on/off)

Should be updated to indicate it supports also long click, long click release, and count of clicks, as well as brightness value.

Koenkk, 2019-04-08:
@james-fry that can be done by updating: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L898

hgelpke, 2019-05-03:
How does one update that?

Koenkk, 2019-05-05:
@hgelpke by making a pull request.

I've updated the description, will be updated in the docs once zigbee2mqtt 1.4 is released.