Koenkk/zigbee2mqtt #586: What's the power usage difference when using this vs xiaomi smart gateway 2?

thibaultmol, 2018-11-11:
I want my smart home to have as little as possible standby power, so I'm curious what the power usage differences between the two

Koenkk, 2018-11-11:
It depends on your setup, when running zigbee2mqtt on a raspberry pi it is very power efficient. Don't know about the consumption of the xiaomi gateway though.

stoinov, 2018-11-13:
It's hard to compare them without any specifics. I suspect both asetups will use comparable power for running the actual Zigbee network. The difference will come from the auxiliary hardware.

- If you're using Xiaomi complementary to another device like RPi, then it's better to use the USB adapter directly on the Pi. as you'll definitely save on power.
- If you're replacing Xiaomi with RPi then I guess your usage might go slightly up, but you'll gain much more verasitility, that can potentially help you save power on places you were not able to with only the smart gateway.

Overall I suggest you switch without hesitating in that regard. You'll be definitely making the correct choice :)

thibaultmol, 2018-11-13:
Ah yeah, good point(s) @stoinov ! Thanks

closed 2018-11-13 by thibaultmol