Koenkk/zigbee2mqtt #587: Salus radiator thermostat

ciB89, 2018-11-11:
Hey,

I just stumpled across a company called "Salus" ([Website])(https://salus-controls.com) which create a bunch of different Zigbee devices. They offer quite a cheap solution for radiators since they basically don't use a gateway (they sell their zigbee stick for 9,99€ and then sell another thermostat for around 15-30€)

Now their smart [radiators](https://www.conrad.de/de/salus-controls-it600-heizkoerperthermostat-trv10rfm-1484561.html) only cost about 24€ - and I'm really curious about looking into them. Is there anyone who has tested something with Salus before? I know from the eCozy thermostat #129 that adding a new device that basically has no functionality/buttons on its own to create messages is quite hard to add.

sebastianheierhoff, 2018-11-14:
Thank's for the hint. Looks like a great alternative to the eCozy thermostats. In theory, once the eCozy is implemented, the same converters should be applicable to the Salus, as well (assuming the adhere to the zigbee standard).

Koenkk, 2018-12-10:
@toha like in: https://github.com/zigbeer/zcl-id/blob/master/definitions/cluster_defs.json#L65?

mtzro2003, 2019-01-01:
@toha 
Any new regarding these Salus TRVs?

P.S. Happy New Year everybody!!! :smile:

ctandi, 2019-01-09:
We (@jamct, @merlinschumacher and I) requested the TRV10RFM as a testing sample from Salus and just got our package. Depending on who has time, we will try to get the thermostat working. As far as I can see, a lot of the general HVAC stuff has already implemented. We'll see how we can move forward with that.

ctandi, 2019-01-09:
> @toha like in: https://github.com/zigbeer/zcl-id/blob/master/definitions/cluster_defs.json#L65?

I gave that a try and got the standard output:
"Device with modelID 'SAL6EM1' is not supported." 

It's a little late, so I'm stopping here and will continue tomorrow. Besides that, I just can't seem to make out how to move on from here, regarding the cluster-definitions for the device. I have written converters before, but not as deeply as this is into the ZigBee protocol. I haven't found any further documentation so far :( Where can I find the log as detailed as the one that @toha provided? @Koenkk could you provide us with a more detailed explanation on how to proceed in this case? We are in contact with Salus and I feel they are pretty open to answering our questions. We also have their gateway and a separate sniffer CC2531. 

Thanks!

toha, 2019-01-09:
@ctandi You can get detailed output by starting zigbee2mqtt with `DEBUG=* npm start`.

Ich hope to get some time at the weekend to continue testing the device. What I found ist that TRV10RFM drains batteries in no time. Had to replace the batteries two times in three days.

pwrangles, 2019-01-14:
Hey all, my dad just brought 8 of there TRV, reply keen to get them working for him to!

ctandi, 2019-01-14:
Great to have someone else working on that as well. I haven't had time to play around last weekend. I might give it a try this afternoon.

It's weird that your batteries drain so quickly @toha. A Salus technician told me that they expect an average lifetime of one year.

pwrangles, 2019-01-14:
I have Hass.io and haven't been able to even map the custom device so not sure I will be much help. This is my first Zigbee2MQTT project.. But I will continue to have a play later today

pwrangles, 2019-01-14:
Im just getting this:

```
  zigbee2mqtt:info 2019-1-14 15:17:54 Logging to directory: '/share/zigbee2mqtt/log/2019-01-14.15-17-54'
/zigbee2mqtt-1.0.1/node_modules/zigbee-shepherd-converters/devices.js:3
    model: 'TRV10RFM', // Vendor model number, look on the device for a model number
         ^
SyntaxError: Unexpected token :
    at createScript (vm.js:80:10)
    at Object.runInThisContext (vm.js:139:10)
    at Module._compile (module.js:617:28)
    at Object.Module._extensions..js (module.js:664:10)
    at Module.load (module.js:566:32)
    at tryModuleLoad (module.js:506:12)
    at Function.Module._load (module.js:498:3)
    at Module.require (module.js:597:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/zigbee2mqtt-1.0.1/node_modules/zigbee-shepherd-converters/index.js:3:17)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.0.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.0.1 start script.
npm 
ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-01-14T15_17_55_085Z-debug.log
2019-01-14T15:17:55: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2019-01-14T15:17:55: PM2 log: App [npm:0] starting in -fork mode-
2019-01-14T15:17:55: PM2 log: App [npm:0] online
> zigbee2mqtt@1.0.1 start /zigbee2mqtt-1.0.1
> node index.js
```

pwrangles, 2019-01-16:
I worked out my error above. But haven't managed to get much further. @ctandi, did you manage to get any further on this?

pwrangles, 2019-01-23:
Hey @ctandi, Any luck on working out these TRV's?

pwrangles, 2019-01-24:
> Hi, sorry for not responding, we've been quite busy here (https://www.heise.de/newsticker/meldung/No-Name-Hausautomation-Luecke-erlaubt-leichten-Firmware-Upload-4284783.html). I will try to make some time to get the thermostats going next week.

That would be awesome! Thanks!

sebastianheierhoff, 2019-02-02:
Happy to give it a try if someone wants to send me his/her thermostat. Else I can definitely recommend the ecozy. They are quite cheap on eBay (Kleinanzeigen), as well.

ciB89, 2019-02-02:
I can send you one. Send me a Mail: Ciaran.baulig Gmail

9600, 2019-02-11:
Watching this with interest as I need to invest in a set of TRVs and the cheap eCozy ones seem to have sold.

DrDale, 2019-02-16:
Me too, Good afternoon @sebastianheierhoff - Have you managed to make any progress with these TRV's

sebastianheierhoff, 2019-02-16:
Working on it, but unfortunately I am running into another issue at the moment (see https://github.com/Koenkk/zigbee2mqtt/issues/1102)

sebastianheierhoff, 2019-02-18:
All those wanting to contribute can check out: https://github.com/sebastianheierhoff/zigbee-shepherd-converters/tree/salustrv10rfm; https://github.com/sebastianheierhoff/zcl-packet/tree/salustrv10rfm and https://github.com/sebastianheierhoff/zcl-id/tree/salustrv10rfm. Work in progress.

ctandi, 2019-02-18:
Sorry for not responding for so long. I had just way too much work to do, to start working on it. But I talked to Salus and one thing that might be of interest is that the thermostat doesn't have a temperatur sensor. It's just there to open an close the valve. I suspect that's why they say that you need another wall thermostat.

sebastianheierhoff, 2019-02-18:
Interesting. They seem to be using quite a lot of manufactur specific messages/ cluster (see also https://github.com/Koenkk/zigbee2mqtt/issues/917). To speed up implementation, it would be really nice if they'd be willing to tell us about these messages (structure, content, etc.)

b4dpxl, 2019-02-18:
> Sorry for not responding for so long. I had just way too much work to do, to start working on it. But I talked to Salus and one thing that might be of interest is that the thermostat doesn't have a [...]

I don’t know about other people, but that would actually be preferable for me. I have thermometers in each room linked to HASS that are much better placed than by the TRV’s. That’s part of what’s put me off other TRV-based systems.

DrDale, 2019-03-09:
Good morning all, any progress guys?

sebastianheierhoff, 2019-03-20:
I didn't really have time to continue working on this. Is anyone else still trying?

closed 2019-05-26 by stale[bot]

zhapwey, 2019-09-20:
hello:

We managed to do something useful with this device?

BR

pwrangles, 2019-09-20:
I have up and went with some zwave devices.

ministryofsillywalks, 2019-10-04:
I'm also interested if someone got this working as this looks to be the cheapest zigbee thermostat around by far.

ciB89, 2019-11-08:
There are some on Amazon for 16 Euros.. it's a shame that we did not get them working :(

abeorch, 2020-05-01:
Ditto. They have a nicely designed ZigBee thermostat that I would be willing to lend to a developer if they wanted to investigate further

ciB89, 2020-05-01:
I have given mine to @sebastianheierhoff but unfortunately he did not continue working on the project.