Koenkk/zigbee2mqtt #588: Support: Inne SP 120 smart plug

tunip, 2018-11-11:
I have a Inne SP 120 Smart Plug (https://shop.innrlighting.com/en/shop/113/smart-plug) which is a unknown device.

I started with the "How to support new devices" Wiki, but after adding the device in devices.js, I need more help. If I started zigbee2mqtt with `DEBUG=zigbee-shepherd* npm start`and press the button on the smart plug, but nothing happens in the console (no message in the log).

zigbee2mqtt@0.1.8

devices.js
```
    {
        zigbeeModel: ['SP 120'],
        model: 'SP 120',
        vendor: 'Innr',
        description: 'smart plug',
        supports: 'on/off, power measurement',
        fromZigbee: [],
        toZigbee: [],
    },
```

Tried also to add a converter in fromZigbee.js for more messages.
```
    innr_plug_state: {
        cid: 'genBasic',
        type: 'attReport',
        convert: (model, msg, publish, options) => {
            console.log(msg.data)
        },
    },
```

Then added this to the device in line fromZigbee: [fz. innr_plug_state]. Also no message in the (debug) log.

What could I do next? Thx

tunip, 2018-11-11:
Added to the device
```
      configure: (ieeeAddr, shepherd, coordinator, callback) => {
                const device = shepherd.find(ieeeAddr, 1);
                const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
                const actions = [
                        (cb) => device.bind('genOnOff', coordinator, cb),
                        (cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
                        (cb) => device.report('haElectricalMeasurement', 'rmsVoltage', 10, 1000, 1, cb),
                        (cb) => device.report('haElectricalMeasurement', 'rmsCurrent', 10, 1000, 1, cb),
                        (cb) => device.report('haElectricalMeasurement', 'activePower', 10, 1000, 1, cb),
                ];

                execute(device, actions, callback);
```

And now I receive messages in the console if I press the button on the plug.
```
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +29s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +10ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +3ms
  zigbee2mqtt:debug 2018-11-11 22:39:03 Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'SP 120' (0x00158d000217be5c)
  zigbee2mqtt:warn 2018-11-11 22:39:03 No converter available for 'SP 120' with cid 'genOnOff', type 'attReport' and data '{"cid":"genOnOff","data":{"onOff":0}}'
  zigbee2mqtt:warn 2018-11-11 22:39:03 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:debug 2018-11-11 22:39:03 Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'SP 120' (0x00158d000217be5c)
  zigbee2mqtt:warn 2018-11-11 22:39:03 No converter available for 'SP 120' with cid 'genOnOff', type 'devChange' and data '{"cid":"genOnOff","data":{"onOff":0}}'
  zigbee2mqtt:warn 2018-11-11 22:39:03 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +3s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +9ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +2ms
  zigbee2mqtt:debug 2018-11-11 22:39:06 Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'SP 120' (0x00158d000217be5c)
  zigbee2mqtt:warn 2018-11-11 22:39:06 No converter available for 'SP 120' with cid 'genOnOff', type 'attReport' and data '{"cid":"genOnOff","data":{"onOff":1}}'
  zigbee2mqtt:warn 2018-11-11 22:39:06 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:debug 2018-11-11 22:39:06 Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'SP 120' (0x00158d000217be5c)
  zigbee2mqtt:warn 2018-11-11 22:39:06 No converter available for 'SP 120' with cid 'genOnOff', type 'devChange' and data '{"cid":"genOnOff","data":{"onOff":1}}'
  zigbee2mqtt:warn 2018-11-11 22:39:06 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
```

Koenkk, 2018-11-12:
Can you try with

```js
    {
        zigbeeModel: ['SP 120'],
        model: 'SP 120',
        vendor: 'Innr',
        description: 'Smart Plug',
        supports: 'on/off, power measurement',
        fromZigbee: [fz.generic_state, fz.ignore_onoff_change, fz.ignore_electrical_change, fz.SP120_power],
        toZigbee: [tz.onoff],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
                const device = shepherd.find(ieeeAddr, 1);
                const cfgOnOff = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
                const cfgVoltage = {direction: 0, attrId: 1285, dataType: 33, minRepIntval: 10, maxRepIntval: 1000, repChange: 1};
                const cfgCurrent = {direction: 0, attrId: 1288, dataType: 33, minRepIntval: 10, maxRepIntval: 1000, repChange: 1};
                const cfgPower = {direction: 0, attrId: 1291, dataType: 41, minRepIntval: 10, maxRepIntval: 1000, repChange: 1};

                const actions = [
                    (cb) => device.bind('genOnOff', coordinator, cb),
                    (cb) => device.foundation('genOnOff', 'configReport', [cfgOnOff], foundationCfg, cb),
                    (cb) => device.foundation('haElectricalMeasurement', 'configReport', [cfgVoltage], foundationCfg, cb),
                    (cb) => device.foundation('haElectricalMeasurement', 'configReport', [cfgCurrent], foundationCfg, cb),
                    (cb) => device.foundation('haElectricalMeasurement', 'configReport', [cfgPower], foundationCfg, cb),
                ];
                execute(device, actions, callback);
        },
    },
```

tunip, 2018-11-12:
Thx @Koenkk for your replay.

No errors after start of zigbee2mqtt:
```
  zigbee2mqtt:info 2018-11-12 13:09:50 MQTT publish, topic: 'zigbee2mqtt/smart_plug_office', payload: '{"state":"ON","linkquality":89}'
  zigbee-shepherd:msgHdlr IND <-- ZDO:permitJoinInd +59ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:mgmtPermitJoinRsp +0ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:bindRsp +52ms
  zigbee-shepherd-converters:devices Configured '(cb) => device.bind('genOnOff', coordinator, cb)' with result 'OK' +0ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 1 +315ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +15ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +23ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 1 +1ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +12ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +10ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +1ms
  zigbee-shepherd-converters:devices Configured '(cb) => device.foundation('genOnOff', 'configReport', [cfgOnOff], foundationCfg, cb)' with result 'OK' +377ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +73ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +8ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +1ms
  zigbee2mqtt:debug 2018-11-12 13:09:50 Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'SP 120' (0x00158d000217be5c)
  zigbee2mqtt:info 2018-11-12 13:09:50 MQTT publish, topic: 'zigbee2mqtt/smart_plug_office', payload: '{"state":"ON","linkquality":89}'
  zigbee2mqtt:debug 2018-11-12 13:09:50 Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'SP 120' (0x00158d000217be5c)
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 2 +228ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +20ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +22ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 2 +1ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +21ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +6ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +1ms
  zigbee-shepherd-converters:devices Configured '(cb) => device.foundation('haElectricalMeasurement', 'configReport', [cfgVoltage], foundationCfg, cb)' with result 'OK' +381ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 3 +306ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +20ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +21ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 3 +1ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +9ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +6ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +2ms
  zigbee-shepherd-converters:devices Configured '(cb) => device.foundation('haElectricalMeasurement', 'configReport', [cfgCurrent], foundationCfg, cb)' with result 'OK' +365ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 4 +307ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +16ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +22ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 4 +1ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +26ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +6ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +2ms
  zigbee-shepherd-converters:devices Configured '(cb) => device.foundation('haElectricalMeasurement', 'configReport', [cfgPower], foundationCfg, cb)' with result 'OK' +380ms
  zigbee2mqtt:info 2018-11-12 13:09:52 Succesfully configured smart_plug_office 0x00158d000217be5c
  zigbee2mqtt:debug 2018-11-12 13:14:46 Saving state to file /opt/zigbee2mqtt/data/state.json
```

How often would the electrical measurement be reported in the console? No output after 5 minutes.

Koenkk, 2018-11-12:
I expect it to be reported when the consumption changes. Try attaching a bulb to it.

tunip, 2018-11-12:
No output about the measurement if I attach a bulb (40W) to the plug.

tunip, 2018-11-13:
Another point is, if I publish the networkmap, the SP 120 is the only device which is in status offline (on/off works):
```
digraph G {
node[shape=record];
  "0x00124b0012023339" [label="{0x00124b0012023339|Coordinator|No model information available|online}"];
  "0x00124b0012023339" -> "0x0017880100c18ca0" [label="55"]
  "0x00158d000236b3a9" [label="{motion_sensor_1_office|EndDevice|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|online}"];
  "0x00158d000236b3a9" -> "0x00124b0012023339" [label="82"]
  "0x00158d000237202f" [label="{motion_sensor_2_office|EndDevice|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|online}"];
  "0x00158d000237202f" -> "0x00124b0012023339" [label="94"]
  "0x00158d00023eb1a9" [label="{window_sensor_office|EndDevice|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|online}"];
  "0x00158d00023eb1a9" -> "0x00124b0012023339" [label="126"]
  "0x00158d0002a4edc9" [label="{shock_sensor_office|EndDevice|Xiaomi Aqara vibration sensor (DJT11LM)|online}"];
  "0x00158d0002a4edc9" -> "0x00124b0012023339" [label="103"]
  "0x00158d000217be5c" [label="{smart_plug_office|Router|Innr Smart Plug (SP 120)|offline}"];
  "0x00158d000217be5c" -> "0x0017880100c18ca0" [label="33"]
  "0x00158d00023761f7" [label="{0x00158d00023761f7|EndDevice|No model information available|online}"];
  "0x00158d00023761f7" -> "0x00158d000217be5c" [label="109"]
  "0x00158d000212fd57" [label="{0x00158d000212fd57|EndDevice|Xiaomi Aqara wireless switch (WXKG11LM)|online}"];
  "0x00158d000212fd57" -> "0x00158d000217be5c" [label="111"]
  "0x0017880100c18ca0" [label="{0x0017880100c18ca0|Router|Philips Hue Bloom (7299760PH)|online}"];
  "0x0017880100c18ca0" -> "0x00158d000217be5c" [label="29"]
}
```

tunip, 2018-11-15:
Found some hints in the deCONZ issues. Maybe it helps.

Unlike the other smart plugs, the innr SP 120 uses the ZLL profile.
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/432#issuecomment-395757777

Adds SP 120 in deCONZ
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/630

Koenkk, 2018-11-15:
No luck yet.

Could you provide all entries of `0x00158d000217be5c` from `database.db`?

tunip, 2018-11-15:
I repaired it, thats why it is now id12 instead id7 from the first post.

```
{"id":12,"type":"Router","ieeeAddr":"0x00158d000217be5c","nwkAddr":41726,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"SP 120","epList":[1,2],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":49246,"epId":1,"devId":16,"inClusterList":[0,3,4,5,6,8,10,1794,2820],"outClusterList":[3,10,25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"10":{},"zclVersion":2,"appVersion":1,"stackVersion":2,"hwVersion":1,"manufacturerName":"innr","modelId":"SP 120","dateCode":"20171027-100","powerSource":1,"swBuildId":"2.0"}},"genIdentify":{"dir":{"value":3},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":0}},"genScenes":{"dir":{"value":1},"attrs":{"count":1,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0,"lastCfgBy":"0xffffffffffffffff"}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"currentLevel":254,"remainingTime":0,"onOffTransitionTime":0}},"genTime":{"dir":{"value":3},"attrs":{"time":0,"timeStatus":0}},"genOta":{"dir":{"value":2},"attrs":{}},"seMetering":{"dir":{"value":1},"attrs":{"currentSummDelivered":{"0":0,"1":0},"status":0,"unitOfMeasure":0,"summaFormatting":42,"meteringDeviceType":0}},"haElectricalMeasurement":{"dir":{"value":1},"attrs":{"measurementType":0,"rmsVoltage":0,"rmsCurrent":0,"activePower":0}}}},"2":{"profId":49246,"epId":2,"devId":4096,"inClusterList":[4096],"outClusterList":[],"clusters":{"lightLink":{"dir":{"value":1},"attrs":{}}}}},"_id":"iwDH6xwrk6op6ah0"}
{"id":12,"type":"Router","ieeeAddr":"0x00158d000217be5c","nwkAddr":41726,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"SP 120","epList":[1,2],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":49246,"epId":1,"devId":16,"inClusterList":[0,3,4,5,6,8,10,1794,2820],"outClusterList":[3,10,25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"10":{},"zclVersion":2,"appVersion":1,"stackVersion":2,"hwVersion":1,"manufacturerName":"innr","modelId":"SP 120","dateCode":"20171027-100","powerSource":1,"swBuildId":"2.0"}},"genIdentify":{"dir":{"value":3},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":0}},"genScenes":{"dir":{"value":1},"attrs":{"count":1,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0,"lastCfgBy":"0xffffffffffffffff"}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"currentLevel":254,"remainingTime":0,"onOffTransitionTime":0}},"genTime":{"dir":{"value":3},"attrs":{"time":0,"timeStatus":0}},"genOta":{"dir":{"value":2},"attrs":{}},"seMetering":{"dir":{"value":1},"attrs":{"currentSummDelivered":{"0":0,"1":0},"status":0,"unitOfMeasure":0,"summaFormatting":42,"meteringDeviceType":0}},"haElectricalMeasurement":{"dir":{"value":1},"attrs":{"measurementType":0,"rmsVoltage":0,"rmsCurrent":0,"activePower":0}}}},"2":{"profId":49246,"epId":2,"devId":4096,"inClusterList":[4096],"outClusterList":[],"clusters":{"lightLink":{"dir":{"value":1},"attrs":{}}}}},"_id":"iwDH6xwrk6op6ah0"}
```

Koenkk, 2018-11-16:
Ok, we need to extend the logging a bit.

Can you change https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/af.js#L576 to `debug(`dispatchIncomingMsg(): type: ${type}, msg: ${JSON.stringify(msg)}`);`

tunip, 2018-11-16:
Extended Log:
[Log](https://pastebin.com/nKuKJfNC)

Koenkk, 2018-11-16:
If I understand it right. `buro_bloom` is connected to the SP 120 plug.  The log you provide after you turn on `buro_bloom` is  very short (2018-11-16 11:53:18).

Could you provide a longer log? (5 mins)

tunip, 2018-11-16:
Yes thats right, `buro_bloom` is connected to the SP 120 plug for debugging.

Hopefully 30 mins are ok:
[Log](https://pastebin.com/FFaviD90)

Koenkk, 2018-11-16:
No reports yet :(

Ive also bought this plug so will try adding support for it once i receive it.

Koenkk, 2018-12-14:
Took 28 days to arrive (😐), however it's supported now! (in the dev branch).

closed 2018-12-14 by Koenkk

hansblafoo, 2018-12-25:
Does it really support power measurement? I did not find a website (not even from innr) which lists this feature.

tunip, 2018-12-26:
Of course 
2018-12-26 10:05:37 - info: MQTT publish: topic 'zigbee2mqtt/christmas_tree',payload '{"state":"ON","power":42,"linkquality":63,"current":0.19,"voltage":235}'

Koenkk, 2018-12-26:
@hansblafoo yes! I use it to automatically switch of the plug once the kettle has finished.

nanosonde, 2019-01-09:
Can you recommend this plug as a Zigbee router?

Koenkk, 2019-01-09:
Yes

TribuneX, 2019-09-21:
My innr plugs do seem to report their state via zigbee once I switch them on with the button. Can someone confirm that this is the case?

gally789, 2019-12-16:
Mine do the same.. if someone switches the plug on at the button on the plug I get no status message in zigbee2mqtt. Should this device report status?

> My innr plugs do not seem to report their state via zigbee once I switch them on with the button. Can someone confirm that this is the case?

Koenkk, 2019-12-16:
@gally789 there was a bug in zigbee2mqtt 1.7.1 causing this.

To fix:
- Update to zigbee2mqtt 1.8.0
- Remove the device through https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove
- Repair device

If it still doesn't work, please provide your `database.db`

TribuneX, 2019-12-16:
Thanks. Upgrading and adding the devices again to my network fixed it for me.

runningman84, 2020-01-13:
@Koenkk is the reporting interval configurable? From the code it looks like it is defined in your code...

Koenkk, 2020-01-14:
@runningman84 it's not configurable indeed (but the default should be a good value)

runningman84, 2020-01-14:
It looks like my zigbee network is spammed with messages, it would be great if we could specify the interval. I think sometimes my xiaomi sensors fail to send messages due to the high zigbee network load...

Koenkk, 2020-01-16:
You can try adjusting the reporting interval in the functions (https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L2642) to find a new good default value. Note that every time you change it you need to bump the configure key by one (https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L2637)

runningman84, 2020-01-16:
what would be needed to make this setting as a config option in the configuration.yaml? In this case I could change this value without rebuilding the whole image...

IcedEarth, 2020-01-19:
Indeed, I experience similar issues with status reports flooding my FHEM server putting a huge load on my zigbee network making other sensor less responsible.

monotonus, 2020-03-22:
> You can try adjusting the reporting interval in the functions (https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L2642) to find a new good default value. Note that every [...]

Could you please giva an example how to do this? At the moment there is no reporting interval listed for the SP120.

I also think adding the interval to the configuration.yaml would be good for most of the users. 5 seconds may still be too much for some environments. So these 5 seconds are configured in the devices itself? Because in the syslog I only see what is pushed to the mqtt broker?

I also found an interesting post here, but this seems not to work: https://zigbee2mqtt.discourse.group/t/configure-attribute-reporting-for-innr-sp120-or-any-zigbee-device/457

Webunity, 2020-05-19:
@Koenkk can you also configure this plug to have a default 'on' state? i want it to measure the power from my washing machine but power cuts should not put the plug into an 'off' state.

Koenkk, 2020-05-19:
@Webunity I'm not aware this is possible with this plug. Maybe someone can check if it's possible via the original gateway, in that case we can also add it.

Webunity, 2020-05-20:
i'll check with deconz. In the meantime as an alternative i could also poll the switch and turn it on in case it was off.

Appelg, 2020-08-17:
@monotonus did you ever figure out how to change the reporting interval?
my "voltage" fluctuates between 232 & 236 resulting in 5-10 updates per minute(!).

monotonus, 2020-08-17:
@Appelg unfortunately no. In the meanwhile I sold the device again and I'm using now the Xiaomi Smart Plug which doesn't update that often.

Appelg, 2020-08-17:
@Koenkk sorry for bumping this old issue again...

I changed the devices.js to this

`await configureReporting.rmsVoltage(endpoint, { change: 10});`

Will that actually tell the device to only report changes of 10 over zigbee, or will it only limit these changes against MQTT?

Why does some devices seem to have a more advanced "config"?

```
           await endpoint.configureReporting('hvacThermostat', [{
                attribute: {ID: 0x4008, type: 34},
                minimumReportInterval: 0,
                maximumReportInterval: repInterval.HOUR,
                reportableChange: 1,
            }], options);
```

Koenkk, 2020-08-17:
> Will that actually tell the device to only report changes of 10 over zigbee, or will it only limit these changes against MQTT?

Zigbee

> Why does some devices seem to have a more advanced "config"?

It depends on the granularity of the reported values, this differs per device, the main goal is to prevent spamming. Some devices already have a correct reporting interval by default so additional configuration is not necessary.

> Also: Is it possible to configure this via "bridge/config/device_options"? Is there a way to do a "get" to see the structure of the message to send to config?

This is not configurable

Appelg, 2020-08-18:
@Koenkk  thanks for the feedback. One final question:
DoConz reported a "consumption" value with total kWh used, this is all I get now:

```
{
  "current": 0.61,
  "last_seen": "2020-08-18T11:18:58+02:00",
  "linkquality": 0,
  "power": 146,
  "state": "ON",
  "voltage": 234
}
```

Is consumption not supported?

Koenkk, 2020-08-18:
@Appelg currently not, can you try changing the definition in `devices.js` to:

```js
{
    zigbeeModel: ['SP 120'],
    model: 'SP 120',
    vendor: 'Innr',
    description: 'Smart plug',
    supports: 'on/off, power measurement',
    fromZigbee: [fz.electrical_measurement_power, fz.on_off, fz.ignore_genLevelCtrl_report, fz.metering_power],
    toZigbee: [tz.on_off],
    meta: {configureKey: 5},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
        await configureReporting.onOff(endpoint);
        // Gives UNSUPPORTED_ATTRIBUTE on readEletricalMeasurementPowerConverterAttributes.
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
            acCurrentDivisor: 1000,
            acCurrentMultiplier: 1,
        });
        await configureReporting.activePower(endpoint);
        await configureReporting.rmsCurrent(endpoint);
        await configureReporting.rmsVoltage(endpoint);
        // Gives UNSUPPORTED_ATTRIBUTE on readMeteringPowerConverterAttributes.
        endpoint.saveClusterAttributeKeyValue('seMetering', {
            multiplier: 1,
            divisor: 1,
        });
        await configureReporting.currentSummDelivered(endpoint);
    },
},
```

Might take some time for the `energy` (= consumption ) values to show up.

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html how to edit `devices.js`.

Appelg, 2020-08-19:
@Koenkk that did work, thanks! Took a couple of hours before the energy showed up, if anyone else tries this.

One problem tho: After this change, I can't add more Innr SP 120, it fails during "interview".
I guess the workaround is to revert code change, add them, do code change again.
Problem is tho that we cannot merge this code until that problem is solved I guess....

This is the error I got when trying to add another device with the above code change:

```
Zigbee2MQTT:error 2020-08-19 10:58:32: Failed to configure 'zigbee2mqtt/outlet/innr120/sovrum', 
attempt 2 (Error: ConfigureReporting 0x**********d92d/1 haElectricalMeasurement([{"attribute":"rmsCurrent","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":1}], 
{"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Status 'undefined')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:140:23)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:348:26)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58))
```

Koenkk, 2020-08-19:
@Appelg when you revert does it work again?

Appelg, 2020-08-20:
> @Appelg when you revert does it work again?

@Koenkk yes.

This is what I did:
1) revert to standard code
2) successfully paired 3 more innr sockets
3) changed the code to the one you provided above, with "energy"
4) all 4 devices reconfigured successfully on restart

Koenkk, 2020-08-20:
So pairing with the new code still fails?

Appelg, 2020-08-27:
@Koenkk sorry for getting back this late, after the PR. I finally had time to set up my Grafana graphs, and look at the data...
It seem's like I need to multiply the value by 10 in order to get Wh. 

Is there a way to figure out what unit the device sends?

I base this on two things:
1) The values are a factor 10 less than what DeConz sent me
2) These logs:

16:53 - 156 enery, 70 power
17:54 - 164 enery, 76 power

So an hour gave a difference of 8.
70 W * 1 hr = 70Wh
But the power sometimes has readings over 100 W for that time, so 8 seems resonable.

Therefore a multiplication by 10 seems what I need in order to get Wh. 
What unit is the other sockets reporing in? Maybe kWh is what is standard?  Then we seem to need a 10000 multiplier instead.

Koenkk, 2020-08-27:
@Appelg energy values should be in kWh not wH so we should divide by 1000 instead of multiplying by 10. This can be done by changing the `divisor` to `1000` (https://github.com/Koenkk/zigbee-herdsman-converters/blob/39bf1c91569fe3aa2e2ceff101203a96ab027185/devices.js#L4452) and bumping the `configureKey` by one `so `6` (https://github.com/Koenkk/zigbee-herdsman-converters/blob/39bf1c91569fe3aa2e2ceff101203a96ab027185/devices.js#L4436)

Appelg, 2020-08-27:
> @Appelg energy values should be in kWh not wH so we should divide by 1000 instead of multiplying by 10. This can be done by changing the `divisor` to `1000` [...]

Yeah, divide, of course. But with 100?
Since I need to multiply with 10 to get Wh. It seems like it reports in 0.1 Wh...
It do seem strange, why report in 1/10 of Wh?

But is there a way to verify? Does the zigbee protocol have a way to ask the device? 

Or is the only way to contact the manufacturer and ask for a spec?

So all the other divisor/multipler in the devices.js has been figured out empirically?

Koenkk, 2020-08-27:
Normally the `divisor` and `multiplier` values are read from the device (e.g. https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L55 / https://github.com/Koenkk/zigbee-herdsman-converters/blob/39bf1c91569fe3aa2e2ceff101203a96ab027185/devices.js#L3151 ) but unfortunately the SP 120 does not support this.

ScumbagSteve, 2022-07-17:
I have the same problem as described above, the SP120 is flodding my network every 5 seconds, which results in commands not going through. I have adjusted the configuration file for the SP 120 (.../iobroker/data/node_modules/zigbee-herdsman-converters/devices/innr.js) and added the "{change: XX}" parameter behind the awaits. However this did not fix my problem (see [https://github.com/ioBroker/ioBroker.zigbee/issues/1483](link).
Any hints what to do?