Koenkk/zigbee2mqtt #589: MQTT whildcard character support

pkedvessy, 2018-11-11:
It would be greate to support wildcard characters (both single- and multi-level) in incoming topics like the MQTT standard does.
This way we can forward single MQTT messages to multiple zigbee devices.

Eg.: having the friendly name for a device like "bedroom/lamps/1" and "bedroom/lamps/2" for an other one
* the MQTT standard allows to use "bedroom/lamps/+/{command}" topic with single-level wildcard character to send messages for both devices but now we can not do it

Related issue: Setting a zigbee node's friendly name to a multi level MQTT topic fregment is working for publish but no working for forwarding messages to a zigbee node.

Eg.: having the friendly name for a device like "bedroom/lamp1"
* MQTT messages can be received by an MQTT client in topic "{base_topic}/bedroom/lamps/1"
* MQTT messages can't be forwarded to the device with friendly name of "bedroom/lamps/1" by publishing a message to "{base_topic}/bedroom/lamps/1/{command}" topic

Koenkk, 2018-11-12:
This has already been implemented (https://github.com/Koenkk/zigbee2mqtt/issues/470)

pkedvessy, 2018-11-14:
I'm on the dev branch and as far as I can see the wildchard character support is not working (or not working as expected).
I have two Aqara double key wired wall switchs (QBKG03LM) with the following friendly names:
* `first_floor/working_room/lights/1`
* `first_floor/working_room/lights/2`

Setting the state on an mqtt topic without any wildcard character is working properly eg:
* `zigbee2mqtt/first_floor/working_room/lights/1/left/set`
* `zigbee2mqtt/first_floor/working_room/lights/2/left/set`

But using a single topic with wildcard to change the state both devices is not working by sending `{"state": "ON"}` to the `zigbee2mqtt/first_floor/working_room/lights/+/left/set` topic.

Koenkk, 2018-11-14:
I misunderstood your request, this is indeed a different feature.

closed 2019-01-20 by stale[bot]

PMunch, 2023-08-20:
I'd like to request this be opened again. I have a very similar situation with multiple Zigbee bulbs in a single fixture. I have some fixtures with three bulbs, some with two, and some with a single bulb. I want to be able to turn off fixtures by name from a script I've written, and if zigbee2mqtt had supported the `+` wildcard it would be as simple as publishing to `zigbee2mqtt/kitchen/fixture/+/set/state`. But without I would need to somehow know how many bulbs each fixture has and send to each bulb individually.

I might be able to work around this by creating a group, but not entirely sure yet. Still would be nice to support this feature.

HiFiJ, 2025-01-04:
Hi, 
I believe as well this would be a great feature! I am looking to set off several smoke detectors (SMSZB-120, Frient) all at once, see discussion here https://github.com/Koenkk/zigbee2mqtt/discussions/25487.
Here I unfortunately hit an additional problem, being unable to add those devices to a group.
Thanks very much!