Koenkk/zigbee2mqtt #591: zigbee2mqtt stops working after some time

naitsimp, 2018-11-12:
Hi,

I have installed the zigbee2mqtt Addon to my Hassio.
It starts and runs normal but after a while it just turns off and stops working.

In one log I got this as the last entry, but I only got this message once, normally the log ends with the last sent mqtt message.:
`./run.sh: line 25: 11 Segmentation fault (core dumped) ZIGBEE2MQTT_DATA="$DATA_PATH" pm2-runtime start npm -- start`

This is my normal log I get:
```
2018-11-12 03:54:19 - info: Logging to directory: '/share/zigbee2mqtt/log/2018-11-12.03-54-18'
2018-11-12 03:54:46 - info: Starting zigbee2mqtt version 0.1.8 (commit #unknown)
2018-11-12 03:54:46 - info: Starting zigbee-shepherd
2018-11-12 03:54:56 - info: zigbee-shepherd started
2018-11-12 03:54:56 - info: Coordinator firmware version: '20180815'
2018-11-12 03:54:57 - info: Currently 2 devices are joined:
2018-11-12 03:54:57 - info: aqara_double_switch_1 (0x00158d0001710b65): WXKG02LM - Xiaomi Aqara double key wireless wall switch (EndDevice)
2018-11-12 03:54:57 - info: aqara_cube_1 (0x00158d000289e0bf): MFKZQ01LM - Xiaomi Mi smart home cube (EndDevice)
2018-11-12 03:54:57 - info: Zigbee: disabling joining new devices.
2018-11-12 03:54:57 - info: Connecting to MQTT server at mqtt://homeassistant
2018-11-12 03:55:04 - info: zigbee-shepherd ready
2018-11-12 03:55:04 - info: Connected to MQTT server
2018-11-12 03:55:04 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
2018-11-12 03:55:05 - info: MQTT publish, topic: 'zigbee2mqtt/aqara_double_switch_1', payload: '{"battery":"55.00","voltage":2995,"linkquality":68}'
2018-11-12 03:55:05 - info: MQTT publish, topic: 'zigbee2mqtt/aqara_cube_1', payload: '{"battery":"42.00","voltage":2985,"linkquality":76}'
.... more lines of mqtt messages
```
Can someone help?

My configuration.yaml:
```
homeassistant: false
log_level: debug
mqtt:
  base_topic: zigbee2mqtt
  password: password
  server: mqtt://homeassistant
  user: username
permit_join: false
serial:
  disable_led: true
  port: /dev/ttyACM0
zigbee_shepherd_debug: true
```

Koenkk, 2018-11-12:
@danielwelch have you seen this happening elsewhere?

giejay, 2018-11-14:
Are you running on a raspberry and if so, could you check if its not due to a faulty sd card?

bendews, 2018-11-23:
I get this as well, running on a VM and in docker. Have dropped back to v0.1.7 to see if it helps but no luck.
I don't see any messages on the log out of the ordinary.

Koenkk, 2018-11-23:
@bendews can you provide the log?

rachetfoot, 2018-12-17:
I'm getting this too. I see that the service restarted about 10 times last night, sometimes right after initializing. Since then it has run fine all day. I don't see anything useful in the logs at all. Here's one of them:

```2018-12-17 00:28:41 - info: Logging to directory: '/opt/zigbee2mqtt/data/log/2018-12-17.00-28-41'
2018-12-17 00:28:42 - info: Starting zigbee2mqtt version 1.0.0 (commit #c798ecf)
2018-12-17 00:28:42 - info: Starting zigbee-shepherd
2018-12-17 00:28:43 - info: zigbee-shepherd started
2018-12-17 00:28:43 - info: Coordinator firmware version: '20180815'
2018-12-17 00:28:43 - info: Currently 10 devices are joined:
2018-12-17 00:28:43 - info: 0x00158d00022cbf64 (0x00158d00022cbf64): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
2018-12-17 00:28:43 - info: 0x00158d0002b6dd23 (0x00158d0002b6dd23): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
2018-12-17 00:28:43 - info: 0x00158d0001e801b4 (0x00158d0001e801b4): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)
2018-12-17 00:28:43 - info: 0x00158d0002042206 (0x00158d0002042206): RTCGQ01LM - Xiaomi MiJia human body movement sensor (EndDevice)
2018-12-17 00:28:43 - info: 0x00158d0002b7fb7d (0x00158d0002b7fb7d): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
2018-12-17 00:28:43 - info: 0x00158d00027d85ed (0x00158d00027d85ed): WXKG02LM - Xiaomi Aqara double key wireless wall switch (EndDevice)
2018-12-17 00:28:43 - info: 0x000b57fffee8e7c8 (0x000b57fffee8e7c8): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
2018-12-17 00:28:43 - info: 0x000b57fffedabe19 (0x000b57fffedabe19): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
2018-12-17 00:28:43 - info: 0x000b57fffed9a90c (0x000b57fffed9a90c): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
2018-12-17 00:28:43 - info: 0x000b57fffee2bd9e (0x000b57fffee2bd9e): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
2018-12-17 00:28:43 - warn: `permit_join` set to  `true` in configuration.yaml.
2018-12-17 00:28:43 - warn: Allowing new devices to join.
2018-12-17 00:28:43 - warn: Set `permit_join` to `false` once you joined all devices.
2018-12-17 00:28:43 - info: Zigbee: allowing new devices to join.
2018-12-17 00:28:43 - info: Connecting to MQTT server at mqtt://localhost
2018-12-17 00:28:44 - info: zigbee-shepherd ready
2018-12-17 00:28:44 - info: Connected to MQTT server
```

Koenkk, 2018-12-18:
@rachetfoot please try to catch a log when there is a crash.

rachetfoot, 2018-12-18:
@Koenkk That *was* from a crash. However, I just remembered that I was actually up that late and working on things, so at least some of those crashes are my own fault. However, there is nothing in the logs from the crash itself. I'm pretty sure I've had similar crashes before I started changing code, but that's hard to confirm now. Sorry that's not more helpful.

salopette, 2019-02-01:
Hello, I use the current Dev branch.
and it comes immeer again to the error.
After a reboot it runs some time and then again not more.

Here's the log:

https://paste.ubuntu.com/p/BTjZvCRfx2/

closed 2019-04-09 by stale[bot]