Koenkk/zigbee2mqtt #592: Attempt to pair Livolo Zigbee switch

Ton1965, 2018-11-14:
I have bought a Livolo zigbee switch like this one: https://www.aliexpress.com/item/Livolo-Touch-ZIGBEE/32918552579.html?spm=a2g0v.10010108.1000016%2FB.1.12754ab2JVEKJU

It seems to be a good option for its price.

The original Livolo instruction is to buy their own gateway and pair using their application. The pairing should be done by keeping a switch button pressed for 5 seconds. It should beep and pair after that.

The problem is it does not want to pair to zigbee2mqtt. I tried the following:

- press a button for 5 seconds, hear the beep, release the button and just wait;
- press a button for 5 seconds, hear the beep, keep it pressed for some more time (up to 90 seconds)
- switch on and off several times after the beep.

All the time I kept the device close to the zigbee stick.

I just could not observe any activity in zigbee2mqtt log in any of these cases. I have double checked that 'permit_join' is set to true and log_level set to 'debug'. I even tried to run zigbee2mqtt together with DEBUG=zigbee-shepherd* - no result.

Does anyone have any suggestions/advices/ideas? It would be very nice to add support to this device since this model seems to be much more attractive than the corresponding RF model.

khzd, 2018-11-14:
Hi Ton1965
I order one too, hop it's work with zigbee2mqtt
did not get it yet, i will post my finding too
cheers

Ton1965, 2018-11-15:
Hi @khzd,

I am glad I am not the only one! I still could not succeed in pairing the switch to zigbee2mqtt 😞. It just does not produce any output in the logs. Probably it is just broken but I hope for the best. I do not want to buy their gateway since it means installing it instead of some outlet and I do not have any spare ones.

Best wishes.

khzd, 2018-11-15:
Asap i got it I will give a try :) I will maybe confirme that yours is not broken :)

Ton1965, 2018-11-17:
@khzd ,
I am ready to help although I am not very experienced in this kind of things. But I will try to do my best although I will probably need some direction.
Can wireshark really be used to sniff on zigbee? I thought it is used for http traffic going through the router, but as I said I do not know much about this.

lych, 2018-11-20:
Hi guys.
When I was buying this switch, a seller warned me it works only with their gateway.
I've tried wireshark with zigbee sniffer. I wasn't able to catch any zigbee packets. So, I see two options, it's encrypted or it uses their own proprietary protocol.
I'm waiting for another CC2531 dongle because I have only one and it's difficult to flash it each time.
When I get the new device, I'll continue the experiments to understand if it's encrypted or not.

Lumbricus, 2018-11-23:
I also bought one of the switches and have the same problem. Would be nice to pair it, because the RF version douesn't give a feedback. There's only a receiver inside.

If you open the switch you can see, that they use a CC2630.

Ton1965, 2018-11-24:
@khzd,
I am in almost the same situation: I want it to fit into the European hole. I have bought just one for now since I suspected there might be some difficulties. If you have already ordered all 12 then you probably would be better off to get their coordinator althoug I do not like this idea😞. I still hope that probably someone does have a coordinator and will be able to hack the protocol.
PS. I have Livolo RF switch and it is running fine but it does not return its state so it is not always convenient.

Koenkk, 2018-11-24:
A bit offtopic: but for european holes I would highly recommend the Shelly 1 with tasmota (wifi esp8266 device). It fits easily and allows you to keep your normal switch. Ive installed 20 of them in my home and they work perfect!

lych, 2018-11-25:
According to their advertisment, their gateway is able to work with Amazon Alexa but I haven't seen any APIs yet.

Ton1965, 2018-11-25:
> A bit offtopic: but for european holes I would highly recommend the Shelly 1 with tasmota (wifi esp8266 device). It fits easily and allows you to keep your normal switch. Ive installed 20 of them in my home and they work perfect!

Shelly would have been nice but I do not have any neutral line in my apartment. Just two wires in each hole. That is why Livolo suits me well, their switches do not require neutral.

kirovilya, 2018-11-25:
@lych run zigbee2mqtt on channel 20 and try to pair livolo switch

lych, 2018-11-25:
It looks encouraging!

 zigbee2mqtt:info 2018-11-25 18:58:54 New device with address 0x00124b000befc0b9 connected!

 zigbee2mqtt:info 2018-11-25 18:58:54 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x00124b000befc0b9"}'

zigbee2mqtt:warn 2018-11-25 18:58:54 Device with modelID 'TI0001          ' is not supported.

kirovilya, 2018-11-25:
@lych now try press switch button and record debug log. We need messages ;)

Ton1965, 2018-11-25:
@lych 
Great to hear! I have always believed someone would be able to pair this device!

I would like to do my own experiments now but I have the only stick and it is so to say in production mode now. 

This switch seems to work on channel 11 only. I do not really know the channel my stick is running on but it should be some other one since the pairing failed in my case. If I change my zigbee2mqtt coordinator to channel to 11 now would it mean re-pairing all the devices? Or probably they will all shift to a different channel :crossed_fingers: ?

lych, 2018-11-25:
@Ton1965 You shouldn't change your coordinator channel, at least now, I haven't been able to make it to work yet. I am not sure about re-pairing, after changing the channel I still see data from my temperature sensor without repairing, but another my device (Xiaomi button) doesn't work. I think it depends on enpoint device.
@kirovilya There is nothing interesting in zigbee2mqtt log, see my previous post, there is the most interesting part of log.
[livolo_pairing.zip](https://github.com/Koenkk/zigbee2mqtt/files/2613373/livolo_pairing.zip)
I've attached pcap file for wireshark [livolo_pairing.zip] (https://github.com/Koenkk/zigbee2mqtt/files/2613373/livolo_pairing.zip). There is the pairing process and some touches of sensor.
It looks like the pairing process can't be finished on the switcher side. The switcher wants some kind of response from coordianator, but sends something when I touch the sensor. It's the worst part, the message is encrypted.

lych, 2018-11-25:
@ptvoinfo There is a photo of the board
[image]
I think it's possible to flash it, but to do it one have to figure out how to connect the programmer and hack the protocol between CC2630 and main PIC controller.
I still hope that it's possible to use it without such difficulties.

kirovilya, 2018-11-26:
@lych I wanted to look at a debug log from zigbee2mqtt, but not wireshark :) https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-debug

lych, 2018-11-26:
@kirovilya As I've already said there is nothing in zigbee2mqtt log.

> zigbee2mqtt:info 2018-11-25 18:58:54 New device with address 0x00124b000befc0b9 connected!
>
> zigbee2mqtt:info 2018-11-25 18:58:54 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: [...]
It's all, nothing else.

kirovilya, 2018-11-26:
@lych Even when you press the switch button after that?
Did you run zigbee2mqtt with debug parameter? https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-debug#zigbee-shepherd-debug-logging
Must be more data.

Koenkk, 2018-11-26:
It could be that reporting/binding needs to be setup, can you try with:

```js
    {
        zigbeeModel: ['TI0001 '],
        model: 'TI0001 ',
        description: 'Zigbee switch',
        supports: 'on/off',
        vendor: 'Livolo',
        fromZigbee: [],
        toZigbee: [tz.on_off],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
            ];

            execute(device, actions, callback);
        },
    },

```

lych, 2018-11-26:
There is a log with Koenkk's changes.
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/2616599/log.txt)
Nothing in the log when I touch the sensor.

kirovilya, 2018-11-26:
I think that model is not 'TI0001 '... somthing more :)
need database.db after pairing

lych, 2018-11-26:
[database.zip](https://github.com/Koenkk/zigbee2mqtt/files/2616892/database.zip)

kirovilya, 2018-11-26:
`"TI0001          "`

lych, 2018-11-26:
Message which the switch sends when I touch the sensor contains encrypted data. Can it be the problem or it is ok?

khzd, 2018-11-27:
Good progress @lych, I got mine  today,I will join tests also.
Can you recap all steps ?
Cheers

lych, 2018-11-27:
@khzd Unfortunately, I haven't had any progress yet. I figured out that Livolo works on channel 26. According to logs, the pairing process looks ok. After it, if I touch the sensor there is a packet in Wireshark, but it marked as malformed. I wasn't able to find any trails of this packet in cc-znp, so I suspect that coordinator just discards such packets. Moreover, if I turn Livolo's power off then on, it starts the pairing process again as if it doesn't remember that it had been already paired. I suspect that something wrong with the pairing process. Strange thing, my switch starts the pairing process immediately as I turn power on, doesn't matter if I touch sensor for 5 seconds or not.

kirovilya, 2018-11-28:
I also helped a friend to pair the switch with the coordinator. But then he tried to send genOnOff commands to switch endpoint 6. No success. :(
`6 failed with error Error: AF data request fails, status code: 240. MAC transaction expired.`

Ton1965, 2018-11-28:
Hi @khzd,

I have found some difference between your database and the one pasted here by @lych in [this message](https://github.com/Koenkk/zigbee2mqtt/issues/592#issuecomment-441750383). Your database shows the switch status as offline but his has several records with online status so the device can at least be brought to online state. I am not sure if this fact is helpful though.
Another strange thing is the modelId that has 9 extra spaced in @lych's database but not in yours.

khzd, 2018-11-28:
I confirm, I have the @kirovilya 10 spaces :)

khzd, 2018-11-28:
**"For European holes, zigbee is progressing ..."**
Hi,
the device is recognized correctly what about the specific fromZigbee code to handel it : 
    {
        zigbeeModel: ['TI0001 '],
        model: 'TI0001 ',
        description: 'Zigbee switch',
        supports: 'on/off',
        vendor: 'Livolo',
        fromZigbee: [**MISSING THIS SECTION**],
        toZigbee: [tz.on_off],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
            ];
            execute(device, actions, callback);
        },
    },

lych, 2018-11-29:
There is a mistake in config. I think it must be
`toZigbee: [tz.onoff],`
instead of
`toZigbee: [tz.on_off],`

lych, 2018-11-29:
Now, zigbee2mqtt sends the reqest "Discover Attributes", Livolo answers "Unsupported General Command"
then, zigbee2mqtt sends "Read Attributes" and Livolo answers its current state (on or off)
And it's all, I think next zigbee2mqtt have to send something to change the state but it doesn't.
I suspect something else have to be in config.

khzd, 2018-11-29:
Hi did you got a good results ?

khzd, 2018-11-29:
Hi did you got good results ?
Onoff in converters is not there , only on_off is!

lych, 2018-11-29:
Now, if I try to change a state, zigbee2mqtt sends the command "ZCL OnOff" and Livolo answers "Success" but nothing really happens (reley doesn't change its state).
Then zigbee2mqtt sends "Discover Attributes", Livolo answers "Unsupported General Command"
then, zigbee2mqtt sends "Read Attributes" and Livolo answers its current state (on or off), it is the real state (it's only one thing which looks working in right way)

If a switch has more than one button, how should the command look like? It cannot just be on / off. There must be something else, like pointing to a specific button.

khzd, 2018-11-29:
hi, was able to debug : I reduce all devices from DB and got this after sendig ON via mqtt : results same as you :(

zigbee2mqtt:debug 2018-11-29 21:36:15 Received MQTT message on '**zigbee2mqtt/my_office_light_up/set**' with data 'ON'
  zigbee2mqtt:info 2018-11-29 21:36:15 Zigbee publish to '**0x00124b000bf0725e**', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 3 +8s
  cc-znp:SREQ --> AF:dataRequest, { dstaddr: 63208, destendpoint: 6, srcendpoint: 1, clusterid: 6, transid: 3, options: 48, radius: 30, len: 3, data: <Buffer 01 03 01> } +1ms
  serialport:main _write 18 bytes of data +8s
  serialport:bindings write 18 bytes +8s
  serialport:unixWrite Starting write 18 bytes offset 0 bytesToWrite 18 +3m
  serialport:unixWrite write returned null 18 +1ms
  serialport:unixWrite wrote 18 bytes +0ms
  serialport:unixWrite Finished writing 18 bytes +0ms
  serialport:main binding.write write finished +1ms
  serialport:poller received "readable" +8s
  serialport:bindings read +10ms
  serialport:unixRead Starting read +8s
  serialport:unixRead Finished read 6 bytes +0ms
  serialport:main binding.read finished +9ms
  cc-znp { sof: 254,
  cc-znp   len: 1,
  cc-znp   type: 'SRSP',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'dataRequest',
  cc-znp   payload: { status: 0 },
  cc-znp   fcs: 100,
  cc-znp   csum: 100 } +12ms
  serialport:main _read reading +2ms
  serialport:bindings read +2ms
  serialport:unixRead Starting read +2ms
  cc-znp:SRSP <-- AF:dataRequest, { status: 0 } +1ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +0ms
  serialport:unixRead waiting for readable because of code: EAGAIN +1ms
  serialport:poller Polling for "readable" +3ms
  serialport:poller received "readable" +7s
  serialport:bindings read +7s
  serialport:unixRead Starting read +7s
  serialport:unixRead Finished read 30 bytes +0ms
  serialport:main binding.read finished +7s
  cc-znp { sof: 254,
  cc-znp   len: 25,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'incomingMsg',
  cc-znp   payload: 
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 10,
  cc-znp      srcaddr: 38930,
  cc-znp      srcendpoint: 1,
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 57,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 16055769,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 5,
  cc-znp      data: <Buffer 10 9a 00 00 00> },
  cc-znp   fcs: 173,
  cc-znp   csum: 173 } +7s
  serialport:main _read reading +3ms
  serialport:bindings read +4ms
  serialport:unixRead Starting read +4ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 10, srcaddr: 38930, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 57, securityuse: 0, timestamp: 16055769, transseqnumber: 0, len: 5, data: <Buffer 10 9a 00 00 00> } +1ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +1ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +0ms
  serialport:unixRead waiting for readable because of code: EAGAIN +1ms
  serialport:poller Polling for "readable" +6ms
  serialport:poller received "readable" +11s
  serialport:bindings read +11s
  serialport:unixRead Starting read +11s
  serialport:unixRead Finished read 8 bytes +1ms
  serialport:main binding.read finished +11s
  cc-znp { sof: 254,
  cc-znp   len: 3,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'dataConfirm',
  cc-znp   payload: { status: 183, endpoint: 1, transid: 3 },
  cc-znp   fcs: 114,
  cc-znp   csum: 114 } +11s
  serialport:main _read reading +3ms
  serialport:bindings read +4ms
  serialport:unixRead Starting read +4ms
  cc-znp:AREQ <-- AF:dataConfirm, { status: 183, endpoint: 1, transid: 3 } +2ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +0ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 3 +1ms
  zigbee2mqtt:error 2018-11-29 21:36:33 Zigbee publish to '0x00124b000bf0725e', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 183. APS no ack.
  serialport:unixRead waiting for readable because of code: EAGAIN +3ms
  serialport:poller Polling for "readable" +8ms
  serialport:poller received "readable" +10s
  serialport:bindings read +10s
  serialport:unixRead Starting read +10s
  serialport:unixRead Finished read 30 bytes +1ms
  serialport:main binding.read finished +10s
  cc-znp { sof: 254,
  cc-znp   len: 25,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'incomingMsg',
  cc-znp   payload: 
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 10,
  cc-znp      srcaddr: 36978,
  cc-znp      srcendpoint: 1,
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 76,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 16119899,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 5,
  cc-znp      data: <Buffer 10 5e 00 00 00> },
  cc-znp   fcs: 154,
  cc-znp   csum: 154 } +10s
  serialport:main _read reading +5ms
  serialport:bindings read +6ms
  serialport:unixRead Starting read +5ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 10, srcaddr: 36978, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 76, securityuse: 0, timestamp: 16119899, transseqnumber: 0, len: 5, data: <Buffer 10 5e 00 00 00> } +2ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +1ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +0ms
  serialport:unixRead waiting for readable because of code: EAGAIN +2ms
  serialport:poller Polling for "readable" +8ms

khzd, 2018-11-29:
my mini DB
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0012022f78","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1543522885,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}},"_id":"9qLpaadwaFy8kHOl"}
{"id":64,"type":"EndDevice","ieeeAddr":"0x00124b000bf0725e","nwkAddr":63208,"manufId":0,"manufName":"LIVOLO\u0000\u0010TI0001  ","powerSource":"Mains (single phase)","modelId":"TI0001          ","epList":[6],"status":"offline","joinTime":null,"endpoints":{"6":{"profId":260,"epId":6,"devId":0,"inClusterList":[0,3],"outClusterList":[6],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}}}}},"_id":"vzW78bhgIYIkCu6z"}
{"$$indexCreated":{"fieldName":"id","unique":true,"sparse":false}}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0012022f78","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1543523552,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}},"_id":"9qLpaadwaFy8kHOl"}
{"id":64,"type":"EndDevice","ieeeAddr":"0x00124b000bf0725e","nwkAddr":63208,"manufId":0,"manufName":"LIVOLO\u0000\u0010TI0001  ","powerSource":"Mains (single phase)","modelId":"TI0001          ","epList":[6],"status":"offline","joinTime":null,"endpoints":{"6":{"profId":260,"epId":6,"devId":0,"inClusterList":[0,3],"outClusterList":[6],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}}}}},"_id":"vzW78bhgIYIkCu6z"}

khzd, 2018-11-29:
> Now, if I try to change a state, zigbee2mqtt sends the command "ZCL OnOff" and Livolo answers "Success" but nothing really happens (reley doesn't change its state).
> Then zigbee2mqtt sends "Discover [...]

for 2 switch you may have two diffrents commands, see the difference in to_zigbee for the dual switch xiaomi

lych, 2018-11-29:
My swith has only one button, but I suspect that they all have the same firmware

lych, 2018-11-29:
@khzd What should the mqtt payload look like in this case?

khzd, 2018-11-29:
give me 5 mn, need to put HA in Debug mqtt to have the exact syntax

khzd, 2018-11-29:
hi do you use HA?

lych, 2018-11-29:
No, I don't

khzd, 2018-11-29:
if yes this the example in zigbee2mqtt for a double switch
 switch:
  - platform: "mqtt"
    state_topic: "zigbee2mqtt/<FRIENDLY_NAME>"
    availability_topic: "zigbee2mqtt/bridge/state"
    payload_off: "OFF"
    payload_on: "ON"
    value_template: "{{ **value_json.state_left** }}"
    command_topic: **"zigbee2mqtt/<FRIENDLY_NAME>/left/set**"
    json_attributes: 
      - "linkquality"
      - "**button_left**"

switch:
  - platform: "mqtt"
    state_topic: "zigbee2mqtt/<FRIENDLY_NAME>"
    availability_topic: "zigbee2mqtt/bridge/state"
    payload_off: "OFF"
    payload_on: "ON"
    value_template: "{{ **value_json.state_right** }}"
    command_topic: "**zigbee2mqtt/<FRIENDLY_NAME>/right/set**"
    json_attributes: 
      - "linkquality"
      - "button_right"

khzd, 2018-11-29:
so from previous example we can build the good syntax :+1: right : your broker 
 zigbee2mqtt/<FRIENDLY_NAME>/right/set -m 'ON'

lych, 2018-11-29:
I use mosquitto.
Thanks, I've alredy found how to do it, but something wrong
`Zigbee cannot publish message to device because '0x00124b000befc0b9' not known by zigbee-shepherd`
Maybe I should updaye my zigbee2mqqt, but tomorrow, it's time to go to bed :) thanks

khzd, 2018-11-29:
never give up with zigbee :)

khzd, 2018-11-29:
hi @kirovilya @Koenkk 
do you have enough debug log or do i need to give more?
Merci

Ton1965, 2018-11-30:
Hi @khzd ,

I see you are ready to submit a PR to shepherd-converters to support these switches. It's great news! You guys are awesome! 
I have a couple of questions though (more than a couple actually 😄). It was reported here that the coordinator needs to be on channel 26 to pair with these switches. Is it still true? I currently have a bunch of devices paired to my stick, will I need to repair them if I change the channel? Is it possible to make channel selection automatic again after the pairing is done?

khzd, 2018-11-30:
hello,
i changed also to channel 26 and unhappily  I had to re-pair all router devices the end-devices do not need
see in the startup or in the database the nature of the devices, it will gives you the  total effort :(

Ton1965, 2018-11-30:
Hi,
Fortunately I do not currently have any routers, just the end devices 😄. Does it mean that we can easily encounter devices from different manufacturers that would be impossible to use together in the same network? Like after adding these Livolo switches we get stuck with channel 26 only and will be unable to move to any other channel.

khzd, 2018-11-30:
not sure, i'm not expert on zigbee protocol. 
NB in wifi the channel is recommended to be changed to not interfere with others sources .

lych, 2018-11-30:
Hi, @khzd . Have you managed to make it work?

khzd, 2018-11-30:
unhappily not :(
the device is connected to the network, i can send request without timeout
BUT no response for on/off, what is amazing is that ( because we are on channel 26, thanks) the pairing is reproducible without issue ( i did it 4 or 5 times) and have long log about it

i learn a lot how to debug on javascript, but no competency on coding zigbee2mqtt :(
we need help from devs team, probably busy with a long list of requests, this is the price of the success

cheers

Ton1965, 2018-11-30:
@khzd ,
I am not an expert as well. I have looked through some papers like Zigbee standard and some papers. It looks like channel hopping was introduced to Zigbee around 2007 but it was never widely adopted. There were some statements that in general a device should scan all the channels while pairing but some devices might have a special application profile and they can be programmed at the firmware level to use some subset of channels or a single channel.
Still just as a final test: could you please try to change the coordinator channel and wait for some time to see if it would still work. I just hope that this channel hopping is already there and a specific channel is necessary at pairing stage only.

khzd, 2018-11-30:
give me 10 mn

khzd, 2018-11-30:
i would test this  ep: {'left': 1, 'right': 2, 'both': 3}, but I have a single swich
can you show me what do you have in eplist from DB?

lych, 2018-11-30:
the same, "epList":[6] . I don't have 2 channel switch, I just assumed that they can have the same firmware

khzd, 2018-11-30:
ok, but in zifbee2mqtt , there is differentiation between single, double switch. they probaly have a common code but some thing to differentiate this feature

[639 more replies]

closed 2020-09-08 by github-actions[bot]