Koenkk/zigbee2mqtt #595: CC2530 through popular UART based on Prolific PL2303

MikeVRT, 2018-11-16:
Hello,
I'm got my CC2530 board with external antena (Fig. 1 on ptvo.info) and trying replace my CC2531.
With flashing - no problem that guide works well: http://ptvo.info/how-to-select-and-flash-cc2530-144/
Was used SmartRF04EB with dupont cable.
But after connection to PC throught UART based on Prolific PL2303 (I knew that it needs old drivers)

I'm got error:
`2018-11-16 03:10:20 - error: Failed to start Error: request timeout
    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:255:22)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at CcZnp.emit (events.js:211:7)
    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:234:18)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5) `

Same error happens with 3 firmwares:
 1. ptvo http://ptvo.info/wp-content/uploads/2018/09/cc2530_1.2.2a.44539_firmware.zip
     coordinator\CC2530ZNP-Pro-Secure_LinkKeyJoin.hex.hex
 2. dzungpv https://raw.githubusercontent.com/dzungpv/Z-Stack-firmware/master/coordinator/CC2530/bin/CC2530ZNP-Prod.hex
    included fix for baud rate & flow control: 
     https://github.com/Koenkk/zigbee2mqtt/issues/293#issuecomment-414297741
     https://github.com/dzungpv/Z-Stack-firmware/blob/master/coordinator/CC2530/README.md
 3. kirovilya   
     https://github.com/kirovilya/files/blob/master/CC2531ZNP-Pro-Secure_LinkKeyJoin.hex      
 https://github.com/kirovilya/ioBroker.zigbee/wiki/%D0%9F%D1%80%D0%BE%D1%88%D0%B8%D0%B2%D0%BA%D0%B0

All firmwares was tryed with sort out RX/TX connetion and P20 to GND or empty. Inside UART power wire resoldered from 5V to 3.3V (this CC2530 board supports only VCC 2V - 3.6V).
For sure database of Zigbee2mqtt was cleared.

ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 Nov 16 03:08 usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0 -> ../../ttyUSB0

stty -F /dev/ttyUSB0 -a
shows 115200 boud rate
but shows "crtscts" without "-" but set "-crtscts" by "stty -F /dev/ttyUSB0 -crtscts" changes state for few seconds - than "stty -F /dev/ttyUSB0 -a" returns "crtscts" again.
All in Ubuntu Server 18, CC2531 works well in same environment.

A lot of thanks for ideas how to solve "timeout" error or detect reason.

MikeVRT, 2018-11-16:
[image]
[image]

MikeVRT, 2018-11-16:
Additional info: connect UART + CC2531 to windows machine by TX->P02, RX->P03, GND->P22. In this combination no error in RealTerm and on open port Char Count: 12 (in hex 00FE064180000200020603C2)

ptvoinfo, 2018-11-16:
@MikeVRT My version of firmware uses 115200 & rtscts = false (off). You need to connect P02, P03, GND, +3V (to the corresponding pin on the board).

MikeVRT, 2018-11-16:
> @MikeVRT My version of firmware uses 115200 & rtscts = false (off). You need to connect P02, P03, GND, +3V (to the corresponding pin on the board).

"stty -F /dev/ttyUSB0 -a" shows 115200 boud rate, so here ok. rtscts = false - it's default if not uses advanced section in configuration.yaml of zigbee2mqtt. Is it enought? 
I checked CC2531 and it's shows "-crtscts" by default instead "crtscts" in СС2530 + UART. But as I wrote - changing it by "stty -F /dev/ttyUSB0 -crtscts" take effect for few secconds only. Or it's doesn't matter?

hamstie, 2018-11-16:
1) do you check for hidden process on ttyUSB0 (e.g. terminal)
- Stop zigbee2mqtt
> lsof|grep "/dev/ttyUSB0"  any output ?

2) Check Configuration: The serial  rtscts config is NOT in serial section, it is in the advanced section: 
 
advanced:
  # Optional: Baudrate for serial port
  baudrate: 115200
  # Optional: RTS / CTS Hardware Flow Control for serial port
  rtscts: false

MikeVRT, 2018-11-17:
@hamstie,
`lsof|grep "/dev/ttyUSB0"` gives empty output

hamstie, 2018-11-17:
Check UART hardware:
- remove CC2531 
> mincom -D /dev/ttyUSB0 and configure uart (disable hardware flow control)
type in any string and it blocks (normal)
connect Tx and Rx pin with an wire
type any string and you will be see the string, uart looks good

if you haven't got success, check boot messages:
dmesg|grep "/dev/ttyUSB0"

MikeVRT, 2018-11-18:
@hamstie,
dmesg|grep "/dev/ttyUSB0" empty, but I allready checked it - by chip's name:

> [  131.675790] usbcore: registered new interface driver pl2303
> [  131.678045] usbserial: USB Serial support registered for pl2303
> [  131.678315] pl2303 1-1:1.0: pl2303 converter detected
> [  131.688810] usb 1-1: pl2303 converter now attached to ttyUSB0

here ok.

About minicom. Yes, I made it today.
With dzungpv's fw ("Coordinator firmware version: '20182308'") zigbee2mqtt sometimes starts (sometimes timeout again). Coordinator appiears in database.db
But xiaomi's devices whan't appiear in log or database after trys of pairing.
But we have some progress!
So. Tomorrow I will flash last ptvo's firmware and try again. Thx!

MikeVRT, 2018-11-18:
p.s.
> connect Tx and Rx pin with an wire
> type any string and you will be see the string, uart looks good
not don't made this test because under widows allready [got answer](https://github.com/Koenkk/zigbee2mqtt/issues/595#issuecomment-439273869) from device throught RealTerm

MikeVRT, 2018-11-18:
> @MikeVRT My version of firmware uses 115200 & rtscts = false (off). You need to connect P02, P03, GND, +3V (to the corresponding pin on the board).

One question - in yours firmware [description](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) in "How to use" wrote:

> P2_o pin state: ignored

So, we not need P20 & GND connection when CC2530 used?

MikeVRT, 2018-11-18:
I'm flash ptvo's firmware again and upgrade zigbee2mqtt to last version (0.2.0 (commit #b0d3c2f)).
Now sometime all starts and devices pairing.

BUT. Work not so stable as was befor with CC2531 USB stick and zigbee2mqtt 0.1.6:
- zigbee2mqtt->zigbee-shepherd not started 2 from 3 times (attemping to fix... (takes 60 seconds) than timeout error). Before on CC2531 never happens
- pairing errors "Cannot get the Node Descriptor of the Device". Before on CC2531 never happens
- pairing warinngs "Message without device!". Before on CC2531 never happens
- after pairing not all sensor's params comes, fore example WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor after pairing showing in log only payload: {"pressure":999,"linkquality":126}'. Some sensors got other data late, some - still not. Before on CC2531 never happens

About stability can't said - need time.

MikeVRT, 2018-11-19:
By the way - in CC2530 selleer's description was

> UART:1

Ant i found this in net:

> It uses P0.2 as UART RX and P0.3 as TX by default. This is UART0 alt.1 and you can revise it to UART0 alt.2 (P1.5 as TX and P1.4 as RX)
> or UART1 alt.1 (P0.4 as TX and P0.5 as RX) or UART1 alt.2 (P1.6 as TX and P1.7 as RX)

So, may be for that type of CC2530 better to use P04 & P05 for TX/RX? How to understand?

pagenoare, 2018-12-18:
Hey!

I have a CC2530 board that I have flashed using alternative method (through ESP), the flashing was successfull (based on the output I got). I've flashed this firmware: https://github.com/kirovilya/files/blob/master/CC2530ZNP-Pro-Secure_LinkKeyJoin_mod.hex

Then I've connected the module to the Prolific PL2303 adapter, and have been trying to make it work with zigbee2mqtt, however its a bit tricky. I tried to run `minicom -D /dev/ttyUSB0` but its just empty. What's the way to debug it futher to figure out what could went wrong?

Sorry to hijack this issue @MikeVRT but it looks like we're trying to use the same setup. Also, I have a bit different board -> https://www.waveshare.com/wiki/Core2530_(B)

Any tips appreciated!

Thanks.

MikeVRT, 2018-12-18:
Hi @pagenoare. As result - my CC2530 works as coordinator not stable and has problems with pairing. So, I flashed it to router role. As router it's works good exclude some things: 
pairing possible only near coordinator (most times),
repeater's state in log stay 'false' (some time it can be 'true' after repower for not long time),
all sensors works throught repeater has linkquality = 0

I'm waiting 2 parcels:
1. want check CC2530 as coordinator with another UART
2. second CC2531 for sure that CC2530 good/bad as repeater too.

p.s. of course Prolific PL2303 adapter in router mode used only as power supply (3.3v).

pagenoare, 2018-12-18:
@MikeVRT I can't get it work work at all though. On which firmware did it work for you in coordinator mode? I'm flashing to https://github.com/mtornblad/zstack-1.2.2a.44539/blob/master/CC2530/CC2530ZNP-Pro-Secure_LinkKeyJoin.hex now.

MikeVRT, 2018-12-18:
> I can't get it work work at all though.
I mean, that first you must check UART - is it works?

pagenoare, 2018-12-18:
@MikeVRT Ok - thanks. You have edited the comment hence I didn't see the part about testing out UART. I've installed the drivers now, provided by seller. How can I make sure UART itself works correct now? Is there any way to test that out? Sorry for noobish questions, but I'm kinda new to all this stuff.

I'm gonna wait till the fw I'm currently flashing will be done, test and if it still doesn't work I'll flash again with ptvo's fw, thanks!

pagenoare, 2018-12-18:
OK - I've connected pins TX to RX on UART, connected to PC, opened Putty, selected the COM10, and opened the connection. Typed "hello" and enter, and it moved back to the beginning of the line. Does it mean that UART USB works correctly?

MikeVRT, 2018-12-18:
> How can I make sure UART itself works correct now? Is there any way to test that out? Sorry for noobish questions, but I'm kinda new to all this stuff.

On windows in device manager you must see "Prolific USB-to-Serial Comm Port" without any problems and unknown devices. On other env - something like that.

MikeVRT, 2018-12-18:
> OK - I've connected pins TX to RX on UART
Are you resoldered 5V to 3.3.V in UART or yours CC2530 supports 5V?

pagenoare, 2018-12-18:
@MikeVRT so, I have flashed the https://raw.githubusercontent.com/mtornblad/zstack-1.2.2a.44539/master/CC2530/CC2530ZNP-Pro-Secure_LinkKeyJoin.hex fw. Connected to UART->USB adapter, plugged in into the rpi, and tried to see what minicom would say.

Started as you mentioned `minicom -D /dev/ttyUSB0 -b 115200`, but nothing happens...

pagenoare, 2019-01-17:
@MikeVRT Just want to let you know that I've found my issue. It was all good with flashed software. My wiring was screwed up, after bunch of try and fails I finally got it to work. Thanks for the help!

Koenkk, 2019-01-17:
Could you share the wiring?

lenisko, 2019-01-17:
@Koenkk 
You have to crossover TX,RX on UART
```
[VCC] [GND]     [   ] [   ]
[   ] [   ]     [GND] [   ]
[   ] [   ]     [   ] [   ]
[TX ] [RX ]     [   ] [   ]
[   ] [   ]     [   ] [   ]
[   ] [   ]     [   ] [   ]
```
that's for CC2530 https://www.dhresource.com/0x0s/f2-albu-g7-M01-B1-97-rBVaSluH0baAL6RUAAHBK7_EGI8709.jpg/cc2530-zigbee-module-uart-wireless-core-board.jpg

msadeghz, 2019-01-29:
I've launched the ZNP with CC2530 and `ftdi USB to TTL module`, but with some minor modifications. I made the ZNP firmware serial flow-control disable and connect the `CTS pin` of the `ftdi module` to GND (make "`uartConfig.flowControl          = TRUE;`" to "`uartConfig.flowControl          = FALSE;`" in `znp_app.c`). so, the crtscts parameter in configuration.yaml is set to true.

closed 2019-04-10 by stale[bot]