Koenkk/zigbee2mqtt #598: Version 0.2.0 stops working with error regarding usuccess request: 17

pepp86, 2018-11-16:
Using zigbee2mqtt with your docker container on a raspberrypi running libreelec, latest version.

When updating from 0.1.8 to 0.2.0 I get the following errors in my log and the whole docker container restarts.

https://pastebin.com/00nzAg8f

Koenkk, 2018-11-16:
Can you try:

- stopping zigbee2mqtt
- pressing the reset button on the stick (the one close to the usb port)
- start zigbee2mqtt

pepp86, 2018-11-20:
Hi.

Did not help. No chance to get it up and running

Koenkk, 2018-11-23:
If a reboot/reset doesn't fix the problem I think the only way to fix it is to reflash the stick.

pepp86, 2018-12-05:
Still same issue with latest firmware and a brand new stick

Koenkk, 2018-12-05:
Can you try a different usb port?

pepp86, 2018-12-05:
I will try it. But actually it happens on two different machines with different sticks

pepp86, 2018-12-20:
Hi,

it works with latest version 1.0.1 on both machines with different sticks and stick firmware.

Issue can be closed.

closed 2018-12-20 by pepp86