Koenkk/zigbee2mqtt #600: Lowes IRIS motion sensor not pairing

mcsSolutions, 2018-11-16:
Attempts to pair Lower IRIS Motion Sensor 3326-L are not successful.  Cannot get the Node Descriptor, Timed out after 10000 ms error  is reported on command line.  Debug log segment posted at https://pastebin.com/RELKgj4m

This device has no pairing button.  Pairing occurs when the battery is inserted.  It appears to repeat the pairing attempt on 20 second intervals.  Led blinking on the 3326-L shows its status during the attempt cycles.  Since there is no link button it is not possible to use the solutions that have been posted for other zigbee devices.  

In the debug this motion sensor device shows as 0x000d6f001099e6c1.  Previoulsy four Xiami water leak sensors were paired and their record shows at the start of the posted debug.   Database.db contains no entries for the motion sensor.  Only for the water leak sensors.

What options are available to complete the pairing of this device so it can be added as a supported device?

johnnyletrois, 2018-11-17:
I recently bought the same motions sensors when they were on sale and am not having luck pairing them.  There is a reset button on the side, and I've tried pressing that while inserting the batter to factory reset it.  No luck.  Same message: "Cannot get the Node Descriptor, Timed out after 10000 ms"

Any ideas?  Thanks!

https://pastebin.com/4BaSQL4M

pixeldoc2000, 2018-11-17:
Some Devices require special handeling to get them to work. Maybe thats the case with this devices too. Looks like many Motion Sensor require some specials.

Have a look here: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices

This device is not (yet) supported.

johnnyletrois, 2018-11-20:
> Some Devices require special handeling to get them to work. Maybe thats the case with this devices too. Looks like many Motion Sensor require some specials.
> 
> Have a look here: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
> 
> This device is not (yet) supported.

Thanks.  I'm not sure how to integrate zapp-cie with zigbee2mqtt.  That's above my pay grade.

mcsSolutions, 2018-11-21:
I also do not know what my next step should be to help get this to be a supported device.  I do not understand what /if I should be doing something with zap-cie.  It looks to be some code addition based upon usage description of using var ...  Is there something that I should do to zigbee-shepard?  I can understand how to add a new device one paired, but how do I get zigbee2mqtt to communicate with the motion sensor to complete the pairing handshake?

Koenkk, 2018-11-23:
Some changes need to be made in zigbee-shepherd to fix the pairing.

Commits like https://github.com/athombv/zigbee-shepherd/commit/91c1e8073542c771cff971c3f94baf361707bee7 (which are currently not used in zigbee2mqtt) could fix this.

Unfortunately I cannot debug this further myself as I don't have the device.

johnnyletrois, 2018-11-23:
Thanks @koenkk for the suggestion. I’ll see if I can’t make some progress this weekend.

johnnyletrois, 2018-11-24:
@Koenkk both those commits are in the version of zigbee-shepherd used in the zigbee2mqtt docker container, so apparently those changes made it.  I'm not sure what my next step should be.  It seems the error is common to Osram, SmartThings, and Iris sensors.

Koenkk, 2018-11-25:
There are several iAS zone enrollment methods. I guess zigbee-shepherd doens't support all of them. See https://www.nxp.com/docs/en/user-guide/JN-UG-3077.pdf (26.3 Enrollment)

johnnyletrois, 2018-11-30:
This is beyond my skillset and I haven’t made any progress. I’ll be returning the sensors this weekend. It’s too bad because the motion/temp sensors are very small and apparently very responsive.

Koenkk, 2018-11-30:
Before returning, can you check if pairing works when using: https://github.com/koenkk/zigbee2mqtt/tree/osram_motion

johnnyletrois, 2018-11-30:
Thanks @koenkk. I’ll give that branch a try.

Koenkk, 2018-11-30:
FYI: it worked with the osram motion sensor where we got the same error before.

Koenkk, 2018-12-01:
no

johnnyletrois, 2018-12-01:
OK.  I managed to build my own using the osram_motion branch at johnnyletrois/zigbee2mqtt:osram_branch and it's running successfully.  In debug mode, I'm getting the following error:

`
  zigbee2mqtt:debug 2018-12-1 17:24:19 Received zigbee message of type 'devInterview' with data '"0x000d6f00106cf0e0"'
  zigbee2mqtt:warn 2018-12-1 17:24:19 Message without device!
`

johnnyletrois, 2018-12-01:
More debug log:

`
  zigbee2mqtt:debug 2018-12-1 17:22:03 Received zigbee message of type 'devLeaving' with data '"0x000d6f00106cf0e0"'
  zigbee2mqtt:warn 2018-12-1 17:22:03 Message without device!
Sat, 01 Dec 2018 17:22:03 GMT zigbee-shepherd:msgHdlr IND <-- ZDO:leaveInd
2018-12-01T17:22:03.687Z serialport:unixRead waiting for readable because of code: EAGAIN
2018-12-01T17:22:03.687Z serialport:poller Polling for "readable"
2018-12-01T17:22:04.617Z serialport:poller received "readable"
2018-12-01T17:22:04.617Z serialport:bindings read
2018-12-01T17:22:04.617Z serialport:unixRead Starting read
2018-12-01T17:22:04.617Z serialport:unixRead Finished read 17 bytes
2018-12-01T17:22:04.617Z serialport:main binding.read finished
Sat, 01 Dec 2018 17:22:04 GMT cc-znp { sof: 254,
  len: 12,
  type: 'AREQ',
  subsys: 'ZDO',
  cmd: 'tcDeviceInd',
  payload: { nwkaddr: 52309, extaddr: '0x000d6f00106cf0e0', parentaddr: 0 },
  fcs: 20,
  csum: 20 }
2018-12-01T17:22:04.618Z serialport:main _read reading
2018-12-01T17:22:04.618Z serialport:bindings read
2018-12-01T17:22:04.618Z serialport:unixRead Starting read
Sat, 01 Dec 2018 17:22:04 GMT cc-znp:AREQ <-- ZDO:tcDeviceInd, { nwkaddr: 52309, extaddr: '0x000d6f00106cf0e0', parentaddr: 0 }
Sat, 01 Dec 2018 17:22:04 GMT zigbee-shepherd:msgHdlr IND <-- ZDO:tcDeviceInd
2018-12-01T17:22:04.618Z serialport:unixRead waiting for readable because of code: EAGAIN
2018-12-01T17:22:04.618Z serialport:poller Polling for "readable"
2018-12-01T17:22:05.046Z serialport:poller received "readable"
2018-12-01T17:22:05.046Z serialport:bindings read
2018-12-01T17:22:05.046Z serialport:unixRead Starting read
2018-12-01T17:22:05.046Z serialport:unixRead Finished read 18 bytes
2018-12-01T17:22:05.046Z serialport:main binding.read finished
Sat, 01 Dec 2018 17:22:05 GMT cc-znp { sof: 254,
  len: 13,
  type: 'AREQ',
  subsys: 'ZDO',
  cmd: 'endDeviceAnnceInd',
  payload: 
   { srcaddr: 52309,
     nwkaddr: 52309,
     ieeeaddr: '0x000d6f00106cf0e0',
     capabilities: 128 },
  fcs: 7,
  csum: 7 }
2018-12-01T17:22:05.046Z serialport:main _read reading
2018-12-01T17:22:05.047Z serialport:bindings read
2018-12-01T17:22:05.047Z serialport:unixRead Starting read
Sat, 01 Dec 2018 17:22:05 GMT cc-znp:AREQ <-- ZDO:endDeviceAnnceInd, { srcaddr: 52309, nwkaddr: 52309, ieeeaddr: '0x000d6f00106cf0e0', capabilities: 128 }
Sat, 01 Dec 2018 17:22:05 GMT zigbee-shepherd spinlock: false []
Sat, 01 Dec 2018 17:22:05 GMT zigbee-shepherd:request REQ --> ZDO:nodeDescReq
Sat, 01 Dec 2018 17:22:05 GMT cc-znp:SREQ --> ZDO:nodeDescReq, { dstaddr: 52309, nwkaddrofinterest: 52309 }
2018-12-01T17:22:05.048Z serialport:main _write 9 bytes of data
2018-12-01T17:22:05.048Z serialport:bindings write 9 bytes
Sat, 01 Dec 2018 17:22:05 GMT zigbee-shepherd:msgHdlr IND <-- ZDO:endDeviceAnnceInd
2018-12-01T17:22:05.048Z serialport:unixWrite Starting write 9 bytes offset 0 bytesToWrite 9
2018-12-01T17:22:05.048Z serialport:unixRead waiting for readable because of code: EAGAIN
2018-12-01T17:22:05.048Z serialport:poller Polling for "readable"
2018-12-01T17:22:05.049Z serialport:unixWrite write returned null 9
2018-12-01T17:22:05.049Z serialport:unixWrite wrote 9 bytes
2018-12-01T17:22:05.049Z serialport:unixWrite Finished writing 9 bytes
2018-12-01T17:22:05.049Z serialport:main binding.write write finished
2018-12-01T17:22:05.057Z serialport:poller received "readable"
2018-12-01T17:22:05.057Z serialport:bindings read
2018-12-01T17:22:05.057Z serialport:unixRead Starting read
2018-12-01T17:22:05.057Z serialport:unixRead Finished read 6 bytes
2018-12-01T17:22:05.057Z serialport:main binding.read finished
Sat, 01 Dec 2018 17:22:05 GMT cc-znp { sof: 254,
  len: 1,
  type: 'SRSP',
[... 614 lines cut ...]
   { groupid: 0,
     clusterid: 0,
     srcaddr: 52309,
     srcendpoint: 1,
     dstendpoint: 1,
     wasbroadcast: 0,
     linkquality: 97,
     securityuse: 0,
     timestamp: 16486478,
     transseqnumber: 0,
     len: 42,
     data: <Buffer 08 05 01 06 00 00 42 10 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 07 00 00 30 03 10 00 00 42 03 4e 2f 41 11 00 00 30 00> },
  fcs: 180,
  csum: 180 }
2018-12-01T17:22:19.124Z serialport:main _read reading
2018-12-01T17:22:19.124Z serialport:bindings read
2018-12-01T17:22:19.124Z serialport:unixRead Starting read
Sat, 01 Dec 2018 17:22:19 GMT cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 0, srcaddr: 52309, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 97, securityuse: 0, timestamp: 16486478, transseqnumber: 0, len: 42, data: <Buffer 08 05 01 06 00 00 42 10 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 07 00 00 30 03 10 00 00 42 03 4e 2f 41 11 00 00 30 00> }
Sat, 01 Dec 2018 17:22:19 GMT zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object]
Sat, 01 Dec 2018 17:22:19 GMT zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2018-12-01T17:22:19.127Z serialport:unixRead waiting for readable because of code: EAGAIN
2018-12-01T17:22:19.127Z serialport:poller Polling for "readable"
Sat, 01 Dec 2018 17:22:19 GMT zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object]
  zigbee2mqtt:debug 2018-12-1 17:22:19 Received zigbee message of type 'devInterview' with data '"0x000d6f00106cf0e0"'
  zigbee2mqtt:warn 2018-12-1 17:22:19 Message without device!`

johnnyletrois, 2018-12-01:
Database entries:

`{"id":10,"type":"EndDevice","ieeeAddr":"0x000d6f00106cf0e0","nwkAddr":20181,"manufId":49887,"epList":[1,2],"status":"online","joinTime":1543682785,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,32,1026,1280,2821],"outClusterList":[25],"clusters":{}},"2":{"profId":49887,"epId":2,"devId":263,"inClusterList":[0,1,3,2821,64582],"outClusterList":[3],"clusters":{}}},"_id":"VNEL4mnedVp7aoUl"}
{"id":10,"type":"EndDevice","ieeeAddr":"0x000d6f00106cf0e0","nwkAddr":20181,"manufId":49887,"manufName":"CentraLite","powerSource":"Battery","modelId":"3326-L","epList":[1,2],"status":"online","joinTime":1543682785,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,32,1026,1280,2821],"outClusterList":[25],"clusters":{}},"2":{"profId":49887,"epId":2,"devId":263,"inClusterList":[0,1,3,2821,64582],"outClusterList":[3],"clusters":{}}},"_id":"VNEL4mnedVp7aoUl"}
{"id":10,"type":"EndDevice","ieeeAddr":"0x000d6f00106cf0e0","nwkAddr":20181,"manufId":49887,"manufName":"CentraLite","powerSource":"Battery","modelId":"3326-L","epList":[1,2],"status":"online","joinTime":1543682785,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,32,1026,1280,2821],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"stackVersion":2,"hwVersion":1,"manufacturerName":"CentraLite","modelId":"3326-L","dateCode":"","powerSource":3,"locationDesc":"N/A","physicalEnv":0}},"genPowerCfg":{"dii
r":{"value":1},"attrs":{"batteryVoltage":30,"batterySize":2,"batteryQuantity":1,"batteryVoltMinThres":21,"batteryVoltThres1":22,"batteryVoltThres2":23,"batteryVoltThres3":24,"batteryAlarmState":0}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":1},"attrs":{}},"msTemperatureMeasurement":{"dir":{"value":1},"attrs":{"measuredValue":1966,"minMeasuredValue":-4000,"maxMeasuredValue":6000,"tolerance":150}},"ssIasZone":{"dir":{"value":1},"attrs":{"zoneState":0,"zoneType":13,"zoneStatus":34,"iasCieAddr":"0xffffffffffffffff","zoneId":255}},"haDiagnostic":{"dir":{"value":1},"attrs":{"macRxBcast":536870912,"macTxBcast":536870912,"macRxUcast":536870978,"macTxUcast":536871038,"macTxUcastRetry":0,"macTxUcastFail":0,"aPSRxBcast":3,"aPSTxBcast":2,"aPSRxUcast":32,"aPSTxUcastSuccess":26,"aPSTxUcastRetry":0,"aPSTxUcastFail":0,"routeDiscInitiated":0,"neighborAdded":1,"neighborRemoved":0,"neighborStale":0,"joinIndication":0,"childMoved":0,"nwkFcFailure":0,"apsFcFailure":0,"apsUnauthorizedKey":0,"nwkDecryptFailures":0,"apsDecryptFailures":0,"packetBufferAllocateFailures":0,"relayedUcast":0,"phyToMacQueueLimitReached":0,"packetValidateDropCount":0,"averageMacRetryPerApsMessageSent":1,"lastMessageLqi":255,"lastMessageRssi":-37}}}},"2":{"profId":49887,"epId":2,"devId":263,"inClusterList":[0,1,3,2821,64582],"outClusterList":[3],"clusters":{"64582":{"dir":{"value":1},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"haDiagnostic":{"dir":{"value":1},"attrs":{}}}}},"_id":"VNEL4mnedVp7aoUl"}
{"$$deleted":true,"_id":"VNEL4mnedVp7aoUl"}
{"id":10,"type":"EndDevice","ieeeAddr":"0x000d6f00106cf0e0","nwkAddr":54312,"manufId":49887,"epList":[1,2],"status":"online","joinTime":1543682984,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,32,1026,1280,2821],"outClusterList":[25],"clusters":{}},"2":{"profId":49887,"epId":2,"devId":263,"inClusterList":[0,1,3,2821,64582],"outClusterList":[3],"clusters":{}}},"_id":"p73ZD5HEJqJ5xyQr"}
{"id":10,"type":"EndDevice","ieeeAddr":"0x000d6f00106cf0e0","nwkAddr":54312,"manufId":49887,"manufName":"CentraLite","powerSource":"Battery","modelId":"3326-L","epList":[1,2],"status":"online","joinTime":1543682984,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,32,1026,1280,2821],"outClusterList":[25],"clusters":{}},"2":{"profId":49887,"epId":2,"devId":263,"inClusterList":[0,1,3,2821,64582],"outClusterList":[3],"clusters":{}}},"_id":"p73ZD5HEJqJ5xyQr"}
`

johnnyletrois, 2018-12-01:
I've added the following to devices.js:
                                                                                                           
`    {                                                                                                            
        zigbeeModel: ['3326-L'],                                                                                 
        model: '3326-L',                                                                                         
        vendor: 'Iris',                                                                                          
        description: 'Motion Sensor',                                                            
        supports: 'occupancy, temperature',                                                      
        fromZigbee: [],                                                                          
        toZigbee: [],                                                                            
    },`

johnnyletrois, 2018-12-01:
I'm not seeing anything else beyond the "Message without device!" below:

`zigbee2mqtt:debug 2018-12-1 17:24:19 Received zigbee message of type 'devInterview' with data '"0x000d6f00106cf0e0"' zigbee2mqtt:warn 2018-12-1 17:24:19 Message without device!`

mcsSolutions, 2018-12-01:
I tried https://github.com/koenkk/zigbee2mqtt/tree/osram_motion and installed it by replacing the lib and node_modules folders from the zip into my /opt/zigbee2mqtt install.  Installing the battery in the Iris Motion Sensor to attempt pairing produced the same result that I had in the original post with inability go the Node Descriptor within 10000 ms.

Should I use a different install process to take advantage of this branch?

johnnyletrois, 2018-12-02:
Made lots of progress yesterday and got it working. Will post details later when I get back to my laptop.  Thanks!

mcsSolutions, 2018-12-02:
I am a neophyte with git.  What do I need to do to take advantage of what has been done by johnnyletrois?    I was able to recognize the edits made in devices.js and fromZigbee.js and can port those, but my attempt to incorporate the osram build did not produce any change for me with the 3326-L.

johnnyletrois, 2018-12-03:
@mcssolutions  you might want to wait until the changes are incorporated by @koenkk in the master.  It’s reassuring that the iris devices do work though. 

I wish I bought more motion sensors and contact sensors when they were on sale.

johnnyletrois, 2018-12-10:
@koenkk I think these are now supported in the dev branch, correct? Will they make it to the next main release? Thanks!

Koenkk, 2018-12-10:
@johnnyletrois yes, everything that is in the dev branch will be in the next release. Can this be closed?

johnnyletrois, 2018-12-10:
@koenkk I believe this can be closed because we have confirmed the sensor pairs and works properly with the relevant changes.