Koenkk/zigbee2mqtt #601: problem with QBKG04LM - Xiaomi Aqara single key wired wall switch (Router)

poki123, 2018-11-16:
Hello,

I have problem to **set ON / OFF** the switch QBKG04LM - Xiaomi Aqara single key wired wall switch (Router). Reading the current state working great, signal is also great.

**The error is:**
  zigbee2mqtt:info 2018-11-16 21:04:13 Zigbee publish to '0x00158d0002c5124c', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:error 2018-11-16 21:04:43 Zigbee publish to '0x00158d0002c5124c', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: Timed out after 30000 ms

Can you solve this please?

**And full log:**
> zigbee2mqtt@0.2.0 start /opt/zigbee2mqtt
> node index.js

		Zigbee2mqtt requires node version >=8.11 10, you are running v8.12.0!

  zigbee2mqtt:info 2018-11-16 21:03:54 Logging to directory: '/opt/zigbee2mqtt/data/log/2018-11-16.21-03-54'
  zigbee2mqtt:debug 2018-11-16 21:03:54 Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 2018-11-16 21:03:55 Loaded state from file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:info 2018-11-16 21:03:55 Starting zigbee2mqtt version 0.2.0 (commit #dd21471)
  zigbee2mqtt:info 2018-11-16 21:03:55 Starting zigbee-shepherd
  zigbee2mqtt:info 2018-11-16 21:03:55 zigbee-shepherd started
  zigbee2mqtt:info 2018-11-16 21:03:55 Coordinator firmware version: '20180815'
  zigbee2mqtt:debug 2018-11-16 21:03:55 zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b0018ed3a5b","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20180815},"startTime":1542398635,"joinTimeLeft":0}
  zigbee2mqtt:info 2018-11-16 21:03:55 Currently 1 devices are joined:
  zigbee2mqtt:info 2018-11-16 21:03:55 unknown (0x00158d0002c5124c): QBKG04LM - Xiaomi Aqara single key wired wall switch (Router)
  zigbee2mqtt:warn 2018-11-16 21:03:55 `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 2018-11-16 21:03:55 Allowing new devices to join.
  zigbee2mqtt:warn 2018-11-16 21:03:55 Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 2018-11-16 21:03:55 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2018-11-16 21:03:55 Connecting to MQTT server at mqtt://localhost
  zigbee2mqtt:info 2018-11-16 21:03:56 zigbee-shepherd ready
  zigbee2mqtt:info 2018-11-16 21:03:56 Connected to MQTT server
  zigbee2mqtt:info 2018-11-16 21:03:56 MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 2018-11-16 21:03:56 MQTT publish: topic 'zigbee2mqtt/0x00158d0002c5124c', payload '{"state":"OFF","linkquality":162}'
  zigbee2mqtt:debug 2018-11-16 21:04:00 Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"61440":63700992,"onOff":1}}' of device 'lumi.ctrl_neutral1' (0x00158d0002c5124c)
  zigbee2mqtt:info 2018-11-16 21:04:00 New device with address 0x00158d0002c5124c connected!
  zigbee2mqtt:info 2018-11-16 21:04:00 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00158d0002c5124c"}'
  zigbee2mqtt:info 2018-11-16 21:04:00 MQTT publish: topic 'zigbee2mqtt/0x00158d0002c5124c', payload '{"state":"ON","linkquality":157}'
  zigbee2mqtt:debug 2018-11-16 21:04:00 Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.ctrl_neutral1' (0x00158d0002c5124c)
  zigbee2mqtt:debug 2018-11-16 21:04:03 Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"61440":63700992,"onOff":0}}' of device 'lumi.ctrl_neutral1' (0x00158d0002c5124c)
  zigbee2mqtt:info 2018-11-16 21:04:03 MQTT publish: topic 'zigbee2mqtt/0x00158d0002c5124c', payload '{"state":"OFF","linkquality":157}'
  zigbee2mqtt:debug 2018-11-16 21:04:03 Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.ctrl_neutral1' (0x00158d0002c5124c)
  zigbee2mqtt:debug 2018-11-16 21:04:13 Received MQTT message on 'zigbee2mqtt/0x00158d0002c5124c/set' with data '{"state":"ON"}'
  zigbee2mqtt:info 2018-11-16 21:04:13 Zigbee publish to '0x00158d0002c5124c', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:error 2018-11-16 21:04:43 Zigbee publish to '0x00158d0002c5124c', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: Timed out after 30000 ms
  zigbee2mqtt:debug 2018-11-16 21:04:55 Check online 0x00158d0002c5124c 0x00158d0002c5124c
  zigbee2mqtt:debug 2018-11-16 21:05:15 Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"1":3300,"3":29,"5":18,"8":9750,"10":0,"100":0,"101":0,"110":0,"111":255,"153":0,"155":0}}}' of device 'lumi.ctrl_neutral1' (0x00158d0002c5124c)
  zigbee2mqtt:debug 2018-11-16 21:05:55 Check online 0x00158d0002c5124c 0x00158d0002c5124c

paskyecijano, 2018-11-17:
I have the same problem with 0.2.0 version
[image]

jasimancas, 2018-11-17:
Other with this problem here!

Koenkk, 2018-11-19:
Continuing here: https://github.com/Koenkk/zigbee2mqtt/issues/593

closed 2018-11-19 by Koenkk