Koenkk/zigbee2mqtt #602: it does not work 50049 Paulmann SmartHome Yourled RGB Controller (on/off, brightness, color xy)

salopette, 2018-11-16:
Hello, I'm trying to pair a Paulmann 50049 controller, but that does not seem to work.

https://pastebin.com/Zrnw59Yk

database.db

https://pastebin.com/byRH3TQc

Koenkk, 2018-11-19:
Looks good, can you try following: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices

Koenkk, 2019-01-13:
Can you provide your `database.db`?

bochelork, 2019-01-14:
This is the Paulmann line from database.db. Let me know if you need more data.

`{"id":27,"type":"Router","ieeeAddr":"0x00158d0002931234","nwkAddr":24243,"manufId":4509,"manufName":"Paulmann Licht GmbH ","powerSource":"Mains (single phase)","modelId":"RGBW light ","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}},"lightLink":{"dir":{"value":1},"attrs":{}}}}},"_id":"XbnhHSBK1KuB2FUQ"}
`

Koenkk, 2019-01-14:
@bochelork should be fixed in the latest dev branch, can you check?

bochelork, 2019-01-14:
Checked, seems so work fine. Had some difficulties updating to dev but now everything is working like a charm.

Koenkk, 2019-01-15:
Great!

closed 2019-01-15 by Koenkk