Koenkk/zigbee2mqtt #603: Problem with home assistant integration

WMP, 2018-11-16:
Hello, i dont know how to read data from zigbee sensors in my HA. I have Synology NAS with docker, and i have 2 docker conteneirs: koenkk/zigbee2mqtt and homeassistant/home-assistant:latest .

This is my zigbee2mqtt configuration:
`homeassistant: true
permit_join: true
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://192.XXX.XXX.194'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password
serial:
  port: /dev/ttyACM0
devices:
  '0x00158d000233f9c5':
    friendly_name: 'telewizor'
    retain: false
  '0x00158d00024906c9':
    friendly_name: 'okolice schodow'
    retain: false
  '0x00158d0002490772':
    friendly_name: 'okno'
    retain: false
  '0x00158d0002391d57':
    friendly_name: 'gora'
    retain: false`

And my zigbee2mqtt log:
`2018-11-16 20:39:51 - info: Logging to directory: '/app/data/log/2018-11-16.20-39-51'
2018-11-16 20:39:55 - info: Starting zigbee2mqtt version 0.2.0 (commit #b0d3c2f)
2018-11-16 20:39:55 - info: Starting zigbee-shepherd
2018-11-16 20:39:59 - info: Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
2018-11-16 20:40:59 - info: Starting zigbee-shepherd
2018-11-16 20:41:01 - info: zigbee-shepherd started
2018-11-16 20:41:01 - info: Coordinator firmware version: '20180815'
2018-11-16 20:41:01 - info: Currently 4 devices are joined:
2018-11-16 20:41:01 - info: okolice schodow (0x00158d00024906c9): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
2018-11-16 20:41:01 - info: okno (0x00158d0002490772): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
2018-11-16 20:41:01 - info: gora (0x00158d0002391d57): unkown - undefined unknown (EndDevice)
2018-11-16 20:41:01 - info: telewizor (0x00158d000233f9c5): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
2018-11-16 20:41:01 - warn: `permit_join` set to  `true` in configuration.yaml.
2018-11-16 20:41:01 - warn: Allowing new devices to join.
2018-11-16 20:41:01 - warn: Set `permit_join` to `false` once you joined all devices.
2018-11-16 20:41:01 - info: Zigbee: allowing new devices to join.
2018-11-16 20:41:01 - info: Connecting to MQTT server at mqtt://192.XXX.XXX.194
2018-11-16 20:41:04 - info: zigbee-shepherd ready
2018-11-16 20:41:04 - info: Connected to MQTT server
2018-11-16 20:41:04 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
2018-11-16 20:41:04 - info: MQTT publish, topic: 'homeassistant/sensor/0x00158d00024906c9/temperature/config', payload: '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","json_attributes":["linkquality","battery","voltage"],"state_topic":"zigbee2mqtt/okolice schodow","availability_topic":"zigbee2mqtt/bridge/state","name":"okolice schodow_temperature"}'
2018-11-16 20:41:04 - info: MQTT publish, topic: 'homeassistant/sensor/0x00158d00024906c9/humidity/config', payload: '{"unit_of_measurement":"%","device_class":"humidity","value_template":"{{ value_json.humidity }}","json_attributes":["linkquality","battery","voltage"],"state_topic":"zigbee2mqtt/okolice schodow","availability_topic":"zigbee2mqtt/bridge/state","name":"okolice schodow_humidity"}'
2018-11-16 20:41:04 - info: MQTT publish, topic: 'homeassistant/sensor/0x00158d00024906c9/pressure/config', payload: '{"unit_of_measurement":"hPa","device_class":"pressure","value_template":"{{ value_json.pressure }}","json_attributes":["linkquality","battery","voltage"],"state_topic":"zigbee2mqtt/okolice schodow","availability_topic":"zigbee2mqtt/bridge/state","name":"okolice schodow_pressure"}'
2018-11-16 20:41:04 - info: MQTT publish, topic: 'homeassistant/sensor/0x00158d0002490772/temperature/config', payload: '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","json_attributes":["linkquality","battery","voltage"],"state_topic":"zigbee2mqtt/okno","availability_topic":"zigbee2mqtt/bridge/state","name":"okno_temperature"}'
2018-11-16 20:41:04 - info: MQTT publish, topic: 'homeassistant/sensor/0x00158d0002490772/humidity/config', payload: '{"unit_of_measurement":"%","device_class":"humidity","value_template":"{{ value_json.humidity }}","json_attributes":["linkquality","battery","voltage"],"state_topic":"zigbee2mqtt/okno","availability_topic":"zigbee2mqtt/bridge/state","name":"okno_humidity"}'
2018-11-16 20:41:04 - info: MQTT publish, topic: 'homeassistant/sensor/0x00158d0002490772/pressure/config', payload: '{"unit_of_measurement":"hPa","device_class":"pressure","value_template":"{{ value_json.pressure }}","json_attributes":["linkquality","battery","voltage"],"state_topic":"zigbee2mqtt/okno","availability_topic":"zigbee2mqtt/bridge/state","name":"okno_pressure"}'
2018-11-16 20:41:04 - info: MQTT publish, topic: 'homeassistant/sensor/0x00158d000233f9c5/temperature/config', payload: '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","json_attributes":["linkquality","battery","voltage"],"state_topic":"zigbee2mqtt/telewizor","availability_topic":"zigbee2mqtt/bridge/state","name":"telewizor_temperature"}'
2018-11-16 20:41:04 - info: MQTT publish, topic: 'homeassistant/sensor/0x00158d000233f9c5/humidity/config', payload: '{"unit_of_measurement":"%","device_class":"humidity","value_template":"{{ value_json.humidity }}","json_attributes":["linkquality","battery","voltage"],"state_topic":"zigbee2mqtt/telewizor","availability_topic":"zigbee2mqtt/bridge/state","name":"telewizor_humidity"}'
2018-11-16 20:41:04 - info: MQTT publish, topic: 'homeassistant/sensor/0x00158d000233f9c5/pressure/config', payload: '{"unit_of_measurement":"hPa","device_class":"pressure","value_template":"{{ value_json.pressure }}","json_attributes":["linkquality","battery","voltage"],"state_topic":"zigbee2mqtt/telewizor","availability_topic":"zigbee2mqtt/bridge/state","name":"telewizor_pressure"}'
2018-11-16 20:41:04 - info: MQTT publish, topic: 'zigbee2mqtt/okolice schodow', payload: '{"temperature":21.42,"linkquality":42,"humidity":46.44,"pressure":1013.13,"battery":100,"voltage":3045}'
2018-11-16 20:41:04 - info: MQTT publish, topic: 'zigbee2mqtt/okno', payload: '{"temperature":17.74,"linkquality":49,"humidity":56.1,"pressure":1012.8,"battery":100,"voltage":3035}'
2018-11-16 20:41:04 - info: MQTT publish, topic: 'zigbee2mqtt/telewizor', payload: '{"temperature":21.23,"linkquality":65,"humidity":47.1,"pressure":1021,"battery":42,"voltage":2985}'
2018-11-16 20:44:05 - info: MQTT publish, topic: 'zigbee2mqtt/okno', payload: '{"temperature":17.46,"linkquality":70,"humidity":56.1,"pressure":1012.8,"battery":100,"voltage":3035}'
2018-11-16 20:44:05 - info: MQTT publish, topic: 'zigbee2mqtt/okno', payload: '{"temperature":17.46,"linkquality":70,"humidity":56.87,"pressure":1012.8,"battery":100,"voltage":3035}'
2018-11-16 20:44:05 - info: MQTT publish, topic: 'zigbee2mqtt/okno', payload: '{"temperature":17.46,"linkquality":70,"humidity":56.87,"pressure":1012,"battery":100,"voltage":3035}'
`

This is my HA configuration:
`mqtt:
  discovery: true
  discovery_prefix: homeassistant
  birth_message:
    topic: 'hass/status'
    payload: 'online'
  will_message:
    topic: 'hass/status'
    payload: 'offline'`

When i go in browser to my HA i havent any data from sensor, where i should looking for this data?

WMP, 2018-11-17:
In home-assistant.log i have only one line:
`2018-11-16 21:36:43 WARNING (MainThread) [hbmqtt.mqtt.protocol.handler] BrokerProtocolHandler Unhandled exception in reader coro: IncompleteReadError('0 bytes read on a total of 1 expected bytes',)`

This is same that in #468 issue, now i try install mosquito mqtt server.

WMP, 2018-11-17:
Ok, with mosquitto and enable discovery on home assistant i have my sensors in HA.

gregorymilet, 2018-11-17:
Thanks for your updates!

I have the exact same issue as you had. Could you share your homeassistant and zigbee2mqtt config pls? :-)

Thanks!

WMP, 2018-11-17:
Yes, of course:

192.XXX.XXX.196 - mosquitto server

zigbee2mqtt:
```
homeassistant: true
permit_join: false
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://192.XXX.XXX.196'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password
serial:
  port: /dev/ttyACM0
devices:
  '0x00158d000233f9c5':
    friendly_name: 'salon i kuchnia'
    retain: false
  '0x00158d00024906c9':
    friendly_name: 'łazienka'
    retain: false
  '0x00158d0002490772':
    friendly_name: 'lodowka'
    retain: false
  '0x00158d0002391d57':
    friendly_name: 'góra'
    retain: false
```

Home Assistant, i dont write configuration in configuration.yml, but in web. I simple write ip 192.XXX.XXX.196 and select "Enable discovery".

mosquitto, default config without any changes.

andrexus, 2018-12-24:
I am a little bit confused about the topic name.
```
homeassistant/sensor/0x00158d0002490772/humidity/config
```
Should it really be something like **0x00158d0002490772** and not the friendly name?

WMP, 2018-12-24:
Sorry, i dont remenber, possible is that i using sed in this log.

Koenkk, 2018-12-24:
@andrexus the home assistant topic config should use a unique identifier which does't change.

andrexus, 2018-12-25:
Ok. Then this is not an issue. I have the save problem as @WMP - device successfully connected and constantly sending updates about state changes but it is not visible in Home Assistant.

closed 2019-02-23 by Koenkk