Koenkk/zigbee2mqtt #606: Ikea LED 1623G12 Error on 0.2.0

salopette, 2018-11-17:
problem with Ikea LED 1623G12 after update .

https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/issues/37

when downgrading to 0.1.8 it works normally without errors

salopette, 2018-11-24:
no idea?

Koenkk, 2018-11-24:
Can you provide the zigbee2mqtt log?

salopette, 2018-11-24:
Log is here:

https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin

Koenkk, 2018-11-24:
Thats not the zigbee2mqtt log but the domoticz plugin log.

salopette, 2018-11-25:
Today I have again made update to 0.2.0 and it works!

salopette, 2018-11-25:
Here is my Log

https://pastebin.com/g6dSmQd4

salopette, 2018-11-25:
here is the same Problem

https://github.com/Koenkk/zigbee2mqtt/issues/632

Koenkk, 2018-11-25:
Nice, thats indeed the same issue. Let's continue in #632, I mentioned a probably fix and a temporary workaround is also posted there.

closed 2018-11-25 by Koenkk