Koenkk/zigbee2mqtt #607: can't pair ikea control outlet and ikea on/off switch in zigbee2mqtt

picsou83, 2018-11-18:
Hi,

sorry for my bad english :-)
here's what i did : 
pair ikea control outlet and zigbee2mqtt is ok and i can control outlet
pair ikea on/off switch in zigbee2mqtt is ok (device not supported)

2018-11-18 09:06:26 - debug: Received zigbee message of type 'devStatus' with data '"online"' of device 'TRADFRI on/off switch' (0x000d6ffffe2f0e12)
2018-11-18 09:06:26 - warn: Device with modelID 'TRADFRI on/off switch' is not supported.

now i press for 10 second  on/off switch to pair with control outlet but after 5 seconds on/off switch stop flashing and not pair with outlet.

without zigbee2mqtt on/off switch pair success after 10 seconds 

Thanks in advance for your advice

UnrealKazu, 2018-11-19:
As far as I know the TRADFRI on/off switch (the one that optionally comes with the outlet, right?) is not supported in zigbee2mqtt yet. The outlet itself is supported, so this is why you can pair the outlet, but not the switch.

So pairing the on/off switch directly with zigbee2mqtt will not work. But you can try to add support for it using the wiki page (https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices), and then create a PR if you're up for it.

If you're talking about pairing the outlet to zigbee2mqtt and pairing the on/off switch directly to the outlet, I'm not sure but I think this should still work. But maybe someone else here can correct me on that one.

Koenkk, 2018-11-20:
@picsou83 yes! You can use the new extension api, example: https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/extensionTemplate.js

Create a new extension and add it to https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/controller.js#L94

picsou83, 2018-11-20:
arf ... it's a little complicated but Give a Man a Fish, and You Feed Him for a Day. Teach a Man To Fish, and You Feed Him for a Lifetime :-)

For zigbee2mqtt/lib/extension/softReset.js , I expect to find of zigbee-shepherd

`
shepherd.reset('soft',` function (err) {
    if (!err)
        console.log('reset successfully.');
});
`
but where is shepherd.reset('soft',` function in  zigbee2mqtt/lib/extension/softReset.js

Thanks

Koenkk, 2018-11-20:
You can get access to zigbee-shepherd in an extension with: `this.zigbee.shepherd`

picsou83, 2018-11-22:
@Koenkk It's possible to have removeDevice in zigbee2mqtt/lib/extension/removedevice.js and not in zigbee.js

It's a good example to use mqtt topic and shelpherd command this.shepherd.remove(deviceID, (error)) 

it would be easier to create another extension on the same principle

Koenkk, 2018-11-22:
No, shepherd functionality should be abstracted from the extensions. For debugging purposes you can use `this.zigbee.shepherd`.

picsou83, 2018-11-23:
Hi,

it's finish for ep.dump() , 

add in https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/controller.js
`const Extensiondump = require('./extension/persodump');`
`new Extensiondump(this.zigbee, this.mqtt, this.state, this.mqttPublishDeviceState),`

and the code for https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/extensionTemplate.js

`const logger = require('../util/logger');
const settings = require('../util/settings');
const zigbeeShepherdConverters = require('zigbee-shepherd-converters');

class persodump {
    constructor(zigbee, mqtt, state) {
        this.zigbee = zigbee;
        this.mqtt = mqtt;
        this.state = state;

        // Subscribe to topic.
        this.topic = `${settings.get().mqtt.base_topic}/bridge/persodump`;
        this.mqtt.subscribe(this.topic);
  
    }
 
     handleMQTTMessage(topic, message) {
        message = message.toString();
        if (topic === this.topic) {
		const deviceID = message;	
		const device = this.zigbee.shepherd._findDevByAddr(deviceID);
		const test = device.dump();
		
//        logger.info(`view device '${JSON.stringify(test)}' et'${message}'`); 
			const converted = JSON.stringify(test);
            this.mqtt.publish(`bridge/persodump/list`, converted, {});
            
            return true;
        }

        return false;
    }
    
}
module.exports = persodump;
   

`

now I'm going to prepare .bind(cId, dstEpOrGrpId[, callback])

question : (I am a newbie in node js)
How to have two variables with mqtt and messages 

`     handleMQTTMessage(topic, message) {
        message = message.toString();`

for
`var ep1 = shepherd.find('0x00124b0001ce4beb', 8);
var ep2 = shepherd.find('0x00124b00014a7dd2', 12);`

Thanks

Koenkk, 2018-11-23:
You can use [`JSON.parse`](https://www.w3schools.com/js/js_json_parse.asp) to convert the message to an object.

E.G. you can publish the message `{"eps": [8, 12]"}.

In JS
```js
message = JSON.parse(message.toString);
const eps = message.eps // eps will be an array [8, 12]
```

picsou83, 2018-11-25:
Very cool, 
I managed to create a group with the functionnals commands and it works, My TRADFRI on/off switch control my outlet.

to optimize all that I let you do it :-)

I added the tradfri control outlet on group 2 and 
i added the trafri on off switch  on group 2

and magic no bind to do

the code I used
````
const telecommande = '0x000d6ffffea4efc6';
           const ep2 = this.zigbee.shepherd.find(telecommande, 1);
                           
 ep2.functional('genGroups', 'add', {  groupid: 0x0002 , groupname: 'group2' } , function (err, data) {
  logger.info(err);    
        logger.info(data);    
});

````

and 

````
const prise =  '0x000d6ffffe2f0a14';        
    const ep1 = this.zigbee.shepherd.find(prise, 1);
    
                           
 ep1.functional('genGroups', 'add', {  groupid: 0x0002 , groupname: 'group2' } , function (err, data) {
  logger.info(err);    
        logger.info(data);    
});

````

aaamoeder, 2018-12-02:
actually bought this a few days ago. can't wait for the support to be integrated into the next update :) GREAT work guys !!!

closed 2018-12-05 by picsou83

Koenkk, 2018-12-05:
@picsou83 could you explain: *also with dummy device it's possible to have a return of status of a remote control with the group*?

aaamoeder, 2018-12-14:
Don't mean to nag, but is there any way I can get this working currently without knowledge of code ?

UnrealKazu, 2018-12-14:
@picsou83 Can you create a PR with your working changes?