Koenkk/zigbee2mqtt #608: Glegopto rgbcct after pairing

khzd, 2018-11-18:
Help

I get it paired but I cannot control it with HA:
after pairing (I had to add the ModelId in the DB) i got : 
2018-11-17 19:07:29 - info: rgbcct_01 (0x00124b001ad70f04): GL-C-008 - Gledopto Zigbee LED controller RGB + CCT (Router)

I got systematically :  
2018-11-17 19:07:30 - debug: Recieved mqtt message on topic 'zigbee2mqtt/rgbcct_01/set' with data '{"state": "ON"}'
2018-11-17 19:07:30 - info: Zigbee publish to '0x00124b001ad70f04', genOnOff - on - {} - null
2018-11-17 19:08:00 - error: Zigbee publish to '0x00124b001ad70f04', genOnOff - on - {} - null failed with error Error: Timed out after 30000 ms

in the log , I have some lines !!! 
 2018-11-17 19:11:31 - debug: Recieved zigbee message of type 'devInterview' with data '"0x00124b001ad70f04"'
2018-11-17 19:11:31 - debug: Recieved zigbee message of type 'devInterview' with data '"0x00124b001ad70f04"'
2018-11-17 19:11:31 - debug: Recieved zigbee message of type 'devInterview' with data '"0x00124b001ad70f04"'
2018-11-17 19:11:31 - debug: Recieved zigbee message of type 'devInterview' with data '"0x00124b001ad70f04"'
2018-11-17 19:11:31 - debug: Recieved zigbee message of type 'devInterview' with data '"0x00124b001ad70f04"'
2018-11-17 19:11:31 - debug: Recieved zigbee message of type 'devInterview' with data '"0x00124b001ad70f04"'

Koenkk, 2018-11-19:
Could you provide the contents of your `database.db`?

khzd, 2018-11-19:
Hello
I m fare from my pc, will send it tomorrow morning Paris time
May I extract only lines concerning this device or do you need full dB ?
Cheers

Koenkk, 2018-11-20:
Only the lines of this device is enough.

andreasbrett, 2018-11-20:
I have similar issues pairing my GLEDOPTO RGBW controller to my CC2531. I switched from the native openHAB zigbee binding (which does not support several devices and needs a different firmware on the CC2531) to zigbee2mqtt. All devices that did not work with the openHAB binding now work with zigbee2mqtt. Unfortunately the GLEDOPTO did work with the openHAB binding but now fails to pair with the CC2531.

Here are the GLEDOPTO specific lines from database.db. Hope that helps

`{"id":14,"type":"Router","ieeeAddr":"0x00124b0019c60150","nwkAddr":26075,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GLEDOPTO","epList":[10,11,13],"status":"online","joinTime":1542680665,"endpoints":{"10":{"profId":49246,"epId":10,"devId":256,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}}}},"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"ewk5B9WAKJqEgVWm"}

{"id":14,"type":"Router","ieeeAddr":"0x00124b0019c60150","nwkAddr":26075,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GLEDOPTO","epList":[10,11,13],"status":"offline","joinTime":null,"endpoints":{"10":{"profId":49246,"epId":10,"devId":256,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}}}},"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"ewk5B9WAKJqEgVWm"}
`

Koenkk, 2018-11-20:
@andreasbrett what do you mean with fails to pair? Because if it fails to pair it shouldn't show up in the `database.db`.

andreasbrett, 2018-11-20:
@Koenkk you're right, my bad, I'm still recovering from a flu :-/ So yes, it is paired but I can't seem to communicate with the device. I don't receive any mqtt message nor can I send a color to it.

I'm very new to zigbee2mqtt so please forgive me if that's a stupid question but how do I find out the mqtt topics for such a device that won't send json payload by itself? I haven't found a way to trigger it (the device has no buttons that I could press). The other zigbee binding had a way of somehow automatically determining the possible "channels"/attributes of a zigbee device (even for the GLEDOPTO). Is there something like this or how will I find out where to push the RGB values for this LED controller?

khzd, 2018-11-20:
@Koenkk 
here is my record in DB 
{"id":48,"type":"Router","ieeeAddr":"0x00124b001ad70f04","nwkAddr":849,"manufId":0,"modelId":"GLEDOPTO","epList":[11,12,13],"status":"offline","joinTime":null,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"12":{"profId":260,"epId":12,"devId":258,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"E5To3y4TxscJUH4e"}

I've added manually "modelId":"GLEDOPTO" according of previous records,  is the record correct?
Many thanks for your help.

khzd, 2018-11-21:
Hello @andreasbrett
Any progress frm your case?
Cheers

andreasbrett, 2018-11-21:
Hey @khzd,
Unfortunately not. It's paired and part of my zigbee mesh (taking over its role as zigbee router) but I can't seem to send or receive anything from it. Problem is that it's not sending data itself but rather only reacting when I send payloads. I tried several different payloads but it only reacts upon state and brightness. It's neither changing brightness nor turning off or on but at least it reports back a "state=on". 

I think this RGB controller is not yet fully supported. I know from the openhab zigbee binding that this particular controller I have (RGBW which is a bit different from the RGB+CCT) reports back as 2 devices (endpoints?). Device 1 is for controlling the W part and device 2 for controlling the RGB part. I have this controller on an RGB led strip so it's very much possible that I don't see any changes because I'm only controlling device 1 and that won't don't much to my RGB-only led strip... 

Hope to get this working soon but I'm not sure how. Maybe the devs can help us out debugging this scenario. There are several different GLEDOPTO controllers and I think they seem to have the same model number (mine is the GL-C-008).

khzd, 2018-11-21:
Hello @andreasbrett,

I got a second one this evening from Aliexpress and WORK **like a charm** :)
The pairing was quite long:
I stopped all my CC2530 and miracle (probably not linked) and the pairing was full ( ie with all attributes) and connection to HA was just amazing.
THe fist one is alwas KO and the record is strangly different!!!

there is probably a release

KO : GLEDOPTO : 
{"id":48,"type":"Router","ieeeAddr":"0x00124b001ad70f04","nwkAddr":849,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GLEDOPTO","epList":[11,12,13],"status":"online","joinTime":1542821258,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"12":{"profId":260,"epId":12,"devId":258,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"E5To3y4TxscJUH4e"}
the OK one : (strangly longer + 1 in the epList)
{"id":48,"type":"Router","ieeeAddr":"0x00124b001ad70f04","nwkAddr":849,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GLEDOPTO","epList":[11,12,13],"status":"offline","joinTime":null,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"12":{"profId":260,"epId":12,"devId":258,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"E5To3y4TxscJUH4e"}

lets compare your results?

cheers

andreasbrett, 2018-11-22:
My database output is some posts above. So is your new (and working) GLEDOPTO device any different from the non-working one? GLEDOPTO sells 4 different models:

[RGB+CCT](https://ae01.alicdn.com/kf/HTB1w3NAX6nuK1RkSmFPq6AuzFXaA/zigbee-Zll-link-smart-strip-light-rgb-rgbw-controller-DC12V-24V-zigbee-rgb-APP-control-compatible.jpg_640x640.jpg)
[RGBW](https://ae01.alicdn.com/kf/HTB1Bk8tX5nrK1RjSsziq6xptpXaC/GLEDOPTO-ZIGBEE-bridge-Led-Controller-RGBW-dimmer-strip-Controller-DC12-24V-compatible-with-LED-echo-zll.jpg_640x640.jpg)
[WW/CW](https://ae01.alicdn.com/kf/HTB18.abX6zuK1RjSspeq6ziHVXat/GLEDOPTO-ZIGBEE-bridge-Led-Controller-ww-cw-dimmer-strip-Controller-DC12-24V-zll-standard-led.jpg_640x640.jpg)
[LED Dimmer](https://ae01.alicdn.com/kf/HTB1G0ieX5nrK1RjSsziq6xptpXal/GLEDOPTO-ZIGBEE-bridge-Led-Controller-ww-cw-dimmer-strip-Controller-DC12-24V-zll-standard-led.jpg_640x640.jpg)

Checking your results I see your 2 devices show different endpoints. The working one shows (11, 12, 13) and the other one (11, 13). Mine shows (10, 11, 13). The only endpoints correlating over all 3 devices are 11 and 13. 

Can you post some output of your controller? Would be interesting to see the payload for your controller.

khzd, 2018-11-22:
hi,
i have both devices the RGB+CCT (6 pins)  but both produces diffrent records in the DB:
**the OK one** : 
{"id":50,"type":"Router","ieeeAddr":"0x00124b0019c606cd","nwkAddr":2298,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GLEDOPTO","epList":**[11,13]**,"status":"online","joinTime":1542834365,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"lKvtjnR82PWSUwb5"}
**and the KO one**
{"id":48,"type":"Router","ieeeAddr":"0x00124b001ad70f04","nwkAddr":4175,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GLEDOPTO","epList":**[11,12,13]**,"status":"online","joinTime":1542829896,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"12":{"profId":260,"epId":12,"devId":258,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"E5To3y4TxscJUH4e"}

**and yours is (KO)** 
{"id":14,"type":"Router","ieeeAddr":"0x00124b0019c60150","nwkAddr":26075,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GLEDOPTO","epList":**[10,11,13]**,"status":"offline","joinTime":null,"endpoints":{"10":{"profId":49246,"epId":10,"devId":256,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}}}},"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"ewk5B9WAKJqEgVWm"}

the difference is probably network topology? @Koenkk?

Any help from devs team ?
cheers

khzd, 2018-11-23:
@Koenkk? any help ?

Koenkk, 2018-11-23:
For the one with `[11,12,13]`. does it work when putting after: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L1353

```
```

khzd, 2018-11-24:
Hi @Koenkk , @andreasbrett 
**yes it work like a charm :) thanks.**
as requested, I put in place ep: {'': 12},  after  https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L1353,
What i did to preserve the one with [11,13],  I create a new section 
    {
        zigbeeModel: ['**GLEDOPTOZ**'],
        model: 'GL-C-008',
        vendor: 'Gledopto',
        description: 'Zigbee LED controller RGB + CCT',
        supports: generic.light_onoff_brightness_colortemp_colorxy().supports,
        fromZigbee: generic.light_onoff_brightness_colortemp_colorxy().fromZigbee,
        toZigbee: generic.light_onoff_brightness_colortemp_colorxy().toZigbee,
        ep: {'':12},
    }, 
and modify DB accordingly : replace GLEDOPTO by GLEDOPTOZ in the according record

if not the one with [11,13] stops to work.

how can I process now to preserve this new update?

Thanks again for your supports.
Cheers

Koenkk, 2018-11-25:
Great, as we cannot distinguish the device based on it's model we need to dynamically determine the ep. I will update zigbee2mqtt with this.

Koenkk, 2018-11-27:
Should be supported now in the dev branch (without making any changes yourself). 

@khzd can you verify?

andreasbrett, 2018-11-27:
> Should be supported now in the dev branch (without making any changes yourself).
> 
> @khzd can you verify?

Should this also work for my version of the GLEDOPTO (the RGBW one with EPs 10, 11 and 13 rather than 11, 12, 13 or 11, 13)? I just switched to branch dev and paired it but it's still not working.

Koenkk, 2018-11-28:
Can you add  

```js
            if (device.epList.toString() === '10,11,13') {
                return {'': 11};
            }
```    

here  https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L1374 and check if it works?

andreasbrett, 2018-11-28:
Yes, that did the trick. It's working now. Thank you very much. One thing to note though: I think it would be best to change the description of this device from "Zigbee LED controller RGB + CCT" to something more generic like "Zigbee LED controller RGB + CCT / RGBW / WWCW / Dimmer" as there are 4 versions and we now know that already 2 of them share the same model number (GL-C-008).

https://www.aliexpress.com/item/GLEDOPTO-ZIGBEE-Led-Controller-RGB-CCT-WW-CW-zigbee-controller-LED-DC12-24V-LED-strip-controller/32858603964.html

Koenkk, 2018-11-28:
Merged, thanks!

closed 2018-11-28 by Koenkk

khzd, 2018-12-01:
Yes it work for both , thanks