Koenkk/zigbee2mqtt #609: NetworkMap bug?

jasimancas, 2018-11-19:
This is my map:
[image]

Why are some devices not linked?

lolorc, 2018-11-19:
good question, I have the same problem here.
If you repair those devices, they'll come ok. But I'd also say there's an issue with the network map (or zigbee-sheperd)

apeinspace, 2018-11-20:
I may have a similar problem - either my network map is broken or the routing in my zigbee network has turned in a complete mess for days now.

wether related or not I dont know, but I remember this situation appearing around the time of the upgrade to 0.2.0.

Can anyone point me into a direction on how to troubleshoot a routing problem without having to trust the network map ? All ideas appreciated. thx

jasimancas, 2018-11-20:
> My problem seems solved - after removing a router via MQTT from the bridge the entire network recovered and my network map now looks normal again. I chose to believe it triggered a "reset" of some sort ... is that possible ?
> 
> Of course I cant say for sure wether just the network map or the network itself had a problem ... ?

Number of devices?

Can you post the map?

apeinspace, 2018-11-21:
this is how it looks now:

[image]

Before I did delete the router, I hade 5-6 sensors without connection, the coordinator had 5 connections and one router was holding all other connections. that router was the one with very low link quality to almost all sensors, as I only used it to extend the network into the garage.

The picture now more or less looks like I would expect it, based on the proximity/placement of the routers and sensors.

its weird though, that all my routers are marked as "offline" although they work just fine.

does that usually happen with you guys too ?

jasimancas, 2018-11-21:
My problem is that my routers don't seem to "route", everything always connects to the coordinator, and I don't know what to do, xiaomi problem maybe?

Anyone have this problem too?

apeinspace, 2018-11-21:
delete a router from the bridge and observe whats happening in the network map. It did the trick for me and the network "reset" the routing.

jasimancas, 2018-11-23:
> delete a router from the bridge and observe whats happening in the network map. It did the trick for me and the network "reset" the routing.
> 
> btw: Im sure there must be a more elegant way - but I simply dont know another way

Same result.

jasimancas, 2018-11-26:
[image]

Two devices without route, and all to coordinator?

Koenkk, 2018-11-27:
The network map is indeed still buggy and this has to be fixed. But to be honest, currently it is not high on my priority list.

jasimancas, 2018-11-27:
No problem, thanks!

closed 2018-11-27 by jasimancas