Koenkk/zigbee2mqtt #611: Tradfri ICPSHC24-10EU-IL-1 pairing

mgrefsrud, 2018-11-19:
I am using the Zigbee2mqtt addon in hassio and cc2531, I have added Xiaomi sensors with no issues, but now i am trying to add a Ikea Tradfri ICPSHC24-10EU-IL-1. But it will not Pair. I removed all other devices and still no help. 
I see no entry in the log, only premit join turning on. 
As far as i have understood this device should work, but i am not able to pair it. 
I have version 20180815 on the coordinator (I think cc2531)
Do i need the driver to be connected to a tradfri remote to pair it, or can it pair directly
I might be doing it wrong, but i really need help.

Koenkk, 2018-11-19:
Do you keep it close to the CC2531 while pairing?

mgrefsrud, 2018-11-19:
Yes, i have in laying on the driver. But what is the sequense to set the Ikea device to pair?

Koenkk, 2018-11-19:
For bulbs it's: https://www.youtube.com/watch?v=mJm9YpPrGzk don't know about the driver though, you have to check the manual.

mgrefsrud, 2018-11-19:
The manual says only a short press with a pin in the reset hole. But i have tried now short press, long press, hold. And it does not seem to try to pair. I also have changed the unit at ikea today in case i had a faulty one. As far as i have read there are someone who have paired this with this solution, but i can't understand how.

Koenkk, 2018-11-20:
@kirovilya do you know anyone who has this device?

kirovilya, 2018-11-20:
no

Koenkk, 2018-11-20:
@mgrefsrud when running with `DEBUG=* npm start` do you see any output after factory reset?

mgrefsrud, 2018-11-20:
I had a friend try today with his similar setup, and it worked on the first try. So something is wrong with my setup. I will reload the fw on the cc2531 tomorrow and try again. The reset for this device is as the mansion says one short press on the reset with a small pin. I will let you know how it goes as soon as I have tested again.

knutfh, 2018-11-20:
I work with mgrefsrud, so I brought his new driver home today to test with my setup. It paired immediately with a short click on the reset button. Probably nothing wrong with zigbee2mqtt. Will 

This is the STOUT log from the official HASS.IO container when I did the pairing:
`[32m  zigbee2mqtt:info[39m 2018-11-20 19:32:46 Zigbee: allowing new devices to join.
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:02 Connecting with device...
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:02 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:02 Connecting with device...
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:02 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:02 Connecting with device...
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:02 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:03 Connecting with device...
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:03 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:03 Connecting with device...
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:03 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:03 Connecting with device...
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:03 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:03 Connecting with device...
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:03 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:03 Connecting with device...
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:03 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:04 Connecting with device...
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:04 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:04 Connecting with device...
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:04 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:04 Device incoming...
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:04 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:04 New device with address 0xd0cf5efffe2ed49a connected!
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:04 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0xd0cf5efffe2ed49a"}'
[32m  zigbee2mqtt:info[39m 2018-11-20 19:33:04 MQTT publish, topic: 'homeassistant/light/0xd0cf5efffe2ed49a/light/config', payload: '{"brightness":true,"platform":"mqtt_json","command_topic":"zigbee2mqtt/0xd0cf5efffe2ed49a/set","state_topic":"zigbee2mqtt/0xd0cf5efffe2ed49a","availability_topic":"zigbee2mqtt/bridge/state","name":"0xd0cf5efffe2ed49a","unique_id":"0xd0cf5efffe2ed49a_light_zigbee2mqtt"}'`

Koenkk, 2018-11-20:
@knutfh thanks, would you mind updating the wiki with this information? (https://github.com/Koenkk/zigbee2mqtt/wiki/Pairing-devices#ikea-tradfri)

knutfh, 2018-11-20:
@Koenkk I'll update the wiki when we have a definite explaination on why it does not work for @mgrefsrud.

knutfh, 2018-11-20:
@Koenkk I've updated the wiki a short pairing description for the drivers.

mgrefsrud, 2018-11-21:
After doing a new FW loading (with the help of @knutfh )on the cc2531, the driver paired on the first try. So my problem was probably something with the FW. Even though i could pair the xiaomi devices. You can close the Issue.
Thank for the help

closed 2018-11-22 by Koenkk

rbrt86, 2021-01-04:
I know this is an old thread but I've spend **hours** on this driver to get it working. Just wanted to share what helped me:

the button seems unresponsive, I tried long, short and pressing various amounts of times. I thought it was broken, however after a while it worked.

The following worked for me:

1. unplug the driver
2. **disable** allowing new devices to join in zigbee2mqtt
3. plug in the driver, wait about 2 seconds
4. allow new devices to join in zigbee2mqtt
5. wait untill the 'interviewing xxxxxx' record shows up.
6. **short**-press the button on the driver. Holding it will make the driver flash the lights, but this is NOT what you want. Simply press once short

the important note here is that the driver must be plugged in briefly (1-2 seconds) before zigbee2mqtt starts interviewing

wollo, 2021-02-13:
Thank you very much! This saved my day and should be included in the wiki! It has also to be mention, that between 5 and 6 there should be less than 10 seconds!

Koenkk, 2021-02-13:
@wollo or @rbrt86 could you make a pr for this by clicking the pencil icon in the right here? https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/ICPSHC24-10EU-IL-1.md

piasekps, 2021-02-13:
Where should I look for point 5? 'interviewing xxxxxx' ??

razmoket89, 2021-08-11:
> I know this is an old thread but I've spend **hours** on this driver to get it working. Just wanted to share what helped me:
> 
> the button seems unresponsive, I tried long, short and pressing various amounts of times. I thought it was broken, however after a while it worked.
> 
> The following worked for me:
> 
> 1. unplug the driver
> 2. **disable** allowing new devices to join in zigbee2mqtt
> 3. plug in the driver, wait about 2 seconds
> 4. allow new devices to join in zigbee2mqtt
> 5. wait untill the 'interviewing xxxxxx' record shows up.
> 6. **short**-press the button on the driver. Holding it will make the driver flash the lights, but this is NOT what you want. Simply press once short
> 
> I'm not certain at this point, but if step 5 never shows up, you might need to briefly unplug the driver and plug it back in quickly
> 
> the important note here is that the driver must be plugged in briefly (1-2 seconds) before zigbee2mqtt starts interviewing

Hi,

I'm following all the steps you mentionned but nothing... I successfully paired it twice but was recognized as unsupported device. Any idea ? It's freaking frustrating considering how easy things are usually.

rbrt86, 2021-08-11:
@razmoket89 sorry I only had 'failed interview' issues untill I found that workaround.
Are you sure the IKEA driver is the 10EU edition?
It seems you might have a 10NA [like this guy](https://github.com/Koenkk/zigbee2mqtt/issues/1984) which is also a driver but cannot be paired to a gateway directly it seems?

razmoket89, 2021-08-11:
> @razmoket89 sorry I only had 'failed interview' issues untill I found that workaround.
> Are you sure the IKEA driver is the 10EU edition?
> It seems you might have a 10NA [like this [...]

I dont have the 10EU but the 30EU. I checked before posting that it was the zigbee version, same than yours but 30w instead of 10w.

sieukrem, 2021-11-21:
Thank you @rbrt86 !
I spent hours searching for the solution. I tried also to follow your suggestion and failed multiple times. 

But the timing is really important, not longer than 2sec after plugging in to start allowing joining and directly after green popup about interviewing to push the reset.

DecksLabs, 2022-08-05:
I know it's reviving a very old issue but to whoever that struggle to pair this device, procedure that worked for me was:

0. Have the driver **disconnected** from mains
1. Enable join
2. Connect the driver to mains, pairing begins immediately
3. Keep **short**-pressing (LEDs can't blink, blinking means that the driver has been reset to factory state) the reset on the device until it is successfully configured 

If you failed to pair it before, unplug it, plug it back, long press the reset on the driver so that the LEDs blink. Proceed with above procedure. This is kinda similar to the pairing of the Xiaomi sensors in the early days. I had this issue with all the drivers that I got (3 units), all came with very old firmware out of the box: 20170529 ver 1.2.245, maybe with newer it's easier to pair it?

Without the short presses it was failing with:
` node descriptor request failed for "0x...."`

MarvAmBass, 2022-10-15:
@DecksLabs thanks mate, I just came back from ikea buying three new devices which had exactly the same problem as my old ones (more than on year newly packaged in my cellar)

thanks to your description I was able to pair them in no time!

can we please add this to the wiki - it feels like a joke what's written there to pair those drivers

MarvAmBass, 2022-10-15:
okay I played with it a bit more and I think I know why there a slightly different approaches here.

this is my more generic approach which should help everyone

NB: you HAVE to put the device VERY close to the zigbee device, otherwise it won't be able to detect.

If you struggle to pair this device, the following procedure should help getting it paired:

If you have troubles pairing this device, note that you should always start with a freshly powered on device. Even when you reset the device you should switch the power off afterwards. And retry powering on the device.

Timing is key, I'd advice you to disable zigbee2mqtt auto join and wait a few seconds before each try.

The device is only able to pair for a short period of time after powering it on. After powering on the device you should immediately keep short-pressing (LEDs can't blink, blinking means that the driver has been reset to factory state) the reset on the device until it is successfully configured / you see a green pairing notification in your zigbee2mqtt server.

I'd guess that depending on your zigbee hardware you need to enable zigbees autojoin mode between -3 to 3 seconds to the point you power on the device. I used a conbee II usb stick and had to enable autojoin -1 to 0 seconds before powering on the device, and I immediately started short pressing (put the pin inside the hole before powering on) Maybe you need to play around with the timing.

If you didn't see any connection / pairing notifications within 5 to 10 seconds you can try again. I always powered off, powered on, reset with long push, powered off, and restartet the process.

I've tried this procedure with models in plastic packaging from around 2020 and newer models in paper packaging from 2022. Both came with the firmware 1.2.245 -> I don't know if it gets easier with newer firmware versions

Wifi from your Notebook might interfere with the pairing.