Koenkk/zigbee2mqtt #612: QBCZ11LM - Error in "power"

jasimancas, 2018-11-19:
Hi, i have some QBCZ11LM, but in one of this the power value is 0 all time (enchufe_termo).

I repair the device 3 times, and the first ON ok, but, restart z2m docker and power=0 all time.

> 
  zigbee2mqtt:info 2018-11-19 20:59:20 MQTT publish: topic 'zigbee2mqtt/enchufe_ordenador', payload '{"state":"ON","power":211.75,"voltage":null,"consumption":203.27,"temperature":27,"linkquality":49}',

  zigbee2mqtt:info 2018-11-19 20:59:17 MQTT publish: topic 'zigbee2mqtt/enchufe_termo', payload '{"state":"ON","power":0,"voltage":234,"consumption":498.56,"temperature":23,"linkquality":39}',

  zigbee2mqtt:info 2018-11-19 20:59:17 MQTT publish: topic 'zigbee2mqtt/enchufe_termo', payload '{"state":"ON","power":0,"voltage":234,"consumption":498.56,"temperature":23,"linkquality":52}',

  zigbee2mqtt:info 2018-11-19 21:00:26 MQTT publish: topic 'zigbee2mqtt/enchufe_termo', payload '{"state":"ON","power":0,"voltage":234,"consumption":498.56,"temperature":23,"linkquality":39}'

Koenkk, 2018-11-20:
- Does it work for your other QBCZ11LM?
- Can you provide the logging (so we can look into all reported attributes)?

jasimancas, 2018-11-20:
@Koenkk I'm going to reinstall all my system, and I wanted to know if you could recommend configurations for optimal operation, to see if I fix bugs.

What system do you use?

Koenkk, 2018-11-20:
I use an Intel Nuc (NUC6CAYH) with:
- 128GB SSD 
- 8GB ram
- CoreOS as operating system

jasimancas, 2018-11-20:
And MQTT, Z2M, and HA in Docker?

Koenkk, 2018-11-21:
Everything in Docker, using mosquitto as MQTT broker.

jasimancas, 2018-11-21:
> Everything in Docker, using mosquitto as MQTT broker.

OK, I'll set up the same stage as you, see if I can work something out.
Can the problem come from here? #609 that xiaomi does not route.

Thank you very much for everything, it's a very nice project and I'm very interested.

Koenkk, 2018-11-22:
No, that problem probably lies in the device itself.

jasimancas, 2018-11-26:
Solved, see #254

closed 2018-11-26 by jasimancas