Koenkk/zigbee2mqtt #613: Cannot get the Node Descriptor of the Device on CC2530

qoobaa, 2018-11-20:
I've flashed my CC2530 with the latest firmware (I tried 3 different firmwares so far), and I'm getting `Cannot get the Node Descriptor of the Device` when I'm trying to pair WXKG01LM or MCCGQ01LM.

```
  serialport:poller received "readable" +44s
  serialport:bindings read +44s
  serialport:unixRead Starting read +44s
  serialport:unixRead Finished read 18 bytes +0ms
  serialport:main binding.read finished +44s
  cc-znp { sof: 254,
  cc-znp   len: 13,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'ZDO',
  cc-znp   cmd: 'leaveInd',
  cc-znp   payload: 
  cc-znp    { srcaddr: 38608,
  cc-znp      extaddr: '0x00158d00016c3861',
  cc-znp      request: 0,
  cc-znp      removechildren: 0,
  cc-znp      rejoin: 0 },
  cc-znp   fcs: 107,
  cc-znp   csum: 107 } +44s
  serialport:main _read reading +4ms
  serialport:bindings read +5ms
  serialport:unixRead Starting read +5ms
  cc-znp:AREQ <-- ZDO:leaveInd, { srcaddr: 38608, extaddr: '0x00158d00016c3861', request: 0, removechildren: 0, rejoin: 0 } +3ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:leaveInd +0ms
  serialport:unixRead waiting for readable because of code: EAGAIN +2ms
  serialport:poller Polling for "readable" +8ms
  serialport:poller received "readable" +2s
  serialport:bindings read +2s
  serialport:unixRead Starting read +2s
  serialport:unixRead Finished read 17 bytes +0ms
  serialport:main binding.read finished +2s
  cc-znp { sof: 254,
  cc-znp   len: 12,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'ZDO',
  cc-znp   cmd: 'tcDeviceInd',
  cc-znp   payload: { nwkaddr: 6847, extaddr: '0x00158d00016c3861', parentaddr: 0 },
  cc-znp   fcs: 138,
  cc-znp   csum: 138 } +2s
  serialport:main _read reading +4ms
  serialport:bindings read +6ms
  serialport:unixRead Starting read +5ms
  cc-znp:AREQ <-- ZDO:tcDeviceInd, { nwkaddr: 6847, extaddr: '0x00158d00016c3861', parentaddr: 0 } +3ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:tcDeviceInd +0ms
  serialport:unixRead waiting for readable because of code: EAGAIN +2ms
  serialport:poller Polling for "readable" +8ms
  serialport:poller received "readable" +266ms
  serialport:bindings read +268ms
  serialport:unixRead Starting read +267ms
  serialport:unixRead Finished read 18 bytes +0ms
  serialport:main binding.read finished +270ms
  cc-znp { sof: 254,
  cc-znp   len: 13,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'ZDO',
  cc-znp   cmd: 'endDeviceAnnceInd',
  cc-znp   payload: 
  cc-znp    { srcaddr: 6847,
  cc-znp      nwkaddr: 6847,
  cc-znp      ieeeaddr: '0x00158d00016c3861',
  cc-znp      capabilities: 128 },
  cc-znp   fcs: 165,
  cc-znp   csum: 165 } +271ms
  serialport:main _read reading +5ms
  serialport:bindings read +6ms
  serialport:unixRead Starting read +6ms
  cc-znp:AREQ <-- ZDO:endDeviceAnnceInd, { srcaddr: 6847, nwkaddr: 6847, ieeeaddr: '0x00158d00016c3861', capabilities: 128 } +3ms
  zigbee-shepherd spinlock: false [] +1ms
[... 126 lines cut ...]
  serialport:unixWrite wrote 9 bytes +0ms
  serialport:unixWrite Finished writing 9 bytes +0ms
  serialport:main binding.write write finished +3ms
  serialport:poller received "readable" +10s
  serialport:bindings read +14ms
  serialport:unixRead Starting read +10s
  serialport:unixRead Finished read 6 bytes +0ms
  serialport:main binding.read finished +12ms
  cc-znp { sof: 254,
  cc-znp   len: 1,
  cc-znp   type: 'SRSP',
  cc-znp   subsys: 'ZDO',
  cc-znp   cmd: 'nodeDescReq',
  cc-znp   payload: { status: 0 },
  cc-znp   fcs: 102,
  cc-znp   csum: 102 } +17ms
  serialport:main _read reading +2ms
  serialport:bindings read +2ms
  serialport:unixRead Starting read +2ms
  cc-znp:SRSP <-- ZDO:nodeDescReq, { status: 0 } +1ms
  serialport:unixRead waiting for readable because of code: EAGAIN +1ms
  serialport:poller Polling for "readable" +3ms
  zigbee2mqtt:error 2018-11-20 06:33:49 Cannot get the Node Descriptor of the Device: 0x00158d00016c3861 (Error: Timed out after 10000 ms)
```
```

Koenkk, 2018-11-20:
Please try following the pairing procedure from: https://github.com/Koenkk/zigbee2mqtt/wiki/Pairing-devices#xiaomi

qoobaa, 2018-11-20:
It works! Thank you sir.

closed 2018-11-20 by qoobaa