Koenkk/zigbee2mqtt #615: Cannot get the Node Descriptor of the Device when pairing Tradfri bulb

aforch, 2018-11-20:
Tradfri bulb, paired and working for ~2 months. Suddenly 2 days ago stopped responding. When attempting to repair the following error is generated in the logs:
error: Cannot get the Node Descriptor of the Device: 0x90fd9ffffed7232e (Error: Timed out after 10000 ms)

To troubleshoot I set up the Tradfri hub and connected the bulb to it to a) test it's pairing capability and b) look for firmware updates for bulb. 
The bulb paired without issue, and no firmware updates are available.
After resetting and attempting to again pair with my cc2531, I get the above error again. Other devices have been able to pair without issue in the past day as well to the cc2531.

Koenkk, 2018-11-23:
I also encountered this issue sometimes, however when keeping the bulb longer close to the CC2531 it eventually pairs.

aforch, 2018-11-23:
Yes, you're correct. I got it to pair last night. I'll close this.

closed 2018-11-23 by aforch