Koenkk/zigbee2mqtt #616: RGB Bulb Brightness/Color

blawford, 2018-11-20:
I am trying to get a Innr 285C RGB bulb to fully co-operate with Home Assistant. It is working but not as well as I would like!

So the main problem I am having is that when I want to change colour, brightness and power state at the same time. It does it, but in 3 distinct steps. 

The automation I have is motion sensor triggered to turn the bathroom light on red at minimum brightness. If the last time the bulb was on it was bright white at full brightness, it turns back on in this state for a split second, then bright red at full brightness and then finally red at minimum brightness.

As you can imagine, a flash of white light in the middle of the night when you've got up to use the toilet isn't what you want.

If I recreate this by sending the MQTT message manually (sent to `zigbee2mqtt/lightbulbShower/set` through the MQTT option in the developer tools):
`{"state":"ON","color_temp":555,"color":{"x":0.698,"y":0.299},"brightness":2}`

I get the following messages by subscribing to `zigbee2mqtt/lightbulbShower` and `zigbee2mqtt/lightbulbShower/set` in an external program:

> zigbee2mqtt/lightbulbShower/set -> {"state":"ON","color_temp":555,"color":{"x":0.698,"y":0.299},"brightness":2}  #Red, low brightness (the message sent from HomeAssistant)
> 
> zigbee2mqtt/lightbulbShower -> {"state":"ON","color_temp":338,"color":{"x":0.437,"y":0.404},"brightness":253} #White, full brightness
> zigbee2mqtt/lightbulbShower -> {"state":"ON","color_temp":338,"color":{"x":0.437,"y":0.404},"brightness":253} #White, full brightness
> zigbee2mqtt/lightbulbShower -> {"state":"ON","color_temp":338,"color":{"x":0.437,"y":0.404},"brightness":253} #White, full brightness
> 
> zigbee2mqtt/lightbulbShower -> {"state":"ON","color_temp":338,"color":{"x":0.437,"y":0.404},"brightness":2} #White, low brightness
> 
> zigbee2mqtt/lightbulbShower -> {"state":"ON","color_temp":555,"color":{"x":0.697,"y":0.299},"brightness":2} #Red, low brightness
> zigbee2mqtt/lightbulbShower -> {"state":"ON","color_temp":555,"color":{"x":0.697,"y":0.299},"brightness":2} #Red, low brightness

My HomeAssistant configuration for the device is the following:

```
light:
  - platform: "mqtt_json"
    name: "Shower Light"
    state_topic: "zigbee2mqtt/lightbulbShower"
    availability_topic: "zigbee2mqtt/bridge/state"
    brightness: true
    color_temp: true
    xy: true
    command_topic: "zigbee2mqtt/lightbulbShower/set"
```

Anyone any ideas how I can improve things?

Koenkk, 2018-11-22:
The reason for this is that all zigbee commands are handled sequentially.

However, when using only the `brightness` attribute the bulb will be turned on with the given brightness (will produce only one zigbee command).

When the bulb is off, can you try sending a `{"brightness":  50}` and check if this improves things?

blawford, 2018-11-22:
Thanks for the info, if I can get the bulb to turn on at the correct brightness (even in the 'wrong' colour) it will definitely improve things. I will give it a try.

Koenkk, 2018-11-28:
I couldn't reproduce this.

What i've done
- Set the RB 285 C to 100%
- Turn the bulb off
- Send `{"brightness": "2"}

Can you provide the log with debug enabled? (set `log_level` to `debug`, https://github.com/Koenkk/zigbee2mqtt/wiki/Configuration)

kiwinol, 2018-12-20:
I think I can reproduce the issue.  I have a button that if I single click it gives me a "normal" light and if I hold it gives me a much duller red light.

This is what I see in mosquitto for the device:
```
  zigbee2mqtt:debug 2018-12-20 20:43:23 Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.sensor_switch' (0x00158d00010aca1b)
  zigbee2mqtt:debug 2018-12-20 20:43:23 Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.sensor_switch' (0x00158d00010aca1b)
  zigbee2mqtt:debug 2018-12-20 20:43:23 Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.sensor_switch' (0x00158d00010aca1b)
  zigbee2mqtt:info 2018-12-20 20:43:23 MQTT publish, topic: 'zigbee2mqtt/0x00158d00010aca1b', payload: '{"battery":99,"voltage":3012,"linkquality":92,"click":"single"}'
  zigbee2mqtt:debug 2018-12-20 20:43:23 Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.sensor_switch' (0x00158d00010aca1b)
  zigbee2mqtt:debug 2018-12-20 20:43:23 Received MQTT message on 'zigbee2mqtt/0x0017880110460c0f/set' with data '{"state": "ON", "color": {"x": 0.529, "y": 0.347}, "brightness": 250}'
  zigbee2mqtt:info 2018-12-20 20:43:23 Zigbee publish to '0x0017880110460c0f', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2018-12-20 20:43:23 MQTT publish, topic: 'zigbee2mqtt/0x0017880110460c0f', payload: '{"state":"ON","brightness":100,"color_temp":153,"color":{"x":0.675,"y":0.322}}'
  zigbee2mqtt:info 2018-12-20 20:43:23 Zigbee publish to '0x0017880110460c0f', lightingColorCtrl - moveToColor - {"colorx":34668.015,"colory":22740.644999999997,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2018-12-20 20:43:23 Zigbee publish to '0x0017880110460c0f', genLevelCtrl - moveToLevelWithOnOff - {"level":250,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:debug 2018-12-20 20:43:23 Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'LCT001' (0x0017880110460c0f)
  zigbee2mqtt:info 2018-12-20 20:43:23 MQTT publish, topic: 'zigbee2mqtt/0x0017880110460c0f', payload: '{"state":"ON","brightness":100,"color_temp":153,"color":{"x":0.675,"y":0.322}}'
  zigbee2mqtt:info 2018-12-20 20:43:24 MQTT publish, topic: 'zigbee2mqtt/0x0017880110460c0f', payload: '{"state":"ON","brightness":100,"color_temp":153,"color":{"x":0.675,"y":0.322}}'
  zigbee2mqtt:debug 2018-12-20 20:43:24 Received zigbee message of type 'devChange' with data '{"cid":"genLevelCtrl","data":{"currentLevel":250}}' of device 'LCT001' (0x0017880110460c0f)
  zigbee2mqtt:info 2018-12-20 20:43:24 MQTT publish, topic: 'zigbee2mqtt/0x0017880110460c0f', payload: '{"state":"ON","brightness":250,"color_temp":153,"color":{"x":0.675,"y":0.322}}'
  zigbee2mqtt:debug 2018-12-20 20:43:25 Received zigbee message of type 'devChange' with data '{"cid":"lightingColorCtrl","data":{"currentHue":255,"currentSaturation":131,"currentX":34695,"currentY":22734,"colorTemperature":500,"enhancedCurrentHue":65330,"colorLoopStoredEnhancedHue":65330}}' of device 'LCT001' (0x0017880110460c0f)
  zigbee2mqtt:info 2018-12-20 20:43:25 MQTT publish, topic: 'zigbee2mqtt/0x0017880110460c0f', payload: '{"state":"ON","brightness":250,"color_temp":500,"color":{"x":0.529,"y":0.347}}'
```
and this is what I see in my hassio addon log:

```
  zigbee2mqtt:debug 2018-12-20 20:43:23 Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.sensor_switch' (0x00158d00010aca1b)
  zigbee2mqtt:debug 2018-12-20 20:43:23 Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.sensor_switch' (0x00158d00010aca1b)
  zigbee2mqtt:debug 2018-12-20 20:43:23 Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.sensor_switch' (0x00158d00010aca1b)
  zigbee2mqtt:info 2018-12-20 20:43:23 MQTT publish, topic: 'zigbee2mqtt/0x00158d00010aca1b', payload: '{"battery":99,"voltage":3012,"linkquality":92,"click":"single"}'
  zigbee2mqtt:debug 2018-12-20 20:43:23 Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.sensor_switch' (0x00158d00010aca1b)
  zigbee2mqtt:debug 2018-12-20 20:43:23 Received MQTT message on 'zigbee2mqtt/0x0017880110460c0f/set' with data '{"state": "ON", "color": {"x": 0.529, "y": 0.347}, "brightness": 250}'
  zigbee2mqtt:info 2018-12-20 20:43:23 Zigbee publish to '0x0017880110460c0f', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2018-12-20 20:43:23 MQTT publish, topic: 'zigbee2mqtt/0x0017880110460c0f', payload: '{"state":"ON","brightness":100,"color_temp":153,"color":{"x":0.675,"y":0.322}}'
  zigbee2mqtt:info 2018-12-20 20:43:23 Zigbee publish to '0x0017880110460c0f', lightingColorCtrl - moveToColor - {"colorx":34668.015,"colory":22740.644999999997,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2018-12-20 20:43:23 Zigbee publish to '0x0017880110460c0f', genLevelCtrl - moveToLevelWithOnOff - {"level":250,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
[... cut ...]
  zigbee2mqtt:debug 2018-12-20 20:43:25 Received zigbee message of type 'devChange' with data '{"cid":"lightingColorCtrl","data":{"currentHue":255,"currentSaturation":131,"currentX":34695,"currentY":22734,"colorTemperature":500,"enhancedCurrentHue":65330,"colorLoopStoredEnhancedHue":65330}}' of device 'LCT001' (0x0017880110460c0f)
  zigbee2mqtt:info 2018-12-20 20:43:25 MQTT publish, topic: 'zigbee2mqtt/0x0017880110460c0f', payload: '{"state":"ON","brightness":250,"color_temp":500,"color":{"x":0.529,"y":0.347}}'
```

closed 2019-02-25 by stale[bot]