Koenkk/zigbee2mqtt #617: Problem with installing Zigbee2mqtt with cc2531

michaeldbr, 2018-11-20:
Hi guys, I need some help! I came from pimatic to home assistant because I couldn’t get it stable.. after some tweaking I got Hassio working well! At the moment I `have:`
RPI3B with an Arduino and also a relayshield working on it. The relayshield is for my sealing light. I have a Mi-Light bridge connected by limitlessLed and some networking scanning for home detection by Nmap. Now I came at the point I want some sensors so my house can control itself. Idea is to use xiaomi sensors with Zigbee2mqtt. But after some attempts of installing I can’t get it to work...
I flashed the CC2531 with the firmware as mentioned with the flasher and software from TI. I think it works?? I really dont know how to check this. And followed the complete steps as mentioned here on github. I installed node version 11.x but Zigbee2mqtt is still mentioning that I installed the wrong node version...

My config of Zigbee2mqtt is:
`{
  "data_path": "/share/zigbee2mqtt",
  "homeassistant": true,
  "permit_join": true,
  "mqtt_base_topic": "zigbee2mqtt",
  "mqtt_server": "mqtt://homeassistant",
  "serial_port": "/dev/ttyACM0"
}`

My config at mosquito broker:
`{
  "logins": [],
  "anonymous": false,
  "customize": {
    "active": false,
    "folder": "mosquitto"
  },
  "certfile": "fullchain.pem",
  "keyfile": "privkey.pem"
}`

And my error at Zigbee2mqtt:

Info] Configuration file found. Will overwrite configurable               fields with values from add-on configuration
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
2018-11-20T21:39:48: PM2 log: Launching in no daemon mode
2018-11-20T21:39:49: PM2 log: App [npm:0] starting in -fork mode-
2018-11-20T21:39:49: PM2 log: App [npm:0] online
> zigbee2mqtt@0.2.0 start /zigbee2mqtt-0.2.0
> node index.js
		Zigbee2mqtt requires node version >=8.11 10, you are running v8.11.4!
  zigbee2mqtt:info 2018-11-20 21:39:55 Logging to directory: '/share/zigbee2mqtt/log/2018-11-20.21-39-55'
  zigbee2mqtt:info 2018-11-20 21:39:57 Starting zigbee2mqtt version 0.2.0 (commit #unknown)
  zigbee2mqtt:info 2018-11-20 21:39:57 Starting zigbee-shepherd
  zigbee2mqtt:info 2018-11-20 21:40:01 zigbee-shepherd started
  zigbee2mqtt:info 2018-11-20 21:40:01 Coordinator firmware version: 'undefined'
  zigbee2mqtt:info 2018-11-20 21:40:01 Currently 0 devices are joined:
  zigbee2mqtt:warn 2018-11-20 21:40:01 `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 2018-11-20 21:40:01 Allowing new devices to join.
  zigbee2mqtt:warn 2018-11-20 21:40:01 Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 2018-11-20 21:40:01 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2018-11-20 21:40:01 Connecting to MQTT server at mqtt://homeassistant
  zigbee2mqtt:info 2018-11-20 21:40:02 zigbee-shepherd ready
events.js:183

Error: Connection refused: Not authorized
    at MqttClient._handleConnack (/zigbee2mqtt-0.2.0/node_modules/mqtt/lib/client.js:920:15)
    at MqttClient._handlePacket (/zigbee2mqtt-0.2.0/node_modules/mqtt/lib/client.js:350:12)
    at work (/zigbee2mqtt-0.2.0/node_modules/mqtt/lib/client.js:292:12)
    at Writable.writable._write (/zigbee2mqtt-0.2.0/node_modules/mqtt/lib/client.js:302:5)
    at doWrite (/zigbee2mqtt-0.2.0/node_modules/readable-stream/lib/_stream_writable.js:428:64)
    at writeOrBuffer (/zigbee2mqtt-0.2.0/node_modules/readable-stream/lib/_stream_writable.js:417:5)
    at Writable.write (/zigbee2mqtt-0.2.0/node_modules/readable-stream/lib/_stream_writable.js:334:11)
    at Socket.ondata (_stream_readable.js:639:20)
    at emitOne (events.js:116:13)
    at Socket.emit (events.js:211:7)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@0.2.0 start: `node index.js`
npm 
ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@0.2.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-11-20T21_40_41_654Z-debug.log
2018-11-20T21:40:41: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2018-11-20T21:40:41: PM2 log: App [npm:0] starting in -fork mode-
2018-11-20T21:40:41: PM2 log: App [npm:0] online
> zigbee2mqtt@0.2.0 start /zigbee2mqtt-0.2.0
> node index.

Thanks for the help in advance guys!!!

1981nick1982, 2018-11-21:
Since today i have the same issue. Is this a known issue?

Koenkk, 2018-11-22:
The MQTT authentication fails, can you try connecting to your MQTT server via an MQTT client (e.g. https://mqttfx.jensd.de/)

michaeldbr, 2018-11-22:
I just tried with an IOS MQTT client to connect to my raspberry and this works... I tried changing the MQTT link to localhost, the ip address and I also changed the port all in Zigbee2mqtt configuration. This all resulted in an error “can’t connect to MQTT broker”. The weird thing is that MQTT mosquito once says there is a connection and a second later there is no connection...
1542919857: New client connected from 172.30.32.1 as 08962d6a-8a27-4158-9527-05d6b.
The ip starting with 172. Is not used in my network... but mosquito says that is HomeAssistant...
After the connection above mosquito give the error:
1542918379: New connection from 172.30.32.1 on port 1883.
1542918379: Socket error on client <unknown>, disconnecting.

Koenkk, 2018-11-23:
For those having the problem, do you all have no authentication on the MQTT server?

Koenkk, 2018-11-23:
Seems to be an hassio issue: https://github.com/danielwelch/hassio-zigbee2mqtt/issues/71

01kbpatel, 2018-11-23:
Okay,

So I have hassio installed with just Zigbee2mqtt addon, connecting to a mqtt server on different machine altogether, upgrading to v2 has completely broken zigbee network

Also tried full restore of v1.8 working snapshot, automatically upgrades to v2.0, during addon boot up resulting with no zigbee network. So down grading does not work

Even tried deleting db, and re-pairing, this just causes the device to constantly crash and reboot

Hope this helps in trying to figure out what’s broken, log also warning node-js needs upgrading (not compatible).

As stated above, I have mqtt server on another device, there is no issue to connecting with mqtt, if the database has no devices, it connects just fine, it’s when devices are added to db the addon just crashes and then reboots in constant cycle!

I have just taken it off line, hope this can be sorted out, is it possible to setup a v1.8 branch please, until we have resolution for this issue thank you for all the effort you guys are putting in to project.

1981nick1982, 2018-11-23:
On my system the problem was solved by setting the anonymous setting on true, for the hassio mosquitto broker.

closed 2019-01-30 by stale[bot]