Koenkk/zigbee2mqtt #618: Support Calex LED bubl

Fossdal, 2018-11-21:
Please add suport for CALEX LED bulb
it is this bulb 
https://www.calex.nl/en/catalogue/product/LEDNLE27A60-8-5W-2700-6500ZIGBEE/

have tested it and works. after adding this to devices

    {
        zigbeeModel: ['EC-Z3.0-CCT '],
        model: '421786',
        vendor: 'Calex',
        description: 'Calex LED A60 Zigbee GLS-lamp',
        supports: generic.light_onoff_brightness().supports,
        fromZigbee: generic.light_onoff_brightness().fromZigbee,
        toZigbee: generic.light_onoff_brightness().toZigbee,
    }

blawford, 2018-11-21:
Why don't you create a Pull Request with the above code then?

Fossdal, 2018-12-06:
me and git are not best friends :-)

Koenkk, 2018-12-06:
Added, thanks!

closed 2018-12-06 by Koenkk