Koenkk/zigbee2mqtt #619: Hue Tap Switch compatibility?

thibaultmol, 2018-11-21:
The Hue Tap Switch doesn't seem to be supported. The nice thing about that device, is that it doesn't require batteries, it completely functions of the kinetic energy of your button press.

https://www2.meethue.com/en-us/p/hue-tap-switch/046677473365

(I don't have one, so I can't test it)

Koenkk, 2018-11-22:
I've somebody has it, we can try adding support for it.

TjBen, 2018-11-26:
I have one and I would like to use it with Zigbee2mqtt. Now it has some restrictions from hue, like only select 3 rooms and only change hue lights. I know I can solve the first restriction with another app but the second one is probably more difficult...

I’m using hassOS and I’ve installed the add on. What do you need from me to add support?

Ps. More people should buy the tap. I’m using it as central device for multiple rooms downstairs. It looks nice and the kinetic energy is always a good story :p

SalkinDK, 2018-12-27:
@TjBen 
Any progress? i just got one myself and would like to help out to get it supported.
Anything i can do?

TjBen, 2018-12-27:
I tried to follow this guide: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
but without success ...

I'm afraid that because of the low power of the device that it uses a different kind of zigbee signal. When I try to put the hue tap in pairing mode (>10 s on button 1) I don't see anything in the zigbee2mqtt log. With the hue app I had to press button 3 for more then 10 seconds to connect to the app.

I'm now using the Hue API but it's not that fast. And sometimes you have to press 2 times...

more info:
https://www.zigbee.org/zigbee-for-developers/zigbee-pro/

@SalkinDK: You can try follow the guide, maybe I'm doing something wrong. 
@Koenkk:  Does Zigbee2Mqtt support Zigbee Light Link Protocol and ZigBee Green Power-protocol?

SalkinDK, 2018-12-27:
@TjBen 
I unfortunally had the same experience, nothing is showing up in the log when trying to pair.

Koenkk, 2018-12-27:
Zigbee light link is supported, green power not. But as all the other Hue remotes can be paired, I find it strange that this once cannot.

Are you sure that the Hue Zigbee network is down? (remove all bulbs and hub from AC).

SalkinDK, 2018-12-27:
I dont have a hue hub running here. so there isnt another zigbee network.

Koenkk, 2018-12-27:
Green power has been introduced in Zigbee 3.0 so first #211 has to be finished.

thibaultmol, 2019-02-26:
Seems like a nice amount of progress was made on the last mentioned issue here. so I wonder if it actually works now already (but is just a bit unstable still)

WhistleMaster, 2019-03-10:
Any news regarding the support of Hue Tap with zigbee2mqtt ?

WhistleMaster, 2019-04-04:
@Koenkk Any news regarding the support of Hue Tap with zigbee2mqtt ?

Koenkk, 2019-04-04:
@whistlemaster no, I dont have this device.

WhistleMaster, 2019-04-04:
I have a few of them, I can help, tell me what you would need !

Koenkk, 2019-04-04:
Reading https://github.com/Koenkk/zigbee2mqtt/issues/619#issuecomment-450145033 this would require zigbee 3.0 (green power) support, Im currently working on this.

closed 2019-06-10 by stale[bot]

electrofloat, 2019-09-11:
Can we reopen this issue? There's already a downloadable zigbee 3 firmware, maybe someone can test it to see if it works?

WhistleMaster, 2019-09-11:
I tried recently with the new 3.0 firmware, no luck unfortunately.

Koenkk, 2019-09-11:
First green power support has to be added (https://github.com/Koenkk/zigbee-herdsman/issues/22)