Koenkk/zigbee2mqtt #620: IKEA TRADFRI wireless dimmer (ICTC-G-1) not pairing

nerdosity, 2018-11-22:
Bug report
- First make sure your on the latest version of zigbee2mqtt.
Check
- Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware.
Check
- Provide a clear description of the problem.
IKEA TRADFRI wireless dimmer (ICTC-G-1) will not pair. It look like it acutally does nothing at all, no incoming message, nothing in debug.
- If applicable, provide steps how to reproduce the problem.
CC2531 actually working with 11 devices.
Trying to pair an IKEA dimmer very near (5cm).
Resetting, long presses, nothing.
- Provide a log with `log_level` set to `debug`
https://hastebin.com/ladilekosi.pl

Koenkk, 2018-11-22:
Do you see the red lights flashing on the dimmer?

nerdosity, 2018-11-22:
Yes. Today I tried another CC2531, never used.

Error is different now, but at least there is something.

  zigbee2mqtt:error 2018-11-22 11:26:02 Failed to configure 0x000b57fffe11fc50 0x000b57fffe11fc50
  zigbee2mqtt:error 2018-11-22 11:26:02 Zigbee cannot determine endpoint for '0x000b57fffe11fc50'
  zigbee2mqtt:error 2018-11-22 11:26:02
        {"message":"Cannot set property 'onAfIncomingMsg' of null","stack":"TypeError: Cannot set property 'onAfIncomingMsg' of null\n    at Zigbee.registerOnAfIncomingMsg (/opt/zigbee2mqtt/lib/zigbee.js:233:32)\n    at mappedModel.onAfIncomingMsg.forEach (/opt/zigbee2mqtt/lib/controller.js:142:73)\n    at Array.forEach (<anonymous>)\n    at Controller.configureDevice (/opt/zigbee2mqtt/lib/controller.js:142:45)\n    at Controller.handleZigbeeMessage (/opt/zigbee2mqtt/lib/controller.js:245:14)\n    at Zigbee.handleMessage (/opt/zigbee2mqtt/lib/zigbee.js:172:18)\n    at emitOne (events.js:116:13)\n    at ZShepherd.emit (events.js:211:7)\n    at ZShepherd.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/shepherd.js:100:14)\n    at emitOne (events.js:116:13)"}
  zigbee2mqtt:info 2018-11-22 11:26:34 MQTT publish, topic: 'zigbee2mqtt/0x00158d0001e5e338', payload: '{"illuminance":22,"linkquality":63,"occupancy":false,"battery":100,"voltage":3035}'
  zigbee2mqtt:info 2018-11-22 11:26:49 MQTT publish, topic: 'zigbee2mqtt/0x00158d0001e5e338', payload: '{"illuminance":21,"linkquality":52,"occupancy":false,"battery":100,"voltage":3035}'
  zigbee2mqtt:info 2018-11-22 11:26:49 MQTT publish, topic: 'zigbee2mqtt/0x00158d0001e5e338', payload: '{"illuminance":21,"linkquality":52,"occupancy":true,"battery":100,"voltage":3035}'
  zigbee2mqtt:warn 2018-11-22 11:26:57 Message without device!
  zigbee2mqtt:info 2018-11-22 11:26:58 Device incoming...
  zigbee2mqtt:info 2018-11-22 11:26:58 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:error 2018-11-22 11:26:58 Failed to configure 0x000b57fffe11fc50 0x000b57fffe11fc50
  zigbee2mqtt:error 2018-11-22 11:26:58 Zigbee cannot determine endpoint for '0x000b57fffe11fc50'
  zigbee2mqtt:error 2018-11-22 11:26:58
        {"message":"Cannot set property 'onAfIncomingMsg' of null","stack":"TypeError: Cannot set property 'onAfIncomingMsg' of null\n    at Zigbee.registerOnAfIncomingMsg (/opt/zigbee2mqtt/lib/zigbee.js:233:32)\n    at mappedModel.onAfIncomingMsg.forEach (/opt/zigbee2mqtt/lib/controller.js:142:73)\n    at Array.forEach (<anonymous>)\n    at Controller.configureDevice (/opt/zigbee2mqtt/lib/controller.js:142:45)\n    at Controller.handleZigbeeMessage (/opt/zigbee2mqtt/lib/controller.js:245:14)\n    at Zigbee.handleMessage (/opt/zigbee2mqtt/lib/zigbee.js:172:18)\n    at emitOne (events.js:116:13)\n    at ZShepherd.emit (events.js:211:7)\n    at ZShepherd.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/shepherd.js:100:14)\n    at emitOne (events.js:116:13)"}
  zigbee2mqtt:info 2018-11-22 11:27:32 MQTT publish, topic: 'zigbee2mqtt/0x00158d00020194a1', payload: '{"battery":99,"voltage":3015,"linkquality":84,"contact":false}'
  zigbee2mqtt:info 2018-11-22 11:27:42 MQTT publish, topic: 'zigbee2mqtt/0x00158d00020194a1', payload: '{"battery":99,"voltage":3015,"linkquality":92,"contact":true}'
  zigbee2mqtt:info 2018-11-22 11:27:43 MQTT publish, topic: 'zigbee2mqtt/0x00158d000289d1fb', payload: '{"battery":55,"voltage":2995,"linkquality":55,"action":"wakeup"}'
  zigbee2mqtt:info 2018-11-22 11:28:19 MQTT publish, topic: 'zigbee2mqtt/0x00158d0001e5e338', payload: '{"illuminance":21,"linkquality":52,"occupancy":false,"battery":100,"voltage":3035}'
  zigbee2mqtt:info 2018-11-22 11:28:35 Device incoming...
  zigbee2mqtt:info 2018-11-22 11:28:35 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 2018-11-22 11:28:35 MQTT publish, topic: 'homeassistant/sensor/0x000b57fffe11fc50/brightness/config', payload: '{"unit_of_measurement":"brightness","icon":"mdi:brightness-5","value_template":"{{ value_json.brightness }}","json_attributes":["linkquality"],"state_topic":"zigbee2mqtt/0x000b57fffe11fc50","availability_topic":"zigbee2mqtt/bridge/state","name":"0x000b57fffe11fc50","unique_id":"0x000b57fffe11fc50_brightness_zigbee2mqtt"}'
  zigbee2mqtt:info 2018-11-22 11:28:46 MQTT publish, topic: 'zigbee2mqtt/0x00158d0001e5e338', payload: '{"illuminance":18,"linkquality":47,"occupancy":false,"battery":100,"voltage":3035}'
  zigbee2mqtt:info 2018-11-22 11:28:46 MQTT publish, topic: 'zigbee2mqtt/0x00158d0001e5e338', payload: '{"illuminance":18,"linkquality":49,"occupancy":true,"battery":100,"voltage":3035}'
  zigbee2mqtt:error 2018-11-22 11:29:26 Failed to configure 0x000b57fffe11fc50 0x000b57fffe11fc50

nerdosity, 2018-11-22:
I noticed that in the end, after the final restart, the rotation is now alive. A bit jerky but definitely alive.
What was my issue? Why weren't I able to sync it with the first CC2531 stick?

Koenkk, 2018-11-22:
It could be that it connected to a node where it was previously connected to (e.g. a TRADFRI bulb).

nerdosity, 2018-11-22:
Impossible, it was my first IKEA device in my home.
The bulb was still in the bag when I paired the dimmer.
Maybe it was pre-paired in fabric with the bulb? I bought it in a kit (1000lm dimmerable bulb + dimmer).

UnrealKazu, 2018-11-23:
@nerdosity AFAIK if you buy a bulb + dimmer in one package, they are already paired in the factory by IKEA. So that could be a reason why it's not pairing.

apeinspace, 2018-11-23:
Couldnt agree more, that dimmer is s**t. I was able to pair it and then watch my sensors and routers lose connection step by step - removed the dimmer, everything recovers within minutes. Did repeat that for 3 times, then took the battery out - will return it to IKEA.

Why that happened, I have no clue. Ideas welcome.

alinelena, 2018-11-23:
ok just to correct myself... I tried to pair it today and worked like a charm... the pairing... the dimmer itself still crap.

closed 2019-01-30 by stale[bot]

u17194907425, 2019-05-11:
SInce this is one of the top results in Google regarding pairing the subject I'll leave here my experience. ( @Koenkk maybe it's worthwhile to add a note into the "pairing" section of the manual. I might be willing to contribute)
I had pretty much the same experience. Including the
> The sensor inside HA was created, but it's not alive. It stays in "unknown" state.
However **as soon as I have re-started the zigbee2mqtt plugin it just started to work**... (I'm using hass.io on VirtualBox).
So, if you are reading these lines trying to make this dimmer work - try to use that good old rule. Restart :)

u17194907425, 2019-05-12:
After a half-day of successful operation this dimmer just stopped working. Neither re-pairing nor zigbee2mqtt restart helped. I give up. This dimmer is indeed a piece of crap.

ckhmer1, 2020-09-05:
Hi,
first of all, thanks for your work to free the Zigbee devices.

I have a CC2531 working with node-red (I've flashed your firmware).

I've bought 4 of them (two alone and two in a bundle with an Ikea light) and I'd like to make them work with node-red.
I've already a node-red Zigbee plugin working, with a Xiaomi sensor and some Ikea stuff, lights, a button, and an occupancy sensor.
Unfortunately, the Ikea dimmer is not working at all, I was able to pair but no messages are generated. I've tried the two stand-alone dimmer, never paired to another device.
Just for testing, I've paired it directly to the light, and the dimmer is working correctly.

I've also restarted the server, created another one with a new Zigbee key, but nothing.

I'm wondering how to debug it in order to try to fix it.

Is there someone who can help me?
Thanks
Claudio