Koenkk/zigbee2mqtt #621: Add support for Xiaomi Aqara Smart Light Bulbs

sapkra, 2018-11-22:
Aqara released a light bulb with zigbee. Would be great to support it. ;D

Unsupported device
https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20181122060617&SearchText=Aqara+Smart+Light+Bulbs

Koenkk, 2018-11-22:
In case you have this device, we can add support for it.

kiwinol, 2018-11-25:
I have just ordered one of these so if it's not added by the time mine arrives and I can get some help with the process I will add support.

paskyecijano, 2018-11-28:
I have just ordered one of these too and would be phantastic give support for the bulb

kiwinol, 2018-12-05:
My bulb arrived today.  My laptop I usually use is not booting due to an update :(  I tried to connect the bulb anyway.  Hopefully I have some time this weekend:

> zigbee2mqtt:info 2018-12-5 19:23:08 New device with address 0x00158d0002c64b73 connected!
  zigbee2mqtt:info 2018-12-5 19:23:08 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x00158d0002c64b73"}'
  zigbee2mqtt:warn 2018-12-5 19:23:08 Device with modelID 'lumi.light.aqcn02' is not supported.
  zigbee2mqtt:warn 2018-12-5 19:23:08 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
  zigbee2mqtt:debug 2018-12-5 19:23:08 Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"3":24,"5":9,"7":"0x0000000000000000","8":278,"9":1280,"10":13100,"100":1,"101":255,"102":153}}}' of device 'lumi.light.aqcn02' (0x00158d0002c64b73)
  zigbee2mqtt:warn 2018-12-5 19:23:08 Device with modelID 'lumi.light.aqcn02' is not supported.
  zigbee2mqtt:warn 2018-12-5 19:23:08 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
  zigbee2mqtt:debug 2018-12-5 19:23:08 Received zigbee message of type 'devInterview' with data '"0x00158d0002c64b73"'
  zigbee2mqtt:debug 2018-12-5 19:23:09 Received zigbee message of type 'devInterview' with data '"0x00158d0002c64b73"'
  zigbee2mqtt:debug 2018-12-5 19:23:09 Received zigbee message of type 'devInterview' with data '"0x00158d0002c64b73"'
  zigbee2mqtt:debug 2018-12-5 19:23:09 Received zigbee message of type 'devInterview' with data '"0x00158d0002c64b73"'
  zigbee2mqtt:debug 2018-12-5 19:23:09 Received zigbee message of type 'devInterview' with data '"0x00158d0002c64b73"'
  zigbee2mqtt:debug 2018-12-5 19:23:10 Received zigbee message of type 'devInterview' with data '"0x00158d0002c64b73"'
  zigbee2mqtt:debug 2018-12-5 19:23:10 Received zigbee message of type 'devInterview' with data '"0x00158d0002c64b73"'
  zigbee2mqtt:debug 2018-12-5 19:23:10 Received zigbee message of type 'devInterview' with data '"0x00158d0002c64b73"'
  zigbee2mqtt:debug 2018-12-5 19:23:10 Received zigbee message of type 'devInterview' with data '"0x00158d0002c64b73"'
  zigbee2mqtt:debug 2018-12-5 19:23:10 Received zigbee message of type 'devInterview' with data '"0x00158d0002c64b73"'
  zigbee2mqtt:debug 2018-12-5 19:23:10 Received zigbee message of type 'devInterview' with data '"0x00158d0002c64b73"'
  zigbee2mqtt:debug 2018-12-5 19:23:10 Received zigbee message of type 'devInterview' with data '"0x00158d0002c64b73"'
  zigbee2mqtt:debug 2018-12-5 19:23:11 Received zigbee message of type 'devInterview' with data '"0x00158d0002c64b73"'
  zigbee2mqtt:debug 2018-12-5 19:23:11 Received zigbee message of type 'devInterview' with data '"0x00158d0002c64b73"'
  zigbee2mqtt:debug 2018-12-5 19:23:11 Received zigbee message of type 'devInterview' with data '"0x00158d0002c64b73"'
  zigbee2mqtt:debug 2018-12-5 19:23:11 Received zigbee message of type 'devInterview' with data '"0x00158d0002c64b73"'
  zigbee2mqtt:debug 2018-12-5 19:23:11 Received zigbee message of type 'devInterview' with data '"0x00158d0002c64b73"'
  zigbee2mqtt:debug 2018-12-5 19:23:11 Received zigbee message of type 'devIncoming' with data '"0x00158d0002c64b73"' of device 'lumi.light.aqcn02' (0x00158d0002c64b73)
  zigbee2mqtt:info 2018-12-5 19:23:11 Device incoming...
  zigbee2mqtt:info 2018-12-5 19:23:11 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:warn 2018-12-5 19:23:11 Device with modelID 'lumi.light.aqcn02' is not supported.
  zigbee2mqtt:warn 2018-12-5 19:23:11 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
  zigbee2mqtt:debug 2018-12-5 19:23:11 Received zigbee message of type 'devStatus' with data '"online"' of device 'lumi.light.aqcn02' (0x00158d0002c64b73)
  zigbee2mqtt:warn 2018-12-5 19:23:11 Device with modelID 'lumi.light.aqcn02' is not supported.
  zigbee2mqtt:warn 2018-12-5 19:23:11 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices

>

Koenkk, 2018-12-05:
What features does this bulb support; brightness and color temperature?

Koenkk, 2018-12-05:
Can you add to `devices.js`:

```js
    {
        zigbeeModel: ['lumi.light.aqcn02'],
        model: 'ZNLDP12LM',
        vendor: 'Xiaomi',
        description: 'Aqara smart LED bulb',
        supports: generic.light_onoff_brightness_colortemp().supports,
        fromZigbee: generic.light_onoff_brightness_colortemp().fromZigbee,
        toZigbee: generic.light_onoff_brightness_colortemp().toZigbee,
    },
```

kiwinol, 2018-12-07:
I will try this tomorrow.  

I see that its now possible to add an external devices.js in /share/zigbee2mqtt and set "zigbee_shepherd_devices": "true" in the addon config, is this correct?

if so can I just add the following to the devices.js file?

```
     {
        zigbeeModel: ['lumi.light.aqcn02'],
        model: 'ZNLDP12LM',
        vendor: 'Xiaomi',
        description: 'Aqara smart LED bulb',
        supports: generic.light_onoff_brightness_colortemp().supports,
        fromZigbee: generic.light_onoff_brightness_colortemp().fromZigbee,
        toZigbee: generic.light_onoff_brightness_colortemp().toZigbee,
    },
`````

Also is the last coma required?

Koenkk, 2018-12-07:
Yes, but note this replace the whole `devices.js` (so devices which are not in that file won't work anymore) (https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices-on-Hass.io)

kiwinol, 2018-12-07:
I have updated devices.js now I get

```
zigbee2mqtt:debug 2018-12-8 07:51:26 Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"3":74,"5":11,"7":"0x0000000000000000","8":278,"9":768,"10":13100,"100":1,"101":255,"102":153}}}' of device 'lumi.light.aqcn02' (0x00158d0002c64b73)
  zigbee2mqtt:warn 2018-12-8 07:51:26 No converter available for 'ZNLDP12LM' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"65281":{"3":74,"5":11,"7":"0x0000000000000000","8":278,"9":768,"10":13100,"100":1,"101":255,"102":153}}}'
  zigbee2mqtt:warn 2018-12-8 07:51:26 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
```

Koenkk, 2018-12-09:
We need to find out what the values mean.

Does turning on/off via MQTT work?

kiwinol, 2018-12-10:
No.  If I publish:
```
{"state": "OFF"}
```
to:
```
 zigbee2mqtt/0x00158d0002c64b73/set
```

it doesnt seem to have any effect.  I double checked I wasnt making a mistake by publishing the ON and OFF states to one of my working Philips Hue bulbs and that does repond.

Koenkk, 2018-12-10:
Ok, we need to sniff the packets send by the xiaomi gateway in order to support this bulb. (http://www.inzoolee.com/capture-zigbee-packets/)

kiwinol, 2018-12-10:
I was so excited by this project I sold my Xiaomi gateway :(.  I am going to borrow one off my friend (who bought it off me!) hopefully tomorrow and try this in the next couple of days.

Just to clarify.  I would pair the bulb with the Xiaomi gateway then follow the article to capture zigbee packets for On/Off, Brightness Increase,Decrease and color temp Increase,Decrease?

kiwinol, 2018-12-11:
@Koenkk can I use the same firmware I was using for zigbee2mqtt or will I need to flash a different firmware?

Koenkk, 2018-12-12:
You can use the firmware which is included in [PACKET SNIFFER](http://www.ti.com/tool/PACKET-SNIFFER)

kiwinol, 2018-12-12:
I downloaded and installed
```
PACKET-SNIFFER:
SmartRF Protocol Packet Sniffer

http://www.ti.com/general/docs/lit/getliterature.tsp?baseLiteratureNumber=swrc045&fileType=zip
```
I am not sure which firmware to use.  Below is a screenshot of the folders I see:
[https://ibb.co/DKG4y1d](url)

Koenkk, 2018-12-12:
If I remember correctly, it's in the cc debugger directory.

kirovilya, 2018-12-12:
Something did wrong ...
I was informed that someone managed to connect a light bulb and it works through clusters genOnOff, genLevelCtrl, lightingColorCtrl.
But I can not check it myself - there is no lamp.

kiwinol, 2018-12-13:
> genLevelCtrl, lightingColorCtrl

Can you provide more details of how to test?  I have the bulb but not much knowledge on how this works.

kirovilya, 2018-12-13:
@kiwinol After change devices-file, try to run zigbee2mqtt in debug mode https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-debug , pair lamp and then try to turn on/off, set brightness, color temp by send message to topics.
You will have a log file with information about sending messages to the lamp and responses from it. Share it (via pastebin for example).
If the lamp did not accept commands, you would have received error messages. In other cases, commands must be accepted.

kirovilya, 2018-12-13:
https://gist.github.com/itsmepetrov/79ae4aef70cce328d2c93ea1334af07f It is an example of improvements for this lamp from the project https://github.com/itsmepetrov/homebridge-zigbee
These improvements look like a normal lamp. And it is works.

kiwinol, 2018-12-14:
@kirovilya thanks for the information. I will try to investigate more this weekend.

kiwinol, 2018-12-14:
@Koenkk I got the debugger firmware flashed and am now able to capture zigbee traffic.  Tomorrow I will switch off my HA to reduce noise and capture the traffic between the Xiaomi hub and the bulb.

kiwinol, 2018-12-14:
[Aqarabulb_1034_aus_time.zip](https://github.com/Koenkk/zigbee2mqtt/files/2682003/Aqarabulb_1034_aus_time.zip)

I have tried to do the capture.  It was all done at about 10:34am Australian time.

The sequence I tried was.
On
Increase Brightness
Decrease Brightness
Increase Color Temp
Decrease Color Temp
Off

Let me know if I need to redo it or you require more info I wasnt sure how to filter it and there are probably messages from many other zigbee devices I have at the same time.

trfr3ak, 2018-12-15:
Okay, I can now confirm, that everything just works, when you treat the device like any other bulb, I can change color_temp and brightness, everything works, with the config from above.

kirovilya, 2018-12-15:
@trfr3ak Excellent! I thought so. Thanks for checking out - can you make the PR to converters?

trfr3ak, 2018-12-15:
@kirovilya I can try, but im not quite sure, what to do. Should I add something to the fromZigbee.js?

trfr3ak, 2018-12-15:
These seem to be the missing converters:
`No converter available for 'ZNLDP12LM' with cid 'genLevelCtrl', type 'attReport' and data '{"cid":"genLevelCtrl","data":{"61440":11383040,"currentLevel":255}}'
No converter available for 'ZNLDP12LM' with cid 'genOnOff', type 'attReport' and data '{"cid":"genOnOff","data":{"61440":117440706,"onOff":1}}'
No converter available for 'ZNLDP12LM' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"65281":[null,null,null,null,null,null,null,"0x0000000000000000",278,512,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,1,255]}}'
No converter available for 'ZNLDP12LM' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"65281":{"3":60,"5":62,"7":"0x0000000000000000","8":278,"9":512,"10":0,"100":1,"101":255,"102":319}}}'
No converter available for 'ZNLDP12LM' with cid 'lightingColorCtrl', type 'attReport' and data '{"cid":"lightingColorCtrl","data":{"61440":117440703,"colorTemperature":319}}'
  
`

phoxy84, 2018-12-16:
Sounds good, any idea when it is available in the edge version?

liads, 2018-12-17:
Just got my bulbs today.
The commands to the bulb seem to work correctly with the above configuration.
However, some messages from the bulb don't have mapped converters.
The bulb appears to be sending some messages twice, once with 'attReport' type and again with 'devChange' type). For example, after a single turn off request I get:
```
Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"61440":117440683,"onOff":0}}' of device 'lumi.light.aqcn02'
Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"61440":117440683,"onOff":0}}' of device 'lumi.light.aqcn02'
Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"61440":6928384,"onOff":0}}' of device 'lumi.light.aqcn02'
Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"61440":6928384}}' of device 'lumi.light.aqcn02'
```
After a brightness change:
```
Received zigbee message of type 'attReport' with data '{"cid":"genLevelCtrl","data":{"61440":117440689,"currentLevel":100}}' of device 'lumi.light.aqcn02'
Received zigbee message of type 'devChange' with data '{"cid":"genLevelCtrl","data":{"61440":117440689,"currentLevel":100}}' of device 'lumi.light.aqcn02'
Received zigbee message of type 'attReport' with data '{"cid":"genLevelCtrl","data":{"61440":6928384,"currentLevel":100}}' of device 'lumi.light.aqcn02'
Received zigbee message of type 'devChange' with data '{"cid":"genLevelCtrl","data":{"61440":6928384}}' of device 'lumi.light.aqcn02'
```
And so on.
The devChange type is handled properly by the generic light converters, but the attReport type causes a no converter warning.
I think it should be safe to ignore the duplicate attReport messages, but I wonder if those "genBasic" messages might have useful data we should decipher:
```
Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"3":74,"5":100,"7":"0x0000000000000000","8":278,"9":513,"10":0,"100":1,"101":100,"102":500}}}' of device 'lumi.light.aqcn02'
Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"65281":[null,null,null,74,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,1,100]}}' of device 'lumi.light.aqcn02'
```

phoxy84, 2018-12-17:
This is my log:

```
  zigbee2mqtt:warn 2018-12-17 19:41:44 No converter available for 'ZNLDP12LM' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"dateCode":"09-30-2018"}}'
  zigbee2mqtt:warn 2018-12-17 19:41:44 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:warn 2018-12-17 19:41:44 No converter available for 'ZNLDP12LM' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"65281":{"3":95,"5":77,"7":"0x0000000000000000","8":278,"9":768,"10":0,"100":1,"101":255,"102":153}}}'
  zigbee2mqtt:warn 2018-12-17 19:41:44 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:warn 2018-12-17 19:41:44 No converter available for 'ZNLDP12LM' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"65281":[null,null,null,95,null,77,null,"0x0000000000000000",278,768,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,1,255,153]}}'
  zigbee2mqtt:warn 2018-12-17 19:41:44 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:warn 2018-12-17 19:41:44 No converter available for 'ZNLDP12LM' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"65281":{"3":95,"5":77,"7":"0x0000000000000000","8":278,"9":768,"10":0,"100":1,"101":255,"102":153}}}'
  zigbee2mqtt:warn 2018-12-17 19:41:44 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
```

liads, 2018-12-18:
I noticed that whenever this bulb omitted the onOff attribute from the devChange genOnOff messages, an OFF state was being published to MQTT even though the bulb could still be on.
I managed to get stable behavior with the following in devices.js:
```
    {
        zigbeeModel: ['lumi.light.aqcn02'],
        model: 'ZNLDP12LM',
        vendor: 'Xiaomi',
        description: 'Aqara smart LED bulb',
        supports: generic.light_onoff_brightness_colortemp().supports,
        fromZigbee: [fz.light_brightness, fz.light_color_colortemp, fz.generic_state, fz.ignore_basic_change, fz.ignore_basic_report],
        toZigbee: generic.light_onoff_brightness_colortemp().toZigbee,
    },
```
I used the following config to test this in Home Assistant:
```
light:
  - platform: mqtt
    schema: json
    command_topic: "zigbee2mqtt/hall_bulb/set"
    state_topic: zigbee2mqtt/hall_bulb
    availability_topic: zigbee2mqtt/bridge/state
    name: Hall light (test)
    brightness: true
    color_temp: true
```

If this works for everyone, I can add converters to ignore the remaining messages and make a PR.

kiwinol, 2018-12-18:
> I noticed that whenever this bulb omitted the onOff attribute from the devChange genOnOff messages, an OFF state was being published to MQTT even though the bulb could still be on.
> I managed to get stable behavior with the following in devices.js:
> 
> ```
>     {
>         zigbeeModel: ['lumi.light.aqcn02'],
>         model: 'ZNLDP12LM',
>         vendor: 'Xiaomi',
>         description: 'Aqara smart LED bulb',
>         supports: generic.light_onoff_brightness_colortemp().supports,
>         fromZigbee: [fz.light_brightness, fz.light_color_colortemp, fz.generic_state, fz.ignore_basic_change, fz.ignore_basic_report],
>         toZigbee: generic.light_onoff_brightness_colortemp().toZigbee,
>     },
> ```
> I used the following config to test this in Home Assistant: (change topics to match your device id)
> 
> ```
> light:
>   - platform: mqtt
>     schema: json
>     command_topic: "zigbee2mqtt/hall_bulb/set"
>     state_topic: zigbee2mqtt/hall_bulb
>     availability_topic: zigbee2mqtt/bridge/state
>     name: Hall light (test)
>     brightness: true
>     color_temp: true
> ```
> If this works for everyone, I can add converters to ignore the remaining messages and make a PR.

Following your configuration it works fine for me 🥇

phoxy84, 2018-12-18:
I can confirm that the config works. I can turn the bulb on/off, change brightness and temp.

phoxy84, 2018-12-19:
So the supoort is there?

liads, 2018-12-19:
Yes.
Once this gets merged and referenced from the main repository, you'd be able to see it in the edge version.

gualostly, 2018-12-20:
Thanks everyone for their work in supporting this affordable bulb!
Mine arrived yesterday and I'm excited to connect it to HA via zigbee2mqtt. But I don't know how to change it to pairing mode (tried off/on 3 times, no dice). Any ideas? Thank you!

kiwinol, 2018-12-20:
> Thanks everyone for their work in supporting this affordable bulb!
> Mine arrived yesterday and I'm excited to connect it to HA via zigbee2mqtt. But I don't know how to change it to pairing mode (tried off/on 3 times, no dice). Any ideas? Thank you!

I think its 5 times.  Mine just paired when I first tried without setting it.  It will start blinking when you have reset it.

Koenkk, 2018-12-20:
Bulb is supported in the dev branch. thanks all!

closed 2018-12-20 by Koenkk

kindnation, 2018-12-29:
Do you know how long it usually takes before a device is supported in the standard version (not edge)?

phoxy84, 2019-01-04:
When will it come in standard version? It's not supported in latest version. I tried dev but latest gives a couple of errors:

```
  zigbee2mqtt:info 2019-1-4 22:20:04 MQTT publish: topic 'zigbee2mqtt/0x00158d0002c65c4b', payload '{"state":"ON","linkquality":0,"last_seen":"2019-01-04T21:20:04.275Z","color_temp":369,"color":{"x":0.462,"y":0.415}}'
  zigbee2mqtt:info 2019-1-4 22:20:10 Zigbee publish to device '0x00158d0002c65c4b', genLevelCtrl - moveToLevelWithOnOff - {"level":139,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2019-1-4 22:20:10 MQTT publish: topic 'zigbee2mqtt/0x00158d0002c65c4b', payload '{"state":"ON","linkquality":0,"last_seen":"2019-01-04T21:20:10.372Z","brightness":139,"color_temp":369,"color":{"x":0.462,"y":0.415}}'
  zigbee2mqtt:info 2019-1-4 22:20:10 MQTT publish: topic 'zigbee2mqtt/0x00158d0002c65c4b', payload '{"state":"ON","linkquality":0,"last_seen":"2019-01-04T21:20:10.414Z","color_temp":369,"color":{"x":0.462,"y":0.415}}'
  zigbee2mqtt:info 2019-1-4 22:20:10 MQTT publish: topic 'zigbee2mqtt/0x00158d0002c65c4b', payload '{"state":"ON","linkquality":0,"last_seen":"2019-01-04T21:20:10.414Z","color_temp":369,"color":{"x":0.462,"y":0.415}}'
  zigbee2mqtt:info 2019-1-4 22:20:14 Zigbee publish to device '0x00158d0002c65c4b', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:error 2019-1-4 22:20:20 Zigbee publish to device '0x00158d0002c65c4b', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
  zigbee2mqtt:info 2019-1-4 22:20:20 Zigbee publish to device '0x00158d0002c65c4b', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:error 2019-1-4 22:20:21 Zigbee publish to device '0x00158d0002c65c4b', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
```

soosterwaal, 2019-01-05:
I got a bulb since yesterday, but im unable to pair it. It simple doesn't show up. Wat is the pairing method of the bulb?

phoxy84, 2019-01-05:
Switch on five times till the bulb blinks a couple of times

honzaka, 2019-03-31:
Does anyone know what is the min. and max. value for "color_temp" in payload? Thank you.

ranchoiver, 2019-04-06:
Also can't pair the bulb even after resetting it.. How did you do it?

Do you need to hold CC2531 sniffer near the bulb?

honzaka, 2019-04-06:
I have made a “homemade pairing cable” with standard wall connector, switch and bulb socket - about 2 meters long. With this cable I am able to pair bulb just next to sniffer without problem.

GreenVine, 2019-04-06:
Same here. Switched on five times and the bulb starts blinking, but nothing is showing in the log. Can someone confirm this bulb supports channel 25?

urusha, 2019-04-08:
@GreenVine have 3 pieces on channel 25, pairs/works fine. Try to remove from mqtt/configuration.yaml/database.db/state.json and re-pair.

Perlovka, 2019-04-13:
> Does anyone know what is the min. and max. value for "color_temp" in payload? Thank you.

153 - 370

Found here:
https://github.com/TedTolboom/com.xiaomi-mi-zigbee/blob/master/drivers/light.aqcn02/device.js

GreenVine, 2019-04-13:
@urusha It looks like this bulb cannot pair directly with the router (one Philips Hue light bulb). As soon as I move the coordinator closer with Xiaomi bulb it pairs flawlessly. I didn’t spot any issues after moving coordinator back and let Xiaomi communicate with the router.

Looks like we can’t pair it with the router?

Perlovka, 2019-04-13:
There is one issue with these bulbs. They are always powered ON after power outage :-/

honzaka, 2019-04-15:
> > Does anyone know what is the min. and max. value for "color_temp" in payload? Thank you.
> 
> 153 - 370
> 
> Found here:
> https://github.com/TedTolboom/com.xiaomi-mi-zigbee/blob/master/drivers/light.aqcn02/device.js

Great, thanks!